Marus/cortex-debug #479: V0.4.4, Info : dropped 'gdb' connection

tbrlek, 2021-09-14:
I am unable debug with V0.4.4. It droppes gdb connection. It never enters main.

Reverting back to 0.4.3 fixes it, but includes a new issue (will submit it as different issue)

haneefdm, 2021-09-14:
A few details please on 0.4.4? Your launch.json is a starting point and few pointers to help reproduce the issue. What do you see in `Debug Console` and the `gdb-server` windows?

> It droppes gdb connection.

By 'it' you mean Cortex-Debug? or the gdb-server?

tbrlek, 2021-09-15:
Sorry for vague description. Everything kept working fine for a year (or year and a half) until 0.4.4. On 0.4.3 it works well (and all prior).

I have attached the following files:
[DebugConsole.txt](https://github.com/Marus/cortex-debug/files/7170516/DebugConsole.txt)
[GDB-server.txt](https://github.com/Marus/cortex-debug/files/7170517/GDB-server.txt)
[launch.txt](https://github.com/Marus/cortex-debug/files/7170522/launch.txt) (changed .json to .txt so I can upload it)

haneefdm, 2021-09-15:
Okay, that helps. So, 0.4.4 is the issue. I cannot reproduce this myself but I will see what could have gone wrong. I also can't see what changed that would affect this. Anything is possible.

Thanks for reporting back.

Is this a `launch` or `attach`. I ask because it first stopped in Cy_DelayCycles_loop. It normally stops near the reset vector.

For the gb-server output could you attach the output (stdout/stderr) from the server. You can find this in the Terminals tab in a task named 'gdb-server'.

The 'Debug Console' will also have more info if you use this in launch.json. This shows me exactly what commands were sent to gdb and its responses.
```
"showDevDebugOutput": true
```

haneefdm, 2021-10-06:
Any more info on this? I use this extension of a regular basis with PSoC6. Please report back and I will reopen the issue. A lot of things have changed since 4.3

closed 2021-10-06 by haneefdm