Marus/cortex-debug #482: OpenOCD is left running in the background after Attach -> Continue -> Disconnect secuence

bohdan-tymkiv, 2021-09-15:
Hello gentlemen,

I would like to report two similar issues but of different nature.

First one is similar to #181 and #216 - sometimes after executing Attach -> Continue -> Disconnect sequence OpenOCD is left running in the background making second attach impossible.  In previous cortex-debug releases this leads to "Error: couldn't bind tcl to socket on port 6666: Address already in use" error. Recent releases of cortex-debug pass all TCP ports (GDB, TCL and RPC) to the OpenOCD so behavior is slightly different - OpenOCD is not able to connect to the debug probe because it is already opened by still running process. This issue is quite rare, I will post more information when I will find a way to reliably reproduce it. One observation is that Attach -> Continue -> Disconnect -> Attach leads to the problem while Attach -> Disconnect -> Attach seems to work fine.
I am attaching small video which demonstrates the issue:
[oocd_left_in_background.zip](https://github.com/Marus/cortex-debug/files/7171432/oocd_left_in_background.zip)

Second one is reproducible with both SEGGER GDBServer and OpenOCD. GDB Server is not left running in the background but when debug session is launched it fails with:
SEGGER:
> ERROR: Failed to listen at socket (Err = -1)
> ERROR: Failed to open listener port 50000

OpenOCD:
> Info : starting gdb server for psoc64.cpu.cm4 on 50000
> Error: couldn't bind gdb to socket on port 50000: No error

In both cases port scanner in cortex-debug has detected that port 50000 is free so it passed it to the server but server fails to open that port. Running `netstat` shows that there are lots of TCP sockets in TIME_WAIT state. Is it possible that there is a race condition between cortex-debug and the server when server tries to open port which is not closet yet by the cortex-debug?

This can also be related to
[Error: Address already in use while binding socket with address but the port number is shown free by `netstat`](https://stackoverflow.com/questions/5106674/error-address-already-in-use-while-binding-socket-with-address-but-the-port-num)

Thanks,
Bohdan

haneefdm, 2021-09-16:
I was able to duplicate using SEGGER. It turns out that I can duplicate this even without CortexDebug trying to figure out which ports are free (just assume they are free, no test servers created, etc.). All I have to do is to launch JLink Launch session back to back. You have to be quick with the mouse/keyboard and it happens 2 out of 10 times. I just wait -- don't touch anything for a couple of mins and it works fine. Yes, it has to do with Windows leaving processes in TIME_WAIT/CLOSE_WAIT states for far too long when there is no other end and FIN/ACKs have already been done.

https://en.wikipedia.org/wiki/File:Tcp_state_diagram_fixed.svg

What is worse with Segger is that while Task manager reports that there are no JLink related processes running,  they are there in some weird state. Look at the Tray

Windows has a 4 mins time out (they claim) but that is not true either. It feels some random amount of time but not as long as 4 mins. More like a few seconds -- just like that link you pointed to.

Anyways, I don't think we or JLink is doing things wrong. But, we have to find a workaround.

Maybe in another 50 years, MS will get their networking stack right.

[image]

Another user reported this a long time ago and I could never duplicate this. If you open the tray and move your mouse a little bit these green JLink icons start disappearing. At one point there were something like 30 JLink icons in there -- meaning, they can last a super long time, hours. Nothing to do with the port issue though, I don't think. For now, this seems harmless.

Btw, this used to happen with SMB files as well. Files used to remain open long after they had been closed snf we stopped using Windows shares for anything critical.

haneefdm, 2021-09-16:
What could help is that if we don't kill the process(es) when they don't respond to detaches/disconnects. Wait a bit longer or do something different. Related to Issue #481

hwmaier, 2021-09-17:
> At one point there were something like 30 JLink icons in there

I see this all the time here too. Every debug start/stop cycle adds one notification icon. Annoying but seems purely cosmetic.

haneefdm, 2021-09-17:
@hwmaier @bohdan-tymkiv I see OpenOCD do that too, but nearly impossible to duplicate on my Mac. And in this case, I was using `launch` and not `attach`. The only time it was understandable is if gdb crashes. It has happened to me is when Cortex-Debug crashed while I am debugging it, but that is rare and is probably because I killed it.

I wish openocd supported (officially) the equivalent of `--once` option like most other gdb-servers. What that does is to exit itself when gdb drops connection (either intentionally or otherwise). JLink uses `--singlerun`

bohdan-tymkiv, 2021-09-17:
@haneefdm 
OpenOCD can be configured to execute arbitrary TCL commands/scripts when GDB detaches. Adding something like:
```
monitor [target current] configure -event gdb-detach { shutdown }
```
to the launch.json should do the job. I will try to test this solution but I'm still looking for a way to reproduce the original problem reliably. This method works just fine from the command line, it even works if GDB process is killed:

https://user-images.githubusercontent.com/10281082/133783385-6d8c45aa-a7a0-40ce-8729-e806e400e2b3.mp4

Bohdan

haneefdm, 2021-09-17:
Can you try the following pre-release?

https://github.com/Marus/cortex-debug/releases

haneefdm, 2021-09-17:
Yes, I know about the tcl trick and the 'monitor' commands. Thing is we don't want to add yet another set of overrides for this. We may eventually do but have not gotten to that point yet. I have to modify  7-8 gdb-servers

I never thought of adding this (your suggestion) to the startup commands
```
    monitor [target current] configure -event gdb-detach { shutdown }
```

What does '[target current]` mean when multiple processors are active at the same time but we are only debugging one, or multiple. I did not know that OpenOCD has a concept of a `current target`. I thought it always has a set of targets. We have another issue with the target name where it was hard-coded to. We have the following now, that does not work for a few people - PSOC :-)

```
monitor [target current] configure -event gdb-detach { shutdown }
```
I want to change that with what you suggest.

bohdan-tymkiv, 2021-09-17:
OpenOCD often has to deal with multiple targets at the same time. If there are two PSoC6 devices in the JTAG chain, OpenOCD will see four different targets - something like psoc6_1.cm0, psoc6_1.cm4, psoc6_2.cm0, psoc6_2.cm4. PSoC6 has common address space for both cores but, just as an example, lets assume cm0 and cm4 have completely different memory spaces. Imagine the situation, we need to read some memory region of psoc6_1 's cm4 core and compare it to the region of psoc6_2 's cm0 core. This can be done using the following commands in the TCL script:
```
# Switch current target to JTAG device #1, Access Port 2
targets psoc6_1.cm4
# Dump 4k of memory 
dump_image data.bin 0x10000000 0x1000
# Switch current target to JTAG device #2, Access Port 1
targets psoc6_2.cm0
# Compare with previously dumped data
verify_image data.bin 0x10000000
```
If you execute `dump_image data.bin 0x10000000 0x1000` in the script (or command line) but target was not explicitly set before - some "default target" will be used. Usually this will be the last core of the last device in the JTAG chain but this is not guaranteed.

In case of commands issued via GDB connection (including `monitor`) - there is no need to specify target explicitly. Default target is automatically set to to the CPU core GDB is connected to. In this way GDB's `load` command will always program flash of the CPU we are currently debugging. The good thing about using `[target current]` in monitor commands is that you don't have to think about which target you are configuring. The `[target current]` will be evaluated to the name of the target which is behind the given GDB port - the target we are currently debugging. Two GDB processes connected to different OpenOCD ports have different default target (but nothing stops you from switching to a different target ,do something there and then switch back).

Regarding your proposal - I don't think something like
```
monitor [target current] configure -rtos auto
```
will work because RTOS detection happens only once when GDB<->OpenOCD communication starts (when GDB provides symbol information). It will be too late to configure RTOS via `monitor` commands. On the other hand, `[target current]` cannot be 100% successfully used in the OpenOCD command line because `[target current]` will return some "default target". With PSoC6 this will be cm0 core of the last device in the JTAG chain which is usually not what you want.
Using `$_TARGETNAME` will not work either because this is just a TCL variable, the name can be different for different chips/config files.

I think the only correct way to configure RTOS is to explicitly specify the name of the target, this should be done somewhere in `openOCDLaunchCommands` or in `configFiles`, this configuration should happen before GDB connects.

Bohdan

haneefdm, 2021-09-25:
I think having a shutdown event handler to shutdown is best done in the config file. In OpenOCD documentation, it says

I tried many things and the only thing that works is to send a SIGTERM. I tried to send a SIGINT before we send a SIGTERM. Does not seem to help do a graceful shutdown. OpenOCD doc says the following. SIGINT emulates a CTRL-C but a process can chose to ignore it if not coming from a terminal or do something else.

> If user types CTRL-C or kills OpenOCD, either the command shutdown or its replacement will be automatically executed before OpenOCD exits.

While OpenOCD does not hang around all the time, it is still annoying that there is no good way to ask it to quit. And, no 'monitor shutdown' is not a solution as it is interfering with our shutdown process with GDB itself.

JLink is well-behaved in this regard and does the right thing now.

Closing for now until we find a better solution. Re-open if you have ideas

haneefdm, 2021-09-27:
Btw, what commands does Eclipse use to shutdown OpenOCD?

haneefdm, 2021-09-29:
@bohdan-tymkiv I installed a diconnect handler as we discussed. I still do not trust the [target current] or what that means. So, I put it on all targets -- for now. You can check it out if you wish

https://github.com/Marus/cortex-debug/releases/tag/V0.4.7-pre2