Marus/cortex-debug #483: Drop use of objdump in favor of GDB functions

GitMensch, 2021-09-17:
[...] we don't use the disassembly from objdump. Just the function addresses (symbol table) to get a start/end of a function. The symbol table is also used to display Globals and Statics.[...]

_Originally posted by @haneefdm in https://github.com/Marus/cortex-debug/issues/325#issuecomment-921743536_

As noted in #242 `info functions` will also list all functions (and can be limited by adding a regexp), `info address myfunc` and `print &myfunc` / `examine &myfunc` will show the start address of the function.
`info variables` 

The main use of objdump has a note of

>  Wish gdb could give us that info rather than using objdump

I _think_ it does.

haneefdm, 2021-09-17:
Thanks for the feedback, but...

Too many things here. For now, not going to drop objdump. It is needed locally just like we need your source code and gdb as well. Like I said, we do not use objdump for disassembly. We use gdb. We use objdump to find function and variable symbols which is an order of magnitude faster than using gdb -- matters when your executable is large.  As you know, not everything you see as a gdb-command is available as a MI command.

https://sourceware.org/gdb/onlinedocs/gdb/GDB_002fMI.html

Some of the things you mention are HUGE initiatives (multi-targets). I can't deal with them in an issue like this. I have it on the roadmap. I am the only one working on this extension and I have a day job.

So, we can't have an issue like "drop objdump". It gives us none of the features you mention in your list.

closed 2021-09-17 by haneefdm

GitMensch, 2021-09-17:
The list was not updated, this was an error which I adjust in any case.
The _intention_ for this issue was not about doing it now, but to "track" pro/con and related issues in a single place.

From what you've mentioned it looks like "if there is a PR and it is not much slower than the current version it has a good chance to be included".

haneefdm, 2021-09-17:
>> Wish gdb could give us that info rather than using objdump
> I think it does.

Not very well and super slow. I had tried and gave up.

GitMensch, 2021-09-17:
Good to know.