Marus/cortex-debug #485: Would it be possible to use objdump on the client side?

GitMensch, 2021-09-17:
I think file performance is likely the limiting part, but to recheck: If the client has objdump available (which would obviously match the target) we could execute it via gdb and `!objdump`, then "somehow" get the file (gdb offers `remote get` but documents that this only works for some targets).
Theoretically the output of objdump could be read via the gdb console (only a guess) and be sent to a file from there, too.

haneefdm, 2021-09-17:
Yes it could work, maybe. There are also other elf tools that do something similar to objdump. It is interesting that pyocd uses an elf module and it knows how to get info from there. Not sure of the actual mechanics. We don't need anything target-specific from the elf file. Just symbols which are not target specific.

What is your target? Or what other elf tools does your target support?

GitMensch, 2021-09-17:
My targets are actually no match (yet) for this extension: x64 and x86, along with "fat" ARM (Raspi).
The python modules are a cool thing (I use them heavily), but especially on "small tin" I'm not sure that this would work: as I understand GDB is often run on the board, not gdbserver [or is this a wrong impression] and GDB on a board is not always likely to have python support enabled.
If `pyocd` uses an elf module (not a gdb one) and accesses the symbols via that, then this could only be done where python runs (locally), it wouldn't work on the gdbserver side because it needs physical access to the object files.

Hm, is it _likely_ to find grep or sed on the gdb target side? If yes then gdb could start something like `!objdump -Wi $something | grep -e something -e another | sed `s/transform/here/g` which would limit the size of the data to parse... Just thinking...

Considering use of other elf tools is good, too - it think if objdump is available one could expect complete binutitls, too.

haneefdm, 2021-09-17:
> My targets are actually no match (yet) for this extension: x64 and x86, along with "fat" ARM (Raspi).

Wouldn't MS cpptools extension be more appropriate for you?

About reducing the data size using grep, I don't know, may make sense if data transfer is the bottleneck.

GitMensch, 2021-09-17:
> Wouldn't MS cpptools extension be more appropriate for you?

Hm, from https://github.com/microsoft/vscode-cpptools/blob/main/LICENSE.md

> Additional binary files may be downloaded at runtime after the initial installation; these are governed by the more restrictive proprietary license terms found at RuntimeLicenses.

and all of those (the cpptools, -server -debugger) have

> You may install and use any number of copies of the software **only with Microsoft Visual Studio, Visual Studio for Mac, Visual Studio Code**, Azure DevOps, Team Foundation Server, and successor Microsoft products and services to develop and test your applications.

and 

> For example, if Microsoft technically limits or disables extensibility for the software, you may not extend the software by, among other things, loading or injecting into the software any non-Microsoft add-ins, macros, or packages; modifying the software registry settings; or adding features or functionality equivalent to that found in Microsoft products and services. You may not:
a)  work around any technical limitations in the software that only allow you to use it in certain ways;
b)  reverse engineer, decompile or disassemble the software, or otherwise attempt to derive the source code for the software, except and to the extent required by third party licensing terms governing use of certain open source components that may be included in the software;
[...]
e)  share, publish, distribute, or lease the software (except for any distributable code, subject to the terms above), provide the software as a stand-alone offering for others to use, or transfer the software or this agreement to any third party.

So... no, not until I can use it wherever I want to and if needed bug-fix issues and remove unnecessary limitations. MS made it very clear that neither the components of the cpptools nor the components of remote-* package will be available with source code under a free license (and the python extension now also includes a non-free extension as language server, so we may see the same move with other previously working extensions, too).

haneefdm, 2021-10-06:
I am going to mark this as a "no"