Marus/cortex-debug #487: How to use alternative RTT viewer tools

bmcdonnell-fb, 2021-09-22:
In our application firmware, we're using Segger RTT over SWD (without an SWO pin) for console output.

Following cortex-debug release V0.4.0, we changed our project's `.vscode/launch.json` in a branch to utilize the new built-in RTT support (diff pictured below). Now with that change, I can no longer connect to the console output with other tools, such as Segger's RTT Viewer, or a telnet client on port 19021. How can I get that working again?

I see in the [changelog](https://github.com/Marus/cortex-debug/blob/master/CHANGELOG.md#v040), it says,

> It is perfectly fine to have Cortex-Debug enable RTT but not use any display features. This way you can use external tools (like JLink tools or custom ones)

That seems to be referring to what I'm asking, but how do I "not use any display features"? And does that mean I need to edit the `launch.json` whenever I want to use a different tool?

I tried removing the `"decoders"` key & value from the `"rttConfig"` section in `launch.json`, and was able to connect with Segger RTT Viewer, but not a telnet client (KiTTY on port 19021). Is this what the quote above means? If so, why doesn't it work with the telnet client?

Is there a way to make it both output to the built-in Terminal, and support support external viewers simultaneously? Or to let me choose between them at runtime (without making config file changes)?

... Currently, the main features that have me wanting to use other viewers are (1) logging to (uniquely named) files, and (2) per-line timestamping. If those were possible with VSCode / Cortex Debug, I would probably just use that, and not care about the external viewers, at least for the time being. But I don't mind using external viewers to get those features. I just need some way to do it.

Thanks for your work on these tools and features. 😃 

[image]

haneefdm, 2021-09-23:
@bmcdonnell-fb I thought JLink supported having multiple TCP clients. It has to support that since it is the server that owns that TCP port. I may have confused JLink with OpenOCD. But, you do see the output in our `TERMINAL` tab right? I will see if we can support never making a TCP connection if there are no decoders, so you can use it with external tools.

Yes, a timestamp would be a good idea. I always wondered, how do you actually use the timestamp? I am curious how it is useful as the timebase and resolution is very different from MCU.

You probably already know, this same output can also (in addition to) to other decoders (just use same port) and also log fo a file. Unfortunately, no facility to uniquely name the file but you could probably do it (automated) externally -- kinda painful. We only make one connection to JLink though, data comes into Cortex-Debug and gets split.  But don't try to use multiple input consoles for the same RTT channel.

Keep in mind though, unlike OpenOCD, JLink really only supports one TCP channel (and thus one RTT channel) via the gdb-server. This is unfortunate.

hwmaier, 2021-09-23:
@bmcdonnell-fb We keep for that reason 2 launch settings we can switch in between. One where cortex-debug's RTT viewer is used and a second which I use when working on a big screen and have space for a second windows with Segger's RTT Viewer. As @haneefdm has mentioned, JLink's GDB server allows only one TCP connection, so this is either the cortex-debug's RTT viewer or the external one. If a second on is opened, the J-Link gdb server declines it with the message `ERROR: Connection refused - There already is an active connection.`

bmcdonnell-fb, 2021-09-23:
@haneefdm,

> you do see the output in our `TERMINAL` tab right?

Yes.

> Yes, a timestamp would be a good idea. I always wondered, how do you actually use the timestamp? I am curious how it is useful as the timebase and resolution is very different from MCU.

This is very situational. It can be useful just to have an idea of how far apart in time things happened, especially if you're not actively monitoring it. You make a good point though that the developer should be aware of the time resolution.

> You probably already know, this same output can also (in addition to) to other decoders (just use same port) and also log fo a file.

I don't, actually. Could you point me to relevant documentation?

bmcdonnell-fb, 2021-09-23:
@hwmaier,

> We keep for that reason 2 launch settings we can switch in between.

How do you switch between them? Do you just keep the alternate `launch.json` in a branch, and switch to or cherry-pick it? Or is there some friendlier way?

> One where cortex-debug's RTT viewer is used and a second which I use when working on a big screen and have space for a second windows with Segger's RTT Viewer.

Good point, that can be a good reason to want to use an external viewer - just for screen real estate. Especially since VSCode is still limited to a single display.

haneefdm, 2021-09-23:
> How do you switch between them? 

I think @hwmaier meant to have two+ configurations in one launch.json. They would identical except for RTT enabled/disabled. I switch between them using the drop-down for `Run & Debug`. You can also use an environment variable or User/workspace setting, etc. none of which are as simple as switching between configurations.

[image]

I generally have 2-6 configurations like that -- One for launch, one for attach, using JLink or OpenOCD, dual/single core, etc.

haneefdm, 2021-09-23:
> I don't, actually. Could you point me to relevant documentation?

https://github.com/Marus/cortex-debug/wiki/SEGGER-RTT-support

Search for `logfile`. Each decoder can have one. I am sure you will let us know if it is not working. Not many users for it.

If you use this, you can do a `tail -f <log-file-name>` on a larger screen and view the output in a separate terminal

hwmaier, 2021-09-24:
@bmcdonnell-fb I use the *Run and Debug* drop down menu to switch between a RTT and not-RTT as shown below:

[image]

bmcdonnell-fb, 2021-09-24:
I'd like to have a config in my `launch.json` _using the `rttConfig`_ that allows for viewing the firmware output with external viewers, e.g. telnet client on port 19021. Is this expected to work now? What would be the corresponding config options?

With this first config...

```
            "rttConfig": {
                "enabled": true,
                "address": "auto",
                "decoders": [
                    {
                        "port": 0,
                        "type": "console",
                        "noprompt": true
                    }
                ]
            },
```

...it outputs to the built-in RTT console as expected.

I thought (hoped) one of the following configurations would allow me to connect with an external RTT viewer. However, in these cases it appears the debugger isn't even connecting to the device. It never hits the breakpoint at the top of `main`, as indicated by the greyed-out debugger control buttons.

[image]

```
            "rttConfig": {
                "enabled": true,
                "address": "auto",
                "decoders": []
            },
```

```
            "rttConfig": {
                "enabled": true,
                "address": "auto"
            },
```

```
            "rttConfig": {
                "enabled": true
            },
```

The Cortex-Debug [SEGGER-RTT-support wiki page](https://github.com/Marus/cortex-debug/wiki/) says, "There is only one property needed to get started: `enabled`." But as noted above, for me, that didn't even allow the debugger to connect. It doesn't output an error or indicate the failure. The GUI just sits there acting like it's going to connect, but doesn't.

bmcdonnell-fb, 2021-09-24:
Here's the bit from our config that does work for external RTT viewers, in place of the `rttConfig` properties. But I'd rather simplify by using `rttConfig` if possible, as I think that would be easier to maintain.

```
            "swoConfig": {
                "enabled": true,
                "cpuFrequency": 168000000,
                "swoFrequency": 4000000,
                "source": "probe",
                "decoders": [
                    {
                        "type": "console",
                        "label": "ITM",
                        "port": 0
                    }
                ]
            },
```

haneefdm, 2021-09-24:
Right now, you have to set `"enabled": false` to use an external viewer. Once disabled, you can do what you want with the external viewer that is up to you.

If `"enabled": true`, then we try to connect

haneefdm, 2021-09-24:
Note that while `swoConfig` and `rttConfig` may look the same, they are entirely different things and use different mechanisms.

bmcdonnell-fb, 2021-09-24:
This did the trick, thanks. 😃 

```
"rttConfig": {
                "enabled": false
            },
```

In `launch.json`, is there a way to specify both `rttConfig` options (`"enabled": true` with other properties, and `"enabled": false` without) under the same "configuration", such that I could select between them in the GUI? Or will I have to have two otherwise identical configurations, differing in those properties and `"name"` only?

haneefdm, 2021-09-24:
Unfortunately, two nearly identical configs are needed. This file is controlled by VSCode and even the JSON format is not very nice. You should see our package.json where we describe this debugger. I have to duplicate thousands of lines of JSON for launch vs attach.

Btw, in the next release, you will be able to add timestamps. Here is a sample output using SEGGERs RTT demo
```
[2021-09-24T19:07:16.642Z] printf Test: %-8.6d,  -12345 : -012345 .
[2021-09-24T19:07:16.642Z] printf Test: %-08d,   -12345 : -12345  .
[2021-09-24T19:07:16.642Z] printf Test: %-08.6d, -12345 : -012345 .
[2021-09-24T19:07:16.642Z] printf Test: %-0d,    -12345 : -12345.
```

bmcdonnell-fb, 2021-09-24:
> Unfortunately, two nearly identical configs are needed.

Thanks for the info.

> Btw, in the next release, you will be able to add timestamps.

Nice!

bmcdonnell-fb, 2021-09-24:
Back to my original question,

> I see in the [changelog](https://github.com/Marus/cortex-debug/blob/master/CHANGELOG.md#v040), it says,
> 
> > It is perfectly fine *to have Cortex-Debug enable RTT* but not use any display features. This way you can use external tools (like JLink tools or custom ones)

[New emphasis mine.]

It's not clear to me from that statement (nor the wiki) that you *enable* the base RTT functionality (without any of Cortex-Debug's RTT viewers) by setting `"rttConfig": { "enabled": false }`. Did I say that correctly?

Could you clarify that on the [wiki page](https://github.com/Marus/cortex-debug/wiki/)? And maybe that statement in the changelog, too. I'd do it myself, but I don't think I know all the correct names for things.

haneefdm, 2021-09-24:
That was the original intent. I will see what went wrong. The idea was to have it enabled but not have any decoders. Between OpenOCD and JLink we ran into some differences and ended up not getting implemented properly.

haneefdm, 2021-09-24:
I don't have a JLink handy right now. I have been using OpenOCD. So, I will test when I get my hands on a JLink

bmcdonnell-fb, 2021-09-24:
> With JLink, how did you use it?

Self quote:

> With this first config...
>
> ```
>             "rttConfig": {
>                 "enabled": true,
>                 "address": "auto",
>                 "decoders": [ [...]

With this minimal config....

```
            "rttConfig": {
                "enabled": false
            },
```

...it allows me to connect with external viewers.

I'm not having the same luck with an ST-Link. But that's why I switched to J-Link in the first place, so I could get on with application development.

haneefdm, 2021-09-24:
There was a bug for JLink. In the next release, you can leave it enabled, have no decoders, and use the JLink external viewer. Cortex-Debug will do the configuration and pass the command-line argument.

[image]

But there is a trick, however. You have to configure the viewer properly. In my case, the following worked.

[image]

Other methods may also work.

hwmaier, 2021-09-25:
> Btw, in the next release, you will be able to add timestamps. Here is a sample output using SEGGERs RTT demo

Probably should go into a different thread, but you mentioned now time stamping here. This would be a very useful addition. Regards the time format, Wireshark for example allows to show absolute time or relative time. For a lot of cases I find time stamping with relative time in seconds.ms format since system start more easier to read. Just a suggestion.

haneefdm, 2021-09-30:
@hwmaier Yup, there is an elapsed time when you ask for timestamps (both RTT and SWO)
```
[2021-09-30T15:23:34.468Z, +003249ms] printf Test: %.3c,       'E' : E.
```
I am not sure how useful this is. Then again, I don't have a very long-running test. I should probably remove the `+` sign as it implies addition to the previous time. I meant it to be elapsed time.

haneefdm, 2021-10-06:
I think we are done with this issue

closed 2021-10-06 by haneefdm