Marus/cortex-debug #491: SWO Console is not showing in VSCode

boningdong, 2021-09-26:
I've been using Cortex Debug plug-in in VScode for a while.
But for some reason, the SWO console is missing recently even with the following configurations:
`/* SWO configurations */
      "swoConfig": {
        "enabled": true,
        "cpuFrequency": 170000000,
        "swoFrequency": 2000000,
        "source": "probe",
        "decoders": [
            { "type": "console", "label": "SWO", "port": 0 }
        ]
      },`

Before when it's working, after starting debugging, I can choose the SWO console in the output. But right now, it doesn't show up.

haneefdm, 2021-09-26:
Check the Terminal tab. It has moved recently.

boningdong, 2021-09-26:
Thank you for your reply! I still couldn't see it. in the Runtime status of the cortex debug, I saw the following status:
`Unhandled error. ('Error: Failed to connect to port 50003 Error: Error: Failed to connect to port 50003 Error: connect ECONNREFUSED 127.0.0.1:50003')`

Here is the openocd command the DEBUG_CONSOLE shows:
`Launching server: "C:\SOFTWARES\OPENOCD\BIN\OPENOCD.EXE" "-c" "gdb_port 50000" "-c" "tcl_port 50001" "-c" "telnet_port 50002" "-s" "C:\Users\dbndb\Desktop\Projects\Scopen-Dev-Firmware" "-f" "interface/stlink.cfg" "-f" "target/stm32g4x.cfg" "-c" "tpiu config internal :50003 uart off 170000000 2000000"`

haneefdm, 2021-09-26:
That means that OpenOCD has not understood the SWO or SWO has not been enabled. In the Terminal tab (gdb-server), check for errors from OpenOCD. The TPIU that you see on the command line seems not to have worked. OpenOCD has to serve the SWO output on TCP port 50003

```
"-c" "tpiu config internal :50003 uart off 170000000 2000000"
```

boningdong, 2021-09-26:
That makes sense, but I couldn't see any error shows up related to the SWO port. Here is the log. Is that because the Openocd version is too low?

```SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected
Open On-Chip Debugger 0.10.0 (2020-01-14) [https://github.com/sysprogs/openocd]
Licensed under GNU GPL v2
libusb1 09e75e98b4d9ea7909e8837b7a3f00dda4589dc3
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : clock speed 500 kHz
Info : STLINK v2 JTAG v34 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.217426
Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Stlink adapter speed set to 480 kHz
Info : stm32l4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 50000 for gdb connections
Info : accepting 'gdb' connection on tcp/50000
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x81000003 pc: 0x08000f6e msp: 0x2001ff04
Info : device id = 0x20016469
Info : flash size = 512kbytes
Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Stlink adapter speed set to 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800d75c msp: 0x20020000
Info : Padding image section 0 at 0x080001d8 with 8 bytes
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000004e msp: 0x20020000
Warn : block write succeeded
Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Stlink adapter speed set to 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0800d75c msp: 0x20020000```

boningdong, 2021-09-26:
Actually, updating to the latest openocd solved the problem! Thanks!

closed 2021-09-26 by boningdong