Marus/cortex-debug #492: [BUG] Call Stack Empty on STM32H750 Device Using Jlink Debugger

PGIII, 2021-09-27:
Call Stack is Empty on latest version using a JLink Debugger with STM32H750 MCU. Has issue with 0.4.5, reverting to 0.4.2 fixes the issue

haneefdm, 2021-09-27:
Can you attach some details? `launch.json`, Output from JLink gdb-server (see Terminal Tab), output from the 'Debug Console`.

You may also want to enable verbose debug data for the debug console.. Use the following in `launch.json`
```
"showDevDebugOutput": true
```

For 0.4.5, we tested with JLink more than any other server and most of the changes were actually in the end on how the session terminates.

DDevine, 2021-09-28:
I think I briefly saw this issue yesterday... But I can't replicate it today.

haneefdm, 2021-10-06:
Please report back or file a new issue with details to see if we can reproduce it. If you saw it, I am sure it happened.

closed 2021-10-06 by haneefdm