Marus/cortex-debug #493: v0.4.5 does not kill openocd process anymore after pressing stop button

sourcebox, 2021-09-28:
So I have to kill it manually from the terminal everytime after a launch. If this is caused by some new feature or configuration setting, please provide detailed information how to restore the previous behaviour.

haneefdm, 2021-09-28:
Yes, some changes have occurred in 0.4.5 that could affect this. Can you give me some details?

* OS
* OpenOCD details. There are too many (custom) versions with the same version number. Where did you download it from?
* Attach your `launch.json`, is it a `Launch` type or `Attach` type session?
* Any steps to reproduce.

You can revert to 0.4.4 -- VSCode extension manager lets you do this but then you may get stuck at that version. It is better to help me fix the issue, but you decide

sourcebox, 2021-09-28:
Of course:
- OS: Linux
- OpenOCD: latest commit from git c3993d3188da5976a64f47d4cbf4d7e5b63f0c8d

launch.json
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "cortex-debug",
            "request": "launch",
            "name": "Debug (OpenOCD)",
            "servertype": "openocd",
            "cwd": "${workspaceRoot}",
            "preLaunchTask": "make",
            "runToMain": true,
            "executable": "./build/example.elf",
            "device": "STM32L452VCT6",
            "configFiles": [
                "openocd.cfg"
            ],
            "svdFile": "${workspaceRoot}/tools/STM32L4x2.svd",
            "postLaunchCommands": [
                "monitor arm semihosting enable"
            ]
        }
    ]
}
```

After pressing F5, openocd is launched and debugging works as expected. Pressing the Stop button stops it, but openocd still runs in the background. So next invokation via F5 fails because openocd is already running. Setting `request' to attach does not help in this regard.

haneefdm, 2021-09-28:
Ok, that helps. We will try to reproduce this.

AndriiLishchynskyi, 2021-09-28:
Hello,
Adding this post for statistic - I also observe behavior described by @sourcebox.
OSes: Ubuntu 20.04, macOS Big Sur v.11.2.3
OpenOCD error:
libusb_claim_interface() failed with LIBUSB_ERROR_BUSY
unable to open ftdi device with vid 04b4, pid f900, description '', serial '' at bus location '*'

haneefdm, 2021-09-28:
I am able to duplicate this problem. Unfortunately, it works fine when I launch the extension within a debugger but the production version fails. Investigating...

Basically, I cannot debug my own extension, so I have to try other ways.

haneefdm, 2021-09-28:
There is now a new release. Hopefully, this is fixed. I was able to duplicate/fix this on both Mac and Linux

haneefdm, 2021-09-28:
It may take a while for VSCode marketplace to push the release. You can pickup the latest from

https://github.com/Marus/cortex-debug/releases/tag/V0.4.6

and install the VSIX file manually in VSCode. Do not double click on it. Use VSCode for the installation.

haneefdm, 2021-09-28:
Anone following this thread, please confirm if the problem is fixed after installing 0.4.6

closed 2021-09-28 by haneefdm

sourcebox, 2021-09-29:
I installed 0.4.6 and had a look. In `attach` mode, openocd is shut down now, but in `launch` mode, it still remains active after pressing stop.

haneefdm, 2021-09-29:
It quits fine for me on Windows/Mac/Linux with either attach or launch. I have a feeling this is VSCode exiting/killing our debug extension before we had a chance to cleanup ... because it is taking too long. I have pretty fast machines though, but not sure that has an effect.

Will try another approach.

haneefdm, 2021-09-29:
Can you try our new approach. We install a gdb disconnect even handler in OpenOCD to auto shutdown when gdb disconnects. This should happen every time now, attach or launch does not matter.

https://github.com/Marus/cortex-debug/releases/tag/V0.4.7-pre2

sourcebox, 2021-09-29:
This look good. I tried in `launch` mode only a few times, but it works again as expected. Thanks for your efforts!

AndriiLishchynskyi, 2021-10-05:
We are experiencing this issue now with v0.4.7, @sourcebox does it work for you?

haneefdm, 2021-10-05:
@AndriiLishchynskyi I cannot duplicate this with 0.4.7 on either Windows or Mac, launch or attach, stop or disconnect. What platform? Now I am wondering if VSCode is killing us before we have a chance to clean up.

Did not try Linux.

AndriiLishchynskyi, 2021-10-05:
Sorry for missing details. It was identified at least on two machines: macOS BigSur 11.6 and macOS Catalina 10.15.7.
I can collect and sent you logs, just let me know what is needed.

AndriiLishchynskyi, 2021-10-05:
After a conversation with @haneefdm it turned out that the problem was in my launch.json. Thanks a lot for identifying the the root cause!
Thus, no issue in Cortex-Debug.

kl4m4, 2022-01-21:
> After a conversation with @haneefdm it turned out that the problem was in my launch.json. Thanks a lot for identifying the the root cause! Thus, no issue in Cortex-Debug.

Hi! Could you elaborate on what exactly was the problem. I'm  experiencing exactly the same issue now, would appreciate some help.

kl4m4, 2022-01-28:
Hi! I switched OpenOCD from v0.10 to v0.11, and  the problem dissapeared completely.
Anyway: thank you for developing cortex-debug, you work is very usefull and appreciated. Cheers!