Marus/cortex-debug #495: config files .cortex-debug.json in wrong workspace

JojoS62, 2021-09-29:
My project is using a workspace with several directorys. To get the git extension showing all included subdirs with git, I add the main workspace as last entry in the workspace file.
The cortex-debug extension creates .cortex-debug.peripherals.state.json and .cortex-debug.registers.state.json. But not in my main workspace, they are put in the first folder in the workspace.

this is part of my .workspace.code-workspace:
```
{
	"folders": [
		{
			"path": "lvgl"
		},
		{
			"path": "mbed-os"
		},
		{
			"path": "lvglDriver"
		},
		{
			"path": "custom_targets"
		},
		{
			"path": "."
		}
	],
```

The cortex config is now created in lvgl/.vscode, which is not wanted because it changes the git repo.
cortex-debug has
```
"cwd": "${workspaceRoot}",
```
so it should use this for storing the configuration. How can I fix it?

haneefdm, 2021-09-29:
I bet the same thing is happening for registers. We have to look at this carefully. VSCode did not have workspaces and then all of a sudden they introduced it. It wasn't clear at the time what the semantics were. It is still not clear to me, but need to dig deeper.

I have been meaning to move these out of the .vscode altogether as this is transient information (mostly). People check-in the .vscode dir into source-control and they have to put these files in .getignore or equivalent.

Will look into it shortly.

haneefdm, 2021-09-29:
Workspaces are designed for related projects to be managed together. So, it could be there are multiple executables here as well and each executable should be treated on its own. You can have multiple `launch.json`s as well.

https://code.visualstudio.com/docs/editor/workspaces#_workspace-tasks-and-launch-configurations

JojoS62, 2021-09-29:
yes, I start the launch config in project ., the debug-cortex config is written fix  in ./lvgl project because this is first in workspace list.

JojoS62, 2021-10-09:
sorry, its not quite perfect, the debug_info.json is still written into the first workspace in the list.