Marus/cortex-debug #496: JLink shows me GDB timeout since update to 0.4.5 / 0.4.6 (Windows)

hurricanefrog, 2021-09-29:
Hi,

since this extension updated to 0.4.5 or 0.4.6, I get a GDB timeout when trying to launch a debug session. Reverting to 0.4.4 fixes this problem. I'm running it under Windows 10.

Relevenat settings from settings.json:

```
    "cortex-debug.armToolchainPath": "C:/Users/me/Downloads/gcc-arm/bin",
    "cortex-debug.JLinkGDBServerPath": "C:/Program Files (x86)/SEGGER/JLink/JLinkGDBServer.exe",
```

Example launch config:

```
    {
        "cwd": "${workspaceRoot}",
        "executable": "./Debug/executable.elf",
        "name": "Debug Cortex",
        "request": "launch",
        "type": "cortex-debug",
        "servertype": "jlink",
        "device": "NRF52840_XXAA"
    }
```

The JLink GDB Server window appears, but no connection is made.

Setting `"showDevDebugOutput": true` doesn't do anything (probably because GDB doesn't even start?).

Is there any other information I can provide to you to help debug the problem?

haneefdm, 2021-09-29:
> The JLink GDB Server window appears, but no connection is made.

What is the JLink version?

Can you also check in the `OUTPUT` tab (`Adapter Output`). Also, run the JLinkGDBServer from the command line using the exact same options you see in `Debug Console`. We look for a particular pattern on stdout/stderr. If that pattern changed, then it will not work (Issue #494). Here is the regexp

```
/Waiting for GDB connection\.\.\./g
```

There was a change in 0.4.5/6 where we had two ways of look for a server to start and we removed one of them because it was interfering with the server even getting created.

@trond-snekvik Could I ask you to take a look here? The only change here is that delay that was added from JLink initial connection. I don't have a Nordic device and you probably don't either. Perhaps the delay was not enough. I hate add more, but you added the delay for the opposite reason. It would output the initMatch too early. Btw, this would be reported to SEGGER.

christoph-nils, 2021-09-29:
Hello,

I have the same issue with the ST-LINK_gdbserver.exe
`"cortex-debug.stlinkPath": "path_to_gdbserver/ST-LINK_gdbserver.exe",`
launch.json
`        {
            "name": "Debug",
            "cwd": "${workspaceFolder}",
            "executable": "${workspaceFolder}/Build/Debug/STM32WL55CM4.elf",
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "stlink",
            "svdFile": "${workspaceFolder}/STM32WL5x_CM4.svd",
            "device": "STM32WL5x",
            "interface": "swd",
            "runToEntryPoint": "main",
            "serverArgs": [
                "-v",
                // "--frequency 24000", // 24000Khz if your ST-Link can run on it.
                //"-e", // Persistent
                "-l 1", // log verbosity
                "-s", // Verify flash
                "-m 0", // AppID to debug
                "-k", // Init under reset,
                "-t", // shared
            ],
            //"showDevDebugOutput": true,
            // disable target-download at launch through override
            "overrideLaunchCommands": [ 
                "interpreter-exec console \"monitor reset halt\"",
            ]
        },`

haneefdm, 2021-09-29:
@christoph-nils For STLink, you can try the patch release mentioned in #494

hurricanefrog, 2021-09-29:
> What is the JLink version?
> 

It's V6.88a

> Can you also check in the `OUTPUT` tab (`Adapter Output`). Also, run the JLinkGDBServer from the command line using the exact same options you see in `Debug Console`. We look for a particular [...]

If I run it from the console (cmd.exe), there is no output at all.

In VScode, Output tab, Adapter output shows:

> DEPRECATED: Please check the 'TERMINALS' tab for 'gdb-server' output. This 'Adapter Output' will not appear in future releases
> QLayout: Attempting to add QLayout "" to MainWindow "", which already has a layout
> QLayout: Attempting to add QLayout "" to MainWindow "", which already has a layout

Tab Terminal, gdb-server shows

> SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected
> QLayout: Attempting to add QLayout "" to MainWindow "", which already has a layout
> QLayout: Attempting to add QLayout "" to MainWindow "", which already has a layout
> SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
> GDB server session ended. Waiting for next server session to start...

JLinkGDBServer window shows:

> SEGGER J-Link GDB Server V6.88a GUI Version
> 
> JLinkARM.dll V6.88a (DLL compiled Nov 18 2020 15:09:23)
> 
> -----GDB Server start settings-----
> GDBInit file:                  none
> GDB Server Listening port:     50000
> SWO raw output listening port: 50001
> Terminal I/O port:             50002
> Accept remote connection:      localhost only
> Generate logfile:              off
> Verify download:               off
> Init regs on start:            off
> Silent mode:                   off
> Single run mode:               on
> Target connection timeout:     5000 ms
> ------J-Link related settings------
> J-Link Host interface:         USB
> J-Link script:                 none
> J-Link settings file:          none
> ------Target related settings------
> Target device:                 NRF52840_XXAA
> Target interface:              SWD
> Target interface speed:        4000kHz
> Target endian:                 little
> 
> Connecting to J-Link...
> J-Link is connected.
> Firmware: J-Link EDU Mini V1 compiled Jun 24 2021 14:29:39
> Hardware: V1.00
> S/N: 801006614
> Feature(s): FlashBP, GDB
> Checking target voltage...
> Target voltage: 2.94 V
> Listening on TCP/IP port 50000
> Connecting to target...
> Connected to target
> Waiting for GDB connection...

Also, the window (GUI) appears, despite the -nogui switch. Seems to be related.

hurricanefrog, 2021-09-29:
Could it be JLinkGDBServer.exe (GUI version) vs. JLinkGDBServerCl.exe (CLI version)?

hurricanefrog, 2021-09-29:
Closing this, since this is not a bug. The problem occurred due to a wrong setting I did myself.

Previous versions of Cortex-Debug simply accepted using the GUI Version of J-Link GDB server (JLinkGDBServer.exe), since there was no stdout/stderr checking.  
v0.4.5 added stdout checking. Since there is no stdout/stderr for the GUI version, the test of the extension failed, leading to the error I experienced,

closed 2021-09-29 by hurricanefrog

haneefdm, 2021-09-29:
Haha, I didn't even know the GUI version worked. I should have noticed that when you posted this
```
    "cortex-debug.JLinkGDBServerPath": "C:/Program Files (x86)/SEGGER/JLink/JLinkGDBServer.exe",
```