Marus/cortex-debug #499: [GDB] : trouble to break, link with c files

lcharpenwi6labs, 2021-10-05:
Hello.
I work on stm32 with stlink configuration:

"configurations": [
    {
      "cwd": "${workspaceRoot}",
      "executable": "test_module/module.elf",
      "name": "Debug Microcontroller",
      "request": "launch",
      "type": "cortex-debug",
      "showDevDebugOutput": true,
      "servertype": "stlink",
      "svdFile": "C:/stm32-svd/svd/stm32l100.svd",
      "runToMain": true,
      "v1": false,
      "showDevDebugOutput": true
    }
  ]
As you can see I work under windows.
The project is compiled under ubuntu wsl 20.04 with arm-none_eabi-gcc.

Under the debugger, I can correctly see the program in asm but I don't have any access to the C.
I have access to registers, call stack etc...

When I set a breakpoint pointing on my c source file, the following command is sent to gdb:

27-break-insert "c:\\test_module\\module.c:91"
GDB -> App: {"token":27,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","No source file named c:\\test_module\\module.c."]]}}

But if I set the breakpoint inside gdb like this it works:
      break module.c:91

Is there any option in the project to use at least relative path but not complete path?
I suppose this problem is also linked to the fact I am not able to see the c code but the code in asm in the debug window.

Thanks for your help

haneefdm, 2021-10-05:
We don't manage breakpoints or even interpret them. This is done by VSCode and interpreted by gdb. If your VSCode project folder does not match how your program is compiled with relative and absolute paths, things may not work. These pathnames are buried in your elf file, btw.

This `"c:\test_module\module.c:91"` comes from VSCode.

You can use gdb's source path features to change the paths-names but things like Globals and Statics also may not work well in the Variables window.

https://sourceware.org/gdb/onlinedocs/gdb/Source-Path.html

And, sorry, no there are no facilities to interpret or re-interpret pathnames by our extension or by VSCode. It is all gcc/gdb and your make/build recipes. You can also change the `"cwd"` which will affect how gdb looks at things and that may work

lcharpenwi6labs, 2021-10-06:
Thanks a lot for your support!

lcharpenwi6labs, 2021-10-06:
I tried to compile under git bash with the windows version of arm-none-eabi and it works now. As you said, the path build using wsl during the compilation are not aligned with the vscode..

closed 2021-10-06 by haneefdm