Marus/cortex-debug #502: Need to kill st-util manually after every debug session in order to start a new session

anthroid, 2021-10-08:
I'm not sure if this is related to #494 or not, but it sounds related at least, and possibly not a coincidence based on reading through that thread. This also only started occurring recently (`0.4.6` - `0.4.7`), thought I can't be sure, but I can confirm that reverting to `0.4.1` seems to work more reliably.

When starting the first debug session, everything seems to work normally. I immediately see the registers loaded in the sidebar and output in the debug console when pressing `Start`. After debugging, I press `Stop` in the toolbar. Then, when I go to start another debug session (after a code change + recompile for instance), the debugger hangs for a timeout period, and then eventually pops up an error window: `Failed to launch ST-Util GDB Server: Timeout.`

I can try restarting the session a few times, but it will not recover until I open a terminal and manually `killall st-util`. Then immediately switch back, hit `Start`, and it takes off normally. If I revert `cortex-debug` to `0.4.1`, I can follow these steps and repeatedly start and stop sessions without any hangs and without having to `killall st-util`. I can also close out of VSCode completely, open everything up in a terminal (`st-util`, `arm-none-eabi-gdb`), and these all start/stop reliably as expected. It seems that the extension is either not properly stopping or detaching, or not detecting the already-running instance of `st-util`.

In `0.4.7`, I get this output in the debug console when it tries to start a session and fails:
```
Please check TERMINAL tab (gdb-server) for output from /opt/homebrew/bin/st-util
Launching server: "/opt/homebrew/bin/st-util" "-p" "50000" "-v" "--no-reset"
```

And this in the `gdb-server` terminal tab:
```
SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected
libusb: warning [darwin_open] USBDeviceOpen: another process has device opened for exclusive access
libusb: warning [darwin_open] USBDeviceOpen: another process has device opened for exclusive access
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
st-util
2021-10-07T21:32:06 WARN usb.c: Stlink usb device found, but unable to claim (probably already in use?)
2021-10-07T21:32:06 DEBUG common.c: *** stlink_close ***
SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

Checking the output of `ps ax`, there is always an `st-util` process running, launched from the extension:
```
36307   ??  S      0:00.59 /opt/homebrew/bin/st-util -p 50000 -v --no-reset
```

Reverting to `0.4.1` and following the same steps, I get this (and it works):
```
Please check OUTPUT tab (Adapter Output) for output from /opt/homebrew/bin/st-util
Launching server: "/opt/homebrew/bin/st-util" "-p" "50000" "-v" "--no-reset"
Launching GDB: "/opt/arm/bin/arm-none-eabi-gdb" "-q" "--interpreter=mi2"
Reading symbols from (...)
```

And in the `gdb-server` terminal:
```
2021-10-07T21:49:44 DEBUG common.c: *** stlink_read_debug32 0x00010000 at 0x52002010
2021-10-07T21:49:44 DEBUG common.c: *** stlink_write_mem32 64 bytes to 0x801ab00
2021-10-07T21:49:44 DEBUG common.c: *** stlink_read_debug32 0x00010000 at 0x52002010
2021-10-07T21:49:44 DEBUG common.c: *** stlink_write_mem32 64 bytes to 0x801ab40
... (breakpoints, etc) ...
2021-10-07T21:51:41 DEBUG common.c:   core status: running
Receive signal 15. Exiting...
2021-10-07T21:51:41 DEBUG common.c: *** stlink_run ***
2021-10-07T21:51:41 DEBUG common.c: *** stlink_read_reg
2021-10-07T21:51:41 DEBUG common.c:  (16) ***
2021-10-07T21:51:41 DEBUG usb.c: r_idx (16) = 0xe000edf0
2021-10-07T21:51:41 INFO common.c: Go to Thumb mode
2021-10-07T21:51:41 DEBUG common.c: *** stlink_write_reg
2021-10-07T21:51:41 DEBUG common.c: *** stlink_exit_debug_mode ***
2021-10-07T21:51:41 DEBUG common.c: *** stlink_write_debug32 0xa05f0000 to 0xe000edf0
2021-10-07T21:51:41 DEBUG common.c: *** stlink_close ***
SERVER CONSOLE DEBUG: onBackendConnect: gdb-server program closed
GDB server exited. Waiting for next server to start...
```

System info:
- Version: 1.60.2 (Universal)
- Commit: 7f6ab5485bbc008386c4386d08766667e155244e
- Date: 2021-09-22T11:59:27.195Z
- Electron: 13.1.8
- Chrome: 91.0.4472.164
- Node.js: 14.16.0
- V8: 9.1.269.39-electron.0
- OS: Darwin arm64 20.6.0
- Extension versions: 0.4.1 - 0.4.7
- `st-util` version: 1.7.0 (homebrew)
- `arm-none-eabi-gdb` version: 9-2020-q2-update

anthroid, 2021-10-08:
> I assume it is brew install stlink to install?

Yes, this is correct. On an M1 Mac, you should get an `st-util` binary in `/opt/homebrew/bin`.

> And no, this is unrelated to #494 which btw is for stlink and not st-util and problem was with not even starting (initially)

Since there was a problem parsing the output to determine whether there was a connection, I thought it could be a similar issue if that area of the extension was changed recently, given that this might have been related to the extension identifying that there was an `st-util` instance running. This was just a speculation though.

My `launch.json`:
```json
{
	"version": "0.2.0",
	"configurations": [
		{
			"name": "Debug STM32H743",
			"cwd": "${workspaceFolder}",
			"svdFile": "./STM32H743.svd",
			"request": "launch",
			"type": "cortex-debug",
			"executable": "./build/${workspaceFolderBasename}.elf",
			"armToolchainPath": "/opt/arm/bin",
			"servertype": "stutil",
			"interface": "swd",
			"device": "STM32H743x",
			"runToEntryPoint": "main",
			"overrideLaunchCommands": [
				"load ./build/${workspaceFolderBasename}.elf"
			]
		}
	]
}
```

Note: I have tried removing the `overrideLaunchCommands` section, but it doesn't seem to make any difference.

haneefdm, 2021-10-08:
Could you try this patch?

https://github.com/Marus/cortex-debug/releases/tag/V0.4.8-pre1

We are sorry about this. Every server behaves differently and we were trying to be nice because sometimes we would kill the server while it was in the middle of a USB transaction. If the servers behaved like they are supposed to, then we would not have to kill them in the first place. Hopefully, we would have weeded out most of these issues.

There is a file `/tmp/cortex-debug-server-exiting.log` that is being created to see what is going on on your system. If things are not working, this may help.

haneefdm, 2021-10-08:
@anthroid Do you mind trying a new patch release? It is a very minor change/cleanup and it works for me

haneefdm, 2021-10-09:
May I ask what drew you using st-util rather than two other options (OpenOCD or STLink)

That much output can also be a huge performance problem. But you have an M1 so you probably didn't notice

anthroid, 2021-10-09:
> May I ask what drew you using st-util rather than two other options (OpenOCD or STLink)

I don't actually remember. I think when I started working with STM32 boards, `st-util` seemed the most appealing and was probably the first option that I got working, so that's what I've been using since. It works reliably for my purposes with minimal fuss, so I haven't ever been too compelled to look into any of the other options.

PhilippHaefele, 2022-03-28:
@haneefdm Good to close as the fix has been released a long time ago

closed 2022-03-29 by haneefdm