Marus/cortex-debug #505: Help with applying the 'disassembly function'

VzdornovNA88, 2021-10-11:
Hi.

Unfortunately, I can't apply the "Cortex-Debug: Set Force Disassembly" command. I expect a new window in working space of vs code but the extension doesn't show the window with disassembled information but the debugger step on the next instruction in disassembled code when I press the button F10,I think so because the cursor of debugger in my source stays in the place and int output of gdb new address of instruction is shown. So , I have checked the gdb output with 'disassembly' command and the output is right but the extension doesn't show the window with disassembled information.

Also, I have tried to apply "Cortex-Debug: View Disassembly (Function)" command but got error - "No function matching name/regexp 'main' found." 

Short information about of context :

- Windows 10

- version of vs code :
Версия: 1.61.0 (user setup)
Фиксация: ee8c7def80afc00dd6e593ef12f37756d8f504ea
Дата: 2021-10-07T18:13:09.652Z
Electron: 13.5.1
Chrome: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
ОС: Windows_NT x64 10.0.16299

- version of cortex debug 
v0.4.7

- version of open OCD
>= 0.10.0-12-20190422-2015

- version of arm tools
gcc-arm-none-eabi 10.2.1

launch file contains : 

"configurations": [
        {
            "name": "KUT-01 Debug",
            "type": "cortex-debug",
            "request": "launch",
            "runToEntryPoint": "main",
            "showDevDebugOutput": false,
            "servertype": "openocd",

            // "openOCDPreConfigLaunchCommands": ["arm disassemble 0x0 256"],

            "cwd": "${workspaceRoot}",
            "interface": "swd",
            "device": "STM32G473VE",

            "executable": "${workspaceRoot}/build/XXX.elf",

            "svdFile"     :   "${env:CMAKE_NONE_EABI_STM32_TOOL_CHAIN}/cmsis-svd/data/STMicro/STM32G473xx.svd",
            "gdbPath"     :   "${env:CMAKE_NONE_EABI_STM32_TOOL_CHAIN}/eabi/bin/arm-none-eabi-gdb.exe",
            "configFiles" : [ "${workspaceRoot}/openocd.cfg", ],
        },

openocd.cfg file contains : 

source [find interface/stlink.cfg]
transport select "hla_swd"
source [find target/stm32g4x.cfg]

Highly likely, I do something wrong but, unfortunately, I don't have much time to figuring out the right combinations of settings of chain 'vs code -> cortex debug -> open OCD -> gdb'. 

Any help is welcome from more competent people in setting up this chain, thanks

VzdornovNA88, 2021-10-12:
Hi, again, like I said above , I have checked the gdb output with 'disassembly' command and the output is right(it means that gdb output contains the disassembled code), just You should read again my first message. And I have used '-g' options of compiler otherwise I wouldn't get disassembled info from gdb, once again, gdb works fine ! like I said in my first message but cortex debug doesn't show the same information, so what's next ?

VzdornovNA88, 2021-10-12:
In addition, using the "Cortex-Debug: View Disassembly (Function)" command doesn't give me any outgoing information in the debug console but VS Code shows an error, like I said in my first message of the topic - "No function matching name/regexp 'main' found."

VzdornovNA88, 2021-10-12:
Exactly ! PhilippHaefele is right, I use his official source. 
So , now I can provide you with the following info :

- https://github.com/VzdornovNA88/signaler_example_freertos.git - my repo with the similar project 
- https://github.com/VzdornovNA88/vs_code_docker_cmake_stm32_gcc.git - my repo sandbox with vs code in the docker 

I can explain why I sugest them , if you don't want to wait tommorow when I can send the info of settings for cortex-debug , You can run the sandbox and clone 'signaler_example_freertos' into it then you will get the similar state as mine, and then you  can run debug, but also you can download and do all the same in your envirement manualy:

https://developer.arm.com/-/media/Files/downloads/gnu-rm/10-2020q4/gcc-arm-none-eabi-10-2020-q4-major-x86_64-linux.tar.bz2

Anyway , I am gonna send the addition info for you tommorow, so that we can just wait tommorow , thanks for your help.

haneefdm, 2021-10-13:
Also, we don't use objdump to do the disassembly. We use it only to get symbols table.
```
arm-none-eabi-objdump --syms -Wi -C XXX-01.elf
```
Maybe you can attach the output of that on your real executable unless that also is under NDA.

VzdornovNA88, 2022-08-06:
Thanks ,I am gonna try it out later maybe in month.