Marus/cortex-debug #507: local var with same name not update when step into

rqbh, 2021-10-12:
local var with same name not update when step into called function.
local var always show the first catched data when first stop at breakpoint.

platform: ubuntu 21.10
vscode : 1.61.0

haneefdm, 2021-10-12:
That is not good. I see hover showing one value and Locals in Variable window showing another. VSCode caches the variables window contents based on a frame-by-frame basis. If you select a different frame in the stack window, do the variables change and/or match what is shown on hover?

In the `Debug Console` can you enter `info globals` and let me know what you see?

Note: Typing any command in the Debug Console updates **all** windows and state. I am interested in what gdb says.

haneefdm, 2021-10-13:
https://visualgdb.com/gdbreference/commands/info_locals#:~:text=The%20info%20locals%20command%20displays,command%20to%20list%20function%20arguments.

rqbh, 2021-10-14:
[image]

In above picture, you can see 'info locals' gdb cmd result with <widget> var's address is not equal with the <widget> var's address in variables window. 
This must be something wrong,

haneefdm, 2022-03-30:
Some of the ways the globals/locals are shown have changed in the extension and this issue may not be relevant anymore. I am going to close this issue. If the issue persists, then we should reopen or open a new issue.

closed 2022-03-30 by haneefdm