Marus/cortex-debug #511: Cortex-debug with dev-container. How to connect to JLinkRemoteServer outside docker?

andyinno, 2021-10-18:
Upon @haneefdm comment I am creating a new issue for this question.

I am trying to connect to my JLink adapter but vscode stops the debug as soon as I connect.

I am launching JLinkRemoteServerExe on host computer

I am using this configuration:

```json
  {
            "type": "cortex-debug",
            "name": "Remote",
            "servertype": "jlink",
            "ipAddress": "192.168.1.87",
            "device": "ATSAM3A4C",
            "serverpath": "JLinkGDBServerCLExe",
            "request": "launch",
            "executable": "${workspaceRoot}/build/test_executable.elf",
            "runToMain": true,
            "svdFile": "${workspaceRoot}/.vscode/ATSAM3A4C.svd",
            "serverArgs": [
                "-rtos",
                "GDBServer/RTOSPlugin_FreeRTOS"
            ],
            "swoConfig": {
                "enabled": true,
                "cpuFrequency": 84000000,
                "swoFrequency": 7500000,
                "source": "probe",
                "decoders": [
                    {
                        "type": "console",
                        "label": "ITM0",
                        "port": 0,
                        "encoding": "ascii"
                    }
                ]
            }
        },
```

What I get on the vscode terminal is:

```bash
Connecting to J-Link...
J-Link is connected.
...
Checking target voltage...
Target voltage: 3.28 V
```
Therefore docker connects to host and host query the JLink device... but after some line (less than a second):

```
Waiting for GDB connection...SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

My host is a ubuntu OS. I just want to have my own docker instance with all the tools already setup.

thank you in advance

If I try to connect from outside the docker instance (reopen folder on host) I can connect to the board with this setup. 

As a sidenote:

@haneefdm you are right that I was polluting a different topic with this question, however your comment 

> You are doing something this tool was not designed for 

Is something unexpected to me. Can you elaborate on this? I think that:

- Windows -> WSL -> Toolchain in Docker 

is really similar to:

- Any Linux distro -> Docker -> Toolchain in Docker

We both want to be able to use standardized tools on different hosts and being able to code and debug our software

haneefdm, 2021-10-18:
The whole process of how a gdb-server is launched, the ip-addresses involved, TCP port allocation never envisioned the gdb-server to be on another computer/environment other than the host itself.

This does include 'external' type servers where you control/launch the servers external to VSCode/Cortex-Debug. There you are fine doing whatever but may have limited functionality as such.

When we launch the server, however, we never expected the server to be running anywhere other than locally.

In your case, you may be having the local jlink server having trouble talking to a remove jlink server which is again an entirely different scenario. It may not even have anything to do with VSCode/Cortex-Debug either.

I would not have started a new task/project to do remote debugging (WSL/Docker/SSH) if it was already doing all of that.

TorBlackbeard, 2021-10-19:
You *could* also :
1)
try to start a real "JLink GDB Server" on the windows side and then use another vscode module, so it's not   "servertype": "jlink" but probably  "external".
I didn't try this. And if IP/firewalling is root cause, you are probably in for the same problems with that approach.

2)
Try to use a RaspPI, and see if you can target that.
That goes for both JLinkRemoteServer and JLinkGDBServer.
(Not 100% sure what exist on the Arm build of JLink tools.)

andyinno, 2021-10-19:
Thank you both @TorBlackbeard and @haneefdm 

I already hit this issue before but completely forgot the solution.

Basically my configuration was really really similar to the one @TorBlackbeard posted.

What I was noticing was:

- green line in the external JLinkRemote
- internal Jlink not starting

Running the commands from the command line inside the container by hand the error message was really clear.

libncurses5 not installed in the docker environment.

Rebuilding the docker container with libncurses5 installed solved the issue.

Just for reference, it was already addressed here #266 . 

@haneefdm: I said I would NOT have started a new task/project #467

Yes, I read all the comments there. I published my first post there for this reason: It seems to me that using JLink Windows/Linux or Linux/Linux is mostly the same at the moment. The reason, in my opinion, why devcontainers comes to hand is when you do not know which computer your next developer will work on, therefore provide a base container where the different tools are installed is really a plus.
I do not know the WSL2 subsystem would provide any difference but if possible to keep in mind other uses cases for devcontainers other than the expressed in the issue #467 would be really appreciated.

Thank you to both again
Andrea

mateusz-, 2021-12-29:
With WSL2 (Ubuntu 20.04) I also had to install `libncurses` (`sudo apt install libncurses5`) to get debugging to work.

haneefdm, 2021-12-29:
@mateusz- It is not just you. Everyone has to do this. For some reason, this does not install automatically when you install gdb on any kind of Linux that I know of. WSL2 or not.