Marus/cortex-debug #520: The cortex-debug extension used to work great for me.  I am now on the latest MacOs and am not able to get it working any more.  I get a message that says:

cbj0, 2021-11-03:
The cortex-debug extension used to work great for me.  I am now on the latest MacOs and am not able to get it working any more.  I get a message that says:

Failed to launch undefined GDB Server: Timeout.
my launch.json is
```json
{
    "configurations": [
    {
        "cwd": "${workspaceRoot}",
        "executable": "${command:cmake.launchTargetPath}",
        "name": "jlinkdebug",
        "request": "launch",
        "type": "cortex-debug",
        "showDevDebugOutput": false,
        "servertype": "jlink",
        "device": "RP2040_M0_0",
        "interface": "swd",
        "configFiles": ["rp2040.cfg"],
        "runToMain": true,
        "gdbPath": "d:/SysGCC/arm-eabi/bin",
        "svdFile": "${env:PICO_SDK_PATH}/src/rp2040/hardware_regs/rp2040.svd",
        "postRestartCommands": ["load","b main","c"],
    }
    ]
}
```
I know the target and cable are all good.  When I use Segger Ozone it all works correctly.

_Originally posted by @chetferry in https://github.com/Marus/cortex-debug/issues/81#issuecomment-415639602_

haneefdm, 2021-11-03:
We haven't changed anything on Cortex-Debug! You can see the command-line for the JLink gdb-server in the `Debug Console`. Have you tried running that from the command line? See if it starts? You should see messages that you normally saw on stdout/stderr.

Not sure Ozone would use this gdb-server since it can talk directly to the probe.

I haven't jumped to Monterey and am not about to for several days/weeks.

haneefdm, 2021-11-11:
No response, trying other things, so closing it

closed 2021-11-11 by haneefdm