Marus/cortex-debug #521: Does the 'command' command work ?

efox29, 2021-11-04:
I am trying to record some data via attaching a set of commands to a breakpoint. However when I try and do this from within the debug console of cortex-debug - the entire session is stuck with it asking for input. 

I have tried this without cortex-debug and ran gdb straight from commandline, and I attaching command to breakpoints works fine.

Am I missing something ?

haneefdm, 2021-11-04:
Do you have something to show me? I don't know what you are trying to do (gdb or python scripts or something else). and who is asking for input?

haneefdm, 2021-11-04:
And, what is a `command` command?

haneefdm, 2021-11-04:
The circled numbers are used by VSCode to indicate there are duplicate lines with identical text

efox29, 2021-11-04:
Edited previous post because I shared the same image twice.

haneefdm, 2021-11-04:
The debug console is not a terminal so things like Control-C don't mean a thing. Control-C may actually do a Copy on some platforms.

Every time you press enter in the debug console, VSCode assumes you finished entering the command and tries to refresh all the windows related to the debugger.

During the refresh, some buttons may be disabled and it will/may appear like we hit the breakpoint again and an infinite loop may start. Some of this is a guess why things stop responding.

This is happening while you are entering creating the `command` callback? You can put the sequence in a text file and source that file.

Please note that this is not an interactive shell and you are not interacting with gdb's stdin/stdout which is already consumed by another process (Cortex-Debug backend). You are actually talking to a window created by VSCode and they set the rules.

efox29, 2021-11-04:
I added the command to my gdbinit file and it works there. 

Is there a way to interact with gdb while still having the graphical frontend ?

haneefdm, 2021-11-04:
In a very limited way, yes. one-line commands. Commands that modify the status of the program or mess with the stack, etc. can cause things to go out of sync -- there are 5 processes (at least) that are involved. We try to recover from a lot of things but obviously not everything.

I am gonna close this.

closed 2021-11-04 by haneefdm

efox29, 2021-11-04:
I would totally want this feature to be available within cortex-debug. This is one of the most powerful capabilities within gdb along with the python interpreter for scripting and loading gdbinit files and takes basic level debugging to another more advanced level. I would urge the developer(s) to consider adding the ability to use this feature.

Just a note: while sourcing the command file within gdb does "work", it has the same problems as before, where you can't interact with the program anymore and you need to terminate the session.

haneefdm, 2021-11-04:
We don't control that window at all. It is VSCode. VSCode gives it to us, but also initiates a large set of actions by itself. We just pass on whatever was given to gdb's stdin but its stdin is note the same as what you see on gdb-prompt. We have to use machine-interface where there are no multi-line commands. Neither us nor VSCode knows it was a partial command.

It would help me to know what was in the set of commands that you were trying to enter, even to judge/scope anything. Btw, I am the sole `developer(s)`.

We have two problems.
* Inability to enter multi-line commands -- VSCode issue
* Whatever you entered, made the whole system go crazy -- could be anything

efox29, 2021-11-04:
I understand that there are limitations within that window. I think it would be a great feature to add (maybe not directly with that window ). This is not my area of expertise, but a few possibilities (no idea difficulty or feasibly) could be to use another window that is not the debug console or to provide another process that can be started that can interact with gdb and feed data back to its stdout.

> It would help me to know what was in the set of commands that you were trying to enter, even to judge/scope anything. Btw, I am the sole developer(s).

I created a breakpoint via debug console and got the breakpoint number (let's say for this, it's 2), and I want to print out the contents of an in-scope variable called foo.

Then I typed out the following. Each new line is me pressing ENTER.
```
command 2 
p foo
c
end
```

As soon as I hit ENTER at the first line - the system breaks here. After that, the rest of the stuff is irrelevant.

haneefdm, 2021-11-04:
Ooops, you did a 'continue'. That modifies the program/debugger state without them knowing about it. It is probably why things got messy after that. And, that continue happens in the middle of everyone trying to updated themselves because of the breakpoint. All bets are off.

haneefdm, 2021-11-04:
Those commands cannot be entered in the Debug Console. No multi-line commands. Gdb itself is probably confused. You can enable `showDevDebugOutput` to see what was the last few things that happened.

zihlmalb, 2025-02-14:
Any news on this? This issue shall be solved.

efox29, 2025-02-14:
Since no multi-line commands allowed in the terminal, I was able to work around this by creating a text file and sourcing it within gdb.

Create an text file with your multi-line commands
```
command 3
p myvar
c
end
```

And within the gdb command line, use `source your_text_file`.

zihlmalb, 2025-02-14:
Ok, thanks for the tip!