Marus/cortex-debug #522: qemu debugging broken?

perlindgren, 2021-11-05:
Hi folks

Seems there is something up with Cortex-Debug and the latest vscode under arch linux with latest packages.

```shell
>code --version
1.62.0
b3318bc0524af3d74034b8bb8a64df0ccf35549a
x64
```
I tried with both the `arm-none-eabi-gdb` 
```shell
> arm-none-eabi-gdb --version
GNU gdb (GDB) 11.1
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```
and installed `gdb-multiarch` from aur.
```
> gdb-multiarch  --version
Exception caught while booting Guile.
Error in function "make_objcode_from_file":
bad header on object file: "\x7fELF\x02\x01\x01ÿ\x00\x00\x00\x00\x00\x00\x00\x00"
gdb-multiarch: warning: Could not complete Guile gdb module initialization from:
/usr/share/gdb/guile/gdb/boot.scm.
Limited Guile support is available.
Suggest passing --data-directory=/path/to/gdb/data-directory.
GNU gdb (GDB) 11.1
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```
(Seems the `gdb-multiarch` is a bit broken but the behavior in vscode is the same.

Qemu is the latest:
```
qemu-system-arm --version
QEMU emulator version 6.1.0
Copyright (c) 2003-2021 Fabrice Bellard and the QEMU Project developers
```

The `Terminal` shows:
```
Resuming connection to gdb server...
SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected
Timer with period zero, disabling
SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

The  `Console` shows:
```
Please check TERMINAL tab (gdb-server) for output from qemu-system-arm

Launching server: "qemu-system-arm" "-cpu" "cortex-m3" "-machine" "lm3s6965evb" "-nographic" "-semihosting-config" "enable=on,target=native" "-gdb" "tcp::50001" "-S" "-kernel" "/home/pln/rust/d7020e/cortex-rt-app/target/thumbv7m-none-eabi/debug/cortex-rt-app"
```
The problem was replicated on two independent systems, (vanilla arch and under Manjaro arch).

The `launch.json` is as follows:
```
          {
            "type": "cortex-debug",
            "request": "launch",
            "name": "Debug (QEMU)",
            "servertype": "qemu",
            "cwd": "${workspaceRoot}",
            "preLaunchTask": "Cargo Build (debug)",
            "runToMain": true,
            "executable": "./target/thumbv7m-none-eabi/debug/cortex-rt-app",
            "cpu": "cortex-m3",
            "machine": "lm3s6965evb",
        },
```
The `preLaunchTask` is run as expected. And if running `gdb` against `qemu` in terminal the program behaves as expected. So the tooling outside of `Cortex-Debug` and `vscode` seems to be ok. So I wonder if something changed with the behavior of `vscode` (I haven't tried downgrading `vscode`, and I would rather not if a workaround/solution is possible for the latest `vscode`.

Let me know if additional info is of any help.

Thanks in advance.
/ Per

haneefdm, 2021-11-05:
I am not familiar with QEMU. Do you recognize the following?

> Timer with period zero, disabling

Could you enable `showDevDebugOutput` and look in the `Debug Console`?

If we can't figure out using the output above, I will need help to try and reproduce the issue.

Nothing changed for QEMU specifically. A lot of changes occurred on a session ends though. Very little on how a session starts.

perlindgren, 2021-11-05:
A small note to the above, in this case I think it had crashed already once, hence a fresh  `tcp` socket ` "tcp::50001" ` was allocated. `killall -9 qemu-system-arm` works to shut down hanging instances. However it does not solve the problem.

perlindgren, 2021-11-05:
Hold on I will try to get a log.

perlindgren, 2021-11-05:
```
Reading symbols from '/home/pln/rust/d7020e/cortex-rt-app/target/thumbv7m-none-eabi/debug/cortex-rt-app'
Finished reading symbols
Please check TERMINAL tab (gdb-server) for output from qemu-system-arm
Launching server: "qemu-system-arm" "-cpu" "cortex-m3" "-machine" "lm3s6965evb" "-nographic" "-semihosting-config" "enable=on,target=native" "-gdb" "tcp::50000" "-S" "-kernel" "/home/pln/rust/d7020e/cortex-rt-app/target/thumbv7m-none-eabi/debug/cortex-rt-app"```

perlindgren, 2021-11-05:
Its a Rust project by the way (not that it should matter I think...)

perlindgren, 2021-11-05:
Is there anything else I should/could try?

haneefdm, 2021-11-06:
I will take a look tomorrow morning. As an aside, people are putting stuff like

`killall -9 qemu-system-arm`

in their `preLaunchTask` or better yet, 'postDebugTask`. I don't recommend either but letting you know.

Can you start qemu with the exact command line you see in the Debug Console. Looks like it did on startup. GDB did not even launch which means the server ended waaay prematurely.

perlindgren, 2021-11-06:
Yes  it works from terminal. It will start and await a gdb connection.
If I start gdb from terminal:
 arm-none-eabi-gdb -f /home/pln/rust/d7020e/cortex-rt-app/target/thumbv7m-none-eabi/debug/cortex-rt-app
 and then 
 target remote localhost:50000
 
 Remote debugging using localhost:50000
cortex_m_rt::Reset () at /home/pln/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rt-0.6.15/src/lib.rs:497
▒▒/home/pln/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rt-0.6.15/src/lib.rs:497:19299:beg:0x400

 
So it looks all good in terminal.

perlindgren, 2021-11-06:
By the way, I tested it with hardware (stm32f411 Nucleo), and `openocd`. It seems to work as intended, so I guess its related specifically to `qemu`.

haneefdm, 2021-11-08:
I tried it on Ubuntu 20.x and it worked (kinda). Although the qemu crashes after startup. It would help if I have an executable that demos this problem. Even a small hello world. I am getting some random executables from the web and they are not working well but qemu itself starts (crashes later but that is a different problem)

I could not compile qemu under Ubuntu, hours later, gave up and found a binary ARM release

haneefdm, 2021-11-08:
Here is the info from the Terminal
```
SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected
qemu: fatal: Lockup: can't escalate 3 to HardFault (current priority -1)

R00=00000000 R01=00000000 R02=00000000 R03=00000000
R04=00000000 R05=00000000 R06=00000000 R07=00000000
R08=00000000 R09=00000000 R10=00000000 R11=00000000
R12=00000000 R13=ffffffe0 R14=fffffff9 R15=00000000
XPSR=40000003 -Z-- A handler
FPSCR: 00000000
SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

perlindgren, 2021-11-09:
I'm using the [cortex-m-quickstart](https://github.com/rust-embedded/cortex-m-quickstart) as a template for generating a minimal executable. From terminal the 'qemu' works well.

In the `.cargo/config.toml`, you can uncomment:
```toml
# runner = "qemu-system-arm -cpu cortex-m3 -machine lm3s6965evb -nographic -semihosting-config enable=on,target=native -kernel"
```
And use:
```shell
> cargo run --example hello
```
To run the 'hello' example. (That one uses semihosting for printing.)

As seen it uses the Arm Cortex M3 processor (lm3s6965evb).

haneefdm, 2021-11-09:
I could not get past the following in the quickstart -- the code failed to compile/install and workarounds do not work.
```
The cargo generate subcommand. Installation instructions.

```
This is why I was asking for an executable (any executable).

203Null, 2021-12-05:
[image]
This is what I have atm.

203Null, 2021-12-05:
A

> 

Running the same command in terminal: (But press c afterward due to -S)
[image]

perlindgren, 2021-12-05:
So there is something wrong in this sequence of commands, in the `cortex-m-rtic` repo we are doing integration/system test with qemu and that seems to work ok, and it works from terminal (see my post earlier), its just that it does not work in cortex-debug.

perlindgren, 2021-12-05:
I think these commands (to launch `qemu-system-arm`) must have worked before (as I have been using vscode/cortex-debug to run under qemu), but something has changed...

203Null, 2021-12-06:
This is how I set it up. I can't find too much info on cortex debug with debug so I can't tell if I did this correctly
[image]

stamerlan, 2021-12-09:
Hello,

I have the same issue. I can successfully debug when I start both qemu and gdb manually, but always get "SERVER CONSOLE DEBUG: GDBServerConsole: onBackendConnect: gdb-server program client error Error: read ECONNRESET" error in vscode.

I use the following command line to start QEMU (same as the plugin does):
`"C:/Program Files/qemu/qemu-system-arm.exe" -cpu cortex-m3 -machine lm3s811evb -nographic -semihosting-config enable=on,target=native -gdb tcp::50000 -S -kernel fw.axf`

On start QEMU prints "_Timer with period zero, disabling_". In vscode terminal I can see only a part of the warning (each time a different part):
`Timer with period zero, [2021-12-09T23:16:44.526Z] SERVER CONSOLE DEBUG: GDBServerConsole: onBackendConnect: gdb-server program client error Error: read ECONNRESET`
`Timer with period zero[2021-12-09T23:21:43.278Z] SERVER CONSOLE DEBUG: GDBServerConsole: onBackendConnect: gdb-server program client error Error: read ECONNRESET`
`Timer with period zero, disa[2021-12-09T23:21:53.287Z] SERVER CONSOLE DEBUG: GDBServerConsole: onBackendConnect: gdb-server program client error Error: read ECONNRESET`

launch.json:
```
{
	"version": "0.2.0",
	"configurations": [
		{
			"name": "QEMU: lm3s811evb",
			"request": "launch",
			"type": "cortex-debug",
			"cwd": "${workspaceFolder}",
			"servertype": "qemu",
			"executable": "bin/fw.axf",
			"machine": "lm3s811evb",
			"gdbPath": "C:/Program Files (x86)/GNU Arm Embedded Toolchain/10 2020-q4-major/bin/arm-none-eabi-gdb.exe",
			"serverpath": "C:/Program Files/qemu/qemu-system-arm.exe",
			"showDevDebugOutput": true
		}
	]
}
```

debug console output:
```
Reading symbols from '[PATH]\bin\fw.axf'
Finished reading symbols
Please check TERMINAL tab (gdb-server) for output from C:/Program Files/qemu/qemu-system-arm.exe
Launching server: "C:/Program Files/qemu/qemu-system-arm.exe" "-cpu" "cortex-m3" "-machine" "lm3s811evb" "-nographic" "-semihosting-config" "enable=on,target=native" "-gdb" "tcp::50000" "-S" "-kernel" "[PATH]\bin\fw.axf"
```

terminal output:
```
[2021-12-09T23:21:52.832Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
Timer with period zero, disa[2021-12-09T23:21:53.287Z] SERVER CONSOLE DEBUG: GDBServerConsole: onBackendConnect: gdb-server program client error Error: read ECONNRESET
[2021-12-09T23:21:53.292Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

haneefdm, 2021-12-10:
I was finally able to compile a project using Rust on Windows (was trying on Linux with zero luck). This issue had nothing to do with Rust. It was our extension that was crashing. The problem should be fixed now and available as a pre-release. You may be the first to try the pre-release mechanism.

We are moving to a new versioning system that supports pre-releases. You have to update to the latest V1.63.0+ of VSCode. In the extensions area, you should now see a green dot
[image]
You will also see the following for install options.
[image]

haneefdm, 2021-12-10:
I am closing this issue, please reopen if this still an issue

closed 2021-12-10 by haneefdm