Marus/cortex-debug #524: Unexpected behavior with the "conditional expression breakpoints" in the Cortex-Debug extension

klazklaz77, 2021-11-10:
Hi, everyone, when I tested the application in the VS Code IDE I find unexpected behavior with conditional expression breakpoints in the Cortex-Debug VS Code Extension.
Steps to reproduce:
1. Set conditional expression breakpoint with some mistake in the expression (ex. ">2$");
2. Remove or disable the breakpoint.
3. Perform "Continue" a few times;

Expected result:
- Debugger doesn't stop in code at all;
Actual result:
- Debugger stops on the breakpoint;

OS: Windows 10;

Versions:
- VSCode 1.62.0;
- C/C++ 1.7.1;
- Cortex-Debug VS Code Extension 0.4.7;

Attachments:
- Step 1

[image]
- Step 2

[image]
- Step 3

[image]

Regards, Vitalii Loik.

haneefdm, 2021-11-10:
Thanks for reporting this. Able to reproduce this and fix on the way

haneefdm, 2021-11-14:
This will be fixed in the next release. A pre-release is available here 

https://github.com/Marus/cortex-debug/releases/tag/v0.4.8-pre5

Re-open if needed

closed 2021-11-14 by haneefdm