Marus/cortex-debug #525: Unexpected behavior with breakpoints in the Cortex-Debug extension

klazklaz77, 2021-11-10:
Hello, everyone!
I would like to report about unexpected behavior with breakpoints in the Cortex-Debug extension for VS Code IDE.

Step to reproduce:

1.  Click launch the application and wait before the board will be launched.
[image]

2.  Set a breakpoint on the non-command line in the program code (for example lines 158-160 on screenshot 2).
[image]

3. Repeat step 2 in the same line, breakpoints will be put on the next program command line, and in the "breakpoints panel" will be 2 breakpoints per 1 line.
[image]

4. Click on the breakpoint one time. Breakpoints will disappear from the "breakpoints panel" and code editor.
[image]

5. Perform "Continue" a few times.
[image]

Expected result:

- Debugger doesn't stop in code at all;

Actual result:

- Debugger stops on the (invisible) breakpoint;

For example, I attached the video:

https://user-images.githubusercontent.com/54813908/141118713-64485fd7-03a1-40d9-8e27-e6ccb0b163cd.mp4

Regards, Vitalii Loik.

haneefdm, 2021-11-10:
Thanks for reporting this.  The video was helpful and I will try to reproduce this.

Did you notice anything in the `Breakpoints` window? Things appearing, disappearing, greying out, etc? There is also stuff displayed on hover.

Basically, gdb created a breakpoint on line 164 when you asked for line 159 where there is no code after gdb compiled your C file. If gdb cannot find code at the specific line you ask for, it sets the breakpoint at the next available line. But then you set a breakpoint at that same line and VSCode notifies us and we notify gdb and I cannot tell what happened after that. It was a duplicate breakpoint but not sure how it got handled.

You can enable `"showDevDebugOutput": true` in your launch.json and see what is happening for each action you do. In the Debug Console, you can also enter regular gdb commands. After the entire video, if you do a `info breakpoints`, I bet you saw a breakpoint -- although neither VSCode or Cortex-Debug know about it.

klazklaz77, 2021-11-11:
Hi, after I set the breakpoint at the line (ex. 159) first time in the breakpoints panel I see one active breakpoint at the next available line (164). It is expected. 

[image]

If I set the breakpoint at the same line one more time, I see in the breakpoints panel one more active breakpoint with the same number.

[image]

If I want to remove the breakpoints I see the two line breakpoints in the menu:

[image]

If I do this, breakpoints will be successfully removed from the pannel and code:

[image]

And debugger doesn't stop in the code at all.

The same situation will be If I remove the breakpoints from the breakpoints panel (remove all breakpoints or one by one):

[image]

But If I click on the breakpoints in the code editor - they will disappear from the code editor and breakpoints panel, but the debugger stops at this code line:

[image]

[image]

One more situation: 
I can set the two breakpoints per one line (steps 1-3 in the bug report) and set one more breakpoint in the line where is 
program code (ex.167).

[image]

In this case, a number of the second breakpoint will be changed to the last used breakpoint number  (ex. 167):

[image]

I hope that this info can help you and Cortex-Debug team!

haneefdm, 2021-11-14:
This was actually a bug in VSCode in breakpoint processing and duplicates. Complicated to explain but VSCode was calling the adapter to adjust breakpoints before the previous request for the same was even finished. We have to create a workaround.

This will be fixed in the next release. A pre-release is available here 

https://github.com/Marus/cortex-debug/releases/tag/v0.4.8-pre5

Re-open if needed

closed 2021-11-14 by haneefdm