Marus/cortex-debug #527: Using hardware breakpoints from GUI

Uknius, 2021-11-11:
Recently I started using an embedded platform that doesn't support software breakpoints.

As I understand I can set only software breakpoints from the vscode GUI. 
Is there any possibility to force GUI to use hardware breakpoints?

haneefdm, 2021-11-11:
You can set whatever breakpoints you want. Neither VSCode nor this extension knows how to handle them. They will be handled by gdb which in turn gives it to OpenOCD. OpenOCD only supports HW bkpts AFAIK and that is what you are getting. We have no capability for force SW or HW breakpoints. It is up to the gdb and OpenOCD -- and depends on where the address is FLASH or SRAM.

See the Debug Console for info on what you see when you set a bkpt.

Note that we are a front-end to GDB. Anything we do translates into something for gdb.

Uknius, 2021-11-11:
Debug console if I place a breakpoint before starting to debug:
[image]
[image]

It looks like it correctly places the breakpoint but it won't break.

The only way I got it working is by setting `hbreak main.c:35` by hand in the debug console.
[image]

Uknius, 2021-11-11:
It look like to this: https://github.com/microsoft/vscode-cpptools/issues/4745

closed 2021-11-11 by Uknius

haneefdm, 2021-11-11:
Note: MS Cpptools and this extension has almost nothing in common. We are both clients of VSCode infrastructure, use gdb and obey the rules set forth by VSCode, so we may look similar. Under the hood and features wise we are very different, focusing only on Embedded stuff.