Marus/cortex-debug #528: serverArgs parameter passes with quotation marks

Uknius, 2021-11-11:
It looks like serverArgs passes arguments to the gdb-server command line with quotation marks. It leads to an unrecognized command:

[image]

Uknius, 2021-11-11:
For my scenario, you can replace 
serverArgs: ["--command..."] with: "openOCDPreConfigLaunchCommands" and it works, but serverArgs still not working as expected.

haneefdm, 2021-11-11:
`serverArgs` are command line args -- unix style. So, `"--command"` and `"set DEVICE TEST"` are separate args. You can see that in the Debug Console.
```JSON
    "serverArgs": [ "--command", "set DEVICE TEST"],
```
`openOCDPreConfigLaunchCommands` is an array of commands handled by us (not bash or cmd.exe) and we automatically prepend a `-c` to each element (as separate unix-style args. Again, you can see how things are translated in the DebugConsole

closed 2021-11-11 by haneefdm