Marus/cortex-debug #529: Feature request: chained launch configurations

AndriiLishchynskyi, 2021-11-14:
When user makes a setup for multi-core debugging, he wants to have an ability to launch multi-core debug with a single button click. Similar to this functionality offers [Compound launch configurations](https://code.visualstudio.com/docs/editor/debugging#_compound-launch-configurations). However, compound launch configs have some disadvantages, i.e. all configurations are started in parallel, which bring-in some complexity. Multi-core debugging implies few debugging processes to be launched. For non-shared GDB servers that means, you need at first start GDB server, then you are able to connect to it. To make sure GDB server is up and running, you need to add additional delay before connecting to it as described [here](https://stackoverflow.com/questions/42867765/compounds-debugging-in-vs-code-with-delayed-launch).

One of the ways to resolve this is to use chained launch/attach configurations. While compound configs are implemented in VSCode, chained configs implementation are up to the extension. Please consider adding this feature into Cortex-Debug

haneefdm, 2021-12-22:
I am working on `chained configurations`, some of the work has been pushed up to the master. Not ready for general use though. One issue I am having issues with is how to handle top-level features like Reset/Restart/Stop/Disconnect.

For 'Pause' I suggest using the ARM's CTI if you want to make all configurations to Pause. Doing it in SW would be too slow. I have to think about 'Resume/Continue' though. Lower Priority. 

Details for all of the above will appear here as they get implemented.

haneefdm, 2021-12-24:
While I leave this page for discussions, the feature will be documented here...

https://github.com/Marus/cortex-debug/wiki/Multi-core-debugging

haneefdm, 2021-12-31:
Many of the features for Multi-core debugging and other changes (see CHANGELOG.md) are now available as a pre-release. You can get it from the marketplace of download from here

https://github.com/Marus/cortex-debug/releases/tag/v1.1.2

More updates likely come before final release

AndriiLishchynskyi, 2022-01-04:
Hello Haneef,
Thanks a lot for your effort! I am playing around with multi-core debugging. Right now I have troubles with chaining configurations residing in different folders. Could you please suggest something?

As per documentation, `"folder"` option should resolve path to configuration, but it looks like it does not happen in my setup.
My project have the following structure:
- root_folder
   - .vscode
   - app_cm0p
      -   .vscode
   - app_cm4
      - .vscode

Launch order: `Multi-Core Debug (JLink)` (root_folder) -> `Launch PSoC6 CM0+ (JLink) chained` (app_cm0p) -> `Attach PSoC6 CM4 (JLink)` (app_cm4)
My launch.json files: [mtb-example-multi-core-prototype.zip](https://github.com/Marus/cortex-debug/files/7810909/mtb-example-multi-core-prototype.zip)

haneefdm, 2022-01-05:
Hi Andrii, I am not sure what you are trying to do, where you are launching VSCode from and where your workspace file is.

Unless you have a workspace file and are using it (by opening a workspace), then nothing will work at the top level.

If I create a dummy workspace file under the root_folder, it starts to recognize the launch configurations in sub-folders. Otherwise, I can't launch anything,... forget about chained stuff.
The workspace file can be technically anywhere but I make one at the root (which is where I think it should be at unlike what MTB make does)

Btw, your chained config setting in the app_cjm0p/.vscode/launch.json seems correct so long as there are no typos in the name of the config to launch. Folder seems correct.
I notice that your root_folder/.vscode/launch.json configurations are not even recognized by VSCode. It is giving you errors right there.

haneefdm, 2022-01-05:
The 'folder's can be a simple name as seen in the .workspace file. That is what the VSCode documentation says.

haneefdm, 2022-01-12:
No feedback, I am going to close this. Submit new specific issues wrt. chained configurations.

closed 2022-01-12 by haneefdm

AndriiLishchynskyi, 2022-01-12:
Hello Haneef,

Sorry for the delayed response. I am able to use chained configurations without any problems. In my setup I have tried the combination of "launch" and "attach" configurations. Configs are located in different directories.

Just for the record: I was able to use "chained" configuration located at another launch.json. For some reason, ${workspaceRoot} variable that was used to specify path to the chained config does not work as I expected. The things gets working when ${workspaceRoot} has been changed to ${workspaceFoler}. That is my observation on Windows machine.

haneefdm, 2022-01-13:
Thanks. Variables like ${workspaceRoot}, ${workspaceFoler} are all handled by VSCode, and what they expand to depends on how/what-dir you launched VSCode and if you are using workspaces or not.  It works both ways for me.

Anyways, I am glad it is working.