Marus/cortex-debug #531: openOCD does not start from launch

jancoow, 2021-11-15:
I'm unable to get the cortex-debug to actually start openOCD.
It does say `Please check TERMINAL tab (gdb-server) for output from /usr/local/bin/openocd` but this terminal tab never appears. 

My config:

```
        {
            "name": "Cortex-m debugger",
            "type": "cortex-debug",
            "rtos": "FreeRTOS",
            "servertype": "openocd",
            "serverpath": "/usr/local/bin/openocd",
            "configFiles": ["/home/janco/openocd.cfg"],
            "executable": "${workspaceRoot}/build/projectEbikeMonitor1.elf",
            "request": "launch",
            "device": "ATSAMR34J18B",
            "showDevDebugOutput": true
        },
```

haneefdm, 2021-11-15:
You will see the exact command used by Cortex-Debug in the Debug Console. Are you able to run this command from a shell/command-line?

The gdb-server Terminal doesn't show until the first server actually starts. I assume you saw no other error message?

haneefdm, 2021-12-10:
Closing due to no response

closed 2021-12-10 by haneefdm