Marus/cortex-debug #535: Problems using pyocd as servertype

NoaEssiq, 2021-11-17:
I'm trying to debug code on a [MIMXRT1024_evk](https://docs.zephyrproject.org/2.6.0/boards/arm/mimxrt1024_evk/doc/index.html) using pyocd as the server. I'm not getting any errors but it's also not stopping at any breakpoints and the GDB server closes immediately. This is the output I'm getting in the terminal: 
```
SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected
0000959:INFO:board:Target type is mimxrt1024
0001017:INFO:dap:DP IDR = 0x0bd11477 (v1 MINDP rev0)
0001038:INFO:ap:AHB-AP#0 IDR = 0x04770041 (AHB-AP var4 rev0)
0001100:INFO:rom_table:AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fd000 (designer=00e part=88c)
0001108:INFO:rom_table:[0]<e00fe000:ROM class=1 designer=43b part=4c8>
0001108:INFO:rom_table:  AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00fe000 (designer=43b part=4c8)
0001118:INFO:rom_table:  [0]<e00ff000:ROM class=1 designer=43b part=4c7>
0001120:INFO:rom_table:    AHB-AP#0 Class 0x1 ROM table #2 @ 0xe00ff000 (designer=43b part=4c7)
0001131:INFO:rom_table:    [0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0001138:INFO:rom_table:    [1]<e0001000:DWT class=14 designer=43b part=002>
0001192:INFO:rom_table:    [2]<e0002000:FPB class=14 designer=43b part=00e>
0001202:INFO:rom_table:    [3]<e0000000:ITM class=14 designer=43b part=001>
0001209:INFO:rom_table:  [1]<e0041000:ETM-M7 class=9 designer=43b part=975 devtype=13 archid=4a13 devid=0:0:0>
0001216:INFO:rom_table:  [2]<e0042000:CTI class=9 designer=43b part=906 devtype=14 archid=0000 devid=40800:0:0>
0001223:INFO:rom_table:[1]<e0040000:TPIU-M7 class=9 designer=43b part=9a9 devtype=11 archid=0000 devid=ca1:0:0>
0001229:INFO:rom_table:[2]<e0043000:TSGEN class=15 designer=43b part=101>
0001232:INFO:target_imxrt:IMXRT Boot Mode: Internal Boot
0001236:INFO:cortex_m:CPU core #0 is Cortex-M7 r1p1
0001248:INFO:cortex_m:FPU present: FPv5-D16-M
0001315:INFO:dwt:4 hardware watchpoints
0001318:INFO:fpb:8 hardware breakpoints, 1 literal comparators
0001341:INFO:server:Semihost server started on port 50081 (core 0)
0001629:INFO:gdbserver:GDB server started on port 50080 (core 0)
SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

I'm able to debug just using GDB in terminal so I'm fairly sure the gcc arm toolchain is set up correctly.

This is what my ```launch.json``` looks like:
```
{
    "configurations": [
        {
            "cwd": "${workspaceRoot}",
            "executable": "./build/zephyr/zephyr.elf",
            "name": "Debug pyocd",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "pyocd",
            "runToEntryPoint": "main",
            "device": "MIMXRT1024-EVK",
        }
    ]
}
```

Any help getting this to work is very much appreciated!

haneefdm, 2021-11-17:
What is in the 'Debug Console'? You can also enable debug output to see gdb transactions.

It is interesting you got port numbers 50081/50080. Highly unusual.

NoaEssiq, 2021-11-17:
I added ```"showDevDebugOutput": true``` to ```launch.json``` and tried to debug again. The 'Terminal' output didn't change apart from the ports now being 50089 and 50088. The 'Debug Console' output was:
```
Reading symbols from '/home/noa/Documents/kecs/build/zephyr/zephyr.elf'
Finished reading symbols
Please check TERMINAL tab (gdb-server) for output from pyocd
Launching server: "pyocd" "gdbserver" "--port" "50088" "--telnet-port" "50089"
Launching GDB: "arm-none-eabi-gdb" "-q" "--interpreter=mi2"
1-gdb-set target-async on
```

I don't know what the ports mean. Why is weird that I get those ports?

haneefdm, 2021-11-17:
pyOCD is exiting after receiving the first command from Cortex-Debug/gdb. It never even responded it appears.

Those are TCP ports that pyOCD should use. If the ports are busy, then Cortex-Debug will select new ones. Normally, they start at 50000. If you are 50088 that means there may be 44 other instances of pyOCD still running. Every new one that is starting is quitting but still holding on to the TCP ports it opened? That is very strange.

Can you do a 'ps' (or equivalent) to see if there are pyocd processes running? Assuming you can use `lsof`  (or equivalent) to see which ports are open and by which program/process.

NoaEssiq, 2021-11-17:
ps gives an insanely long list of stuff I'm unsure how to interpret. Here's a snipet:
```
teams       1750                               noa    7r      REG              259,2     171563    2639787 /usr/share/teams/v8_context_snapshot.bin
teams       1750                               noa    8r      REG              259,2     180782    2639777 /usr/share/teams/chrome_100_percent.pak
teams       1750                               noa    9r      REG              259,2     322069    2639778 /usr/share/teams/chrome_200_percent.pak
teams       1750                               noa   10r      REG              259,2      81141    2764545 /usr/share/teams/locales/en-US.pak
teams       1750                               noa   11r      REG              259,2    4349723    2639784 /usr/share/teams/resources.pak
teams       1750                               noa   12u     unix 0x0000000000000000        0t0      47492 type=STREAM
teams       1750                               noa   13u     unix 0x0000000000000000        0t0      47493 type=STREAM
teams       1750                               noa   14r     FIFO               0,12        0t0      47494 pipe
teams       1750                               noa   15w     FIFO               0,12        0t0      47494 pipe
teams       1750                               noa   16u  a_inode               0,13          0      14459 [eventpoll]
teams       1750                               noa   17u     unix 0x0000000000000000        0t0      46619 type=STREAM
teams       1750                               noa   18u     unix 0x0000000000000000        0t0      46620 type=STREAM
teams       1750                               noa   19r     FIFO               0,12        0t0      46621 pipe
teams       1750                               noa   20w     FIFO               0,12        0t0      46621 pipe
teams       1750                               noa   21r      CHR                1,9        0t0         10 /dev/urandom
teams       1750                               noa   22u     unix 0x0000000000000000        0t0      44015 type=STREAM
teams       1750                               noa   23r  a_inode               0,13          0      14459 inotify
teams       1750                               noa   24u      REG              259,2        186   26480659 /home/noa/.config/Microsoft/Microsoft Teams/Cache/3fc4d4e656549da0_0
teams       1750                               noa   25u  netlink                           0t0      44016 ROUTE
teams       1750                               noa   26u     unix 0x0000000000000000        0t0      49182 type=STREAM
teams       1750                               noa   28u      REG              259,2      45056   26480841 /home/noa/.config/Microsoft/Microsoft Teams/Cookies
teams       1750                               noa   30u      REG              259,2      28672   26347267 /home/noa/.pki/nssdb/cert9.db
teams       1750                               noa   31u      REG              259,2      36864   26347268 /home/noa/.pki/nssdb/key4.db
```

ps gives this:
```
 PID TTY          TIME CMD
 281233 pts/2    00:00:00 bash
 321151 pts/2    00:00:00 ps
```

haneefdm, 2021-11-17:
you can pipe output of `ps` through `| grep -i pyocd` perhaps.

You can use `lsof -n -iTCP` to see what ports are open and grep for ports starting with 5 ` | grep :5`

https://www.tecmint.com/find-out-which-process-listening-on-a-particular-port/

The second column of `lsof` is the process-id and you can look that up with `ps` as well. You can also use netstat -- I can't tell if you are on a Mac or Linux

NoaEssiq, 2021-11-17:
```lsof -n -iTCP``` worked well. It seems like there are a lot of ports taken by pyocd like you said. Here's the output:
```
COMMAND      PID USER   FD   TYPE  DEVICE SIZE/OFF NODE NAME
pyocd     239104  noa   13u  IPv4  788878      0t0  TCP 127.0.0.1:50000 (LISTEN)
pyocd     239104  noa   14u  IPv4  788881      0t0  TCP 127.0.0.1:50001 (LISTEN)
pyocd     239626  noa   13u  IPv4  798018      0t0  TCP 127.0.0.1:50002 (LISTEN)
pyocd     239626  noa   14u  IPv4  798021      0t0  TCP 127.0.0.1:50003 (LISTEN)
pyocd     239825  noa   13u  IPv4  795997      0t0  TCP 127.0.0.1:50004 (LISTEN)
pyocd     239825  noa   14u  IPv4  796000      0t0  TCP 127.0.0.1:50005 (LISTEN)
pyocd     239890  noa   13u  IPv4  798540      0t0  TCP 127.0.0.1:50006 (LISTEN)
pyocd     239890  noa   14u  IPv4  800188      0t0  TCP 127.0.0.1:50007 (LISTEN)
pyocd     240163  noa   13u  IPv4  797665      0t0  TCP 127.0.0.1:50008 (LISTEN)
pyocd     240163  noa   14u  IPv4  797668      0t0  TCP 127.0.0.1:50009 (LISTEN)
pyocd     240550  noa   13u  IPv4  808078      0t0  TCP 127.0.0.1:50010 (LISTEN)
pyocd     240550  noa   14u  IPv4  808081      0t0  TCP 127.0.0.1:50011 (LISTEN)
pyocd     240685  noa   13u  IPv4  803555      0t0  TCP 127.0.0.1:50012 (LISTEN)
pyocd     240685  noa   14u  IPv4  803558      0t0  TCP 127.0.0.1:50013 (LISTEN)
pyocd     240861  noa   13u  IPv4  811282      0t0  TCP 127.0.0.1:50014 (LISTEN)
pyocd     240861  noa   14u  IPv4  811285      0t0  TCP 127.0.0.1:50015 (LISTEN)
pyocd     241452  noa   13u  IPv4  817251      0t0  TCP 127.0.0.1:50016 (LISTEN)
pyocd     241452  noa   14u  IPv4  815306      0t0  TCP 127.0.0.1:50017 (LISTEN)
pyocd     241878  noa   13u  IPv4  817995      0t0  TCP 127.0.0.1:50018 (LISTEN)
pyocd     241878  noa   14u  IPv4  815564      0t0  TCP 127.0.0.1:50019 (LISTEN)
pyocd     244574  noa   13u  IPv4  823097      0t0  TCP 127.0.0.1:50020 (LISTEN)
pyocd     244574  noa   14u  IPv4  823100      0t0  TCP 127.0.0.1:50021 (LISTEN)
pyocd     245254  noa   13u  IPv4  834779      0t0  TCP 127.0.0.1:50022 (LISTEN)
pyocd     245254  noa   14u  IPv4  834087      0t0  TCP 127.0.0.1:50023 (LISTEN)
pyocd     245623  noa   13u  IPv4  833293      0t0  TCP 127.0.0.1:50024 (LISTEN)
pyocd     245623  noa   14u  IPv4  833296      0t0  TCP 127.0.0.1:50025 (LISTEN)
pyocd     245817  noa   13u  IPv4  839028      0t0  TCP 127.0.0.1:50026 (LISTEN)
pyocd     245817  noa   14u  IPv4  839031      0t0  TCP 127.0.0.1:50027 (LISTEN)
pyocd     245990  noa   13u  IPv4  836432      0t0  TCP 127.0.0.1:50028 (LISTEN)
pyocd     245990  noa   14u  IPv4  842067      0t0  TCP 127.0.0.1:50029 (LISTEN)
pyocd     246249  noa   13u  IPv4  847213      0t0  TCP 127.0.0.1:50030 (LISTEN)
pyocd     246249  noa   14u  IPv4  847216      0t0  TCP 127.0.0.1:50031 (LISTEN)
pyocd     246451  noa   13u  IPv4  847485      0t0  TCP 127.0.0.1:50032 (LISTEN)
pyocd     246451  noa   14u  IPv4  847488      0t0  TCP 127.0.0.1:50033 (LISTEN)
pyocd     247303  noa   13u  IPv4  850620      0t0  TCP 127.0.0.1:50034 (LISTEN)
pyocd     247303  noa   14u  IPv4  850623      0t0  TCP 127.0.0.1:50035 (LISTEN)
pyocd     247376  noa   13u  IPv4  850732      0t0  TCP 127.0.0.1:50036 (LISTEN)
pyocd     247376  noa   14u  IPv4  849840      0t0  TCP 127.0.0.1:50037 (LISTEN)
pyocd     248200  noa   13u  IPv4  860506      0t0  TCP 127.0.0.1:50038 (LISTEN)
pyocd     248200  noa   14u  IPv4  860509      0t0  TCP 127.0.0.1:50039 (LISTEN)
pyocd     248326  noa   13u  IPv4  859602      0t0  TCP 127.0.0.1:50040 (LISTEN)
pyocd     248326  noa   14u  IPv4  859605      0t0  TCP 127.0.0.1:50041 (LISTEN)
pyocd     248912  noa   13u  IPv4  868227      0t0  TCP 127.0.0.1:50042 (LISTEN)
pyocd     248912  noa   14u  IPv4  868230      0t0  TCP 127.0.0.1:50043 (LISTEN)
pyocd     249155  noa   13u  IPv4  878976      0t0  TCP 127.0.0.1:50044 (LISTEN)
pyocd     249155  noa   14u  IPv4  878979      0t0  TCP 127.0.0.1:50045 (LISTEN)
pyocd     249236  noa   13u  IPv4  881666      0t0  TCP 127.0.0.1:50046 (LISTEN)
pyocd     249236  noa   14u  IPv4  881669      0t0  TCP 127.0.0.1:50047 (LISTEN)
pyocd     249428  noa   13u  IPv4  874489      0t0  TCP 127.0.0.1:50048 (LISTEN)
pyocd     249428  noa   14u  IPv4  874492      0t0  TCP 127.0.0.1:50049 (LISTEN)
pyocd     249693  noa   13u  IPv4  886327      0t0  TCP 127.0.0.1:50050 (LISTEN)
pyocd     249693  noa   14u  IPv4  886330      0t0  TCP 127.0.0.1:50051 (LISTEN)
pyocd     249846  noa   13u  IPv4  887480      0t0  TCP 127.0.0.1:50052 (LISTEN)
pyocd     249846  noa   14u  IPv4  887483      0t0  TCP 127.0.0.1:50053 (LISTEN)
pyocd     252995  noa   13u  IPv4  931887      0t0  TCP 127.0.0.1:50054 (LISTEN)
pyocd     252995  noa   14u  IPv4  931890      0t0  TCP 127.0.0.1:50055 (LISTEN)
pyocd     253303  noa   13u  IPv4  933096      0t0  TCP 127.0.0.1:50056 (LISTEN)
pyocd     253303  noa   14u  IPv4  933099      0t0  TCP 127.0.0.1:50057 (LISTEN)
pyocd     256270  noa   13u  IPv4  946850      0t0  TCP 127.0.0.1:50058 (LISTEN)
pyocd     256270  noa   14u  IPv4  946853      0t0  TCP 127.0.0.1:50059 (LISTEN)
pyocd     257271  noa   13u  IPv4  954758      0t0  TCP 127.0.0.1:50060 (LISTEN)
pyocd     257271  noa   14u  IPv4  953789      0t0  TCP 127.0.0.1:50061 (LISTEN)
code      260159  noa   68u  IPv4  967246      0t0  TCP 127.0.0.1:55878 (LISTEN)
pyocd     262434  noa   13u  IPv4  990495      0t0  TCP 127.0.0.1:50062 (LISTEN)
pyocd     262434  noa   14u  IPv4  990498      0t0  TCP 127.0.0.1:50063 (LISTEN)
[... cut ...]
pyocd     288631  noa   14u  IPv4 1079743      0t0  TCP 127.0.0.1:50083 (LISTEN)
pyocd     288763  noa   13u  IPv4 1077712      0t0  TCP 127.0.0.1:50084 (LISTEN)
pyocd     288763  noa   14u  IPv4 1077715      0t0  TCP 127.0.0.1:50085 (LISTEN)
pyocd     318071  noa   13u  IPv4 1143428      0t0  TCP 127.0.0.1:50086 (LISTEN)
pyocd     318071  noa   14u  IPv4 1143431      0t0  TCP 127.0.0.1:50087 (LISTEN)
pyocd     318178  noa   13u  IPv4 1145552      0t0  TCP 127.0.0.1:50088 (LISTEN)
pyocd     318178  noa   14u  IPv4 1149042      0t0  TCP 127.0.0.1:50089 (LISTEN)

```

haneefdm, 2021-11-17:
You can use `killall pyocd` to nuke them all. I am assuming that works on Linux.

It appears if one is blocked, it is blocking all future ones. If you ever see anything other low 50000's, I would check to see what happened. Maybe even report it to us. It is interesting that it exits -- according to the gdb-server console but it is in fact running. It is possible the main thread exited but a spawned thread continues to run> Just guessing.

NoaEssiq, 2021-11-18:
```killall pyocd´´´ worked. I ran ```lsof -n -iTCP``` again and it showed no pyocd COMMANDs. If I try to debug now I get the same terminal and debug console messages apart from the ports being 50000 and 50001. If I try again the ports increase to 50002 and 50003.

haneefdm, 2021-11-18:
More than one strange thing going on here. What debug hardware are you using? Black magic probe? If so, you don't need/want pyocd.  You seemed to connect directly to a serial port. The servertype is "bmp"

I am curious that pyocd also worked for you to some extent. Whey/how? Again, it would help me know what your debug setup looks like.

NoaEssiq, 2021-11-19:
My debug setup is just a usb cable connected directly to the board from my laptop. From what I can understand [here](https://docs.zephyrproject.org/2.6.0/boards/arm/mimxrt1024_evk/doc/index.html) the board has a built in debug probe called [OpenSDA DAPLink](https://docs.zephyrproject.org/2.6.0/guides/flash_debug/probes.html#opensda-daplink-onboard-debug-probe). The chip is a LPC4322.

haneefdm, 2021-11-19:
Okay, that is a setup I have not seen before. We can continue trying to use pyocd (or do it another way if that fails). But, I think we can make pyocd work. I found this page

https://docs.zephyrproject.org/1.9.0/tools/opensda.html

In the debugging section https://docs.zephyrproject.org/1.9.0/tools/opensda.html#debugging

I see that make launches both pyOCD and gdb. We need to find out what the exact command line was used for pyOCD. We can do this in one of several ways. Run make as suggested by the link above, and in another terminal run `ps -eo pid,user,args` Or you can replace pyocd which a shell script that prints its arguments.

You can also look in the make file to see what it is doing. You can see if the makefile provides a verbose option or do a `make -n OPENSDA_FW=daplink debug` or `make SHELL="/bin/bash -x" OPENSDA_FW=daplink debug`

The link you provided says that pyOCD does not support programming which can be a problem but clearly you found a way to program the device. We will cross that bridge later.

NoaEssiq, 2021-11-19:
The links you provided were for version 1.9, I use 2.5. I couldn't find them for version 2.5. I haven't used the ```make``` command when working with zephyr. They use [west](https://docs.zephyrproject.org/latest/guides/west/index.html) and to debug with that I can use [west debug](https://docs.zephyrproject.org/latest/guides/west/build-flash-debug.html#debugging-west-debug-west-debugserver). Using that command I got this output:
```
noa@noa-essiq ~/Documents/kecs (conf_str_read_write-786) $ west debug
-- west debug: rebuilding
[0/1] cd /home/noa/Documents/kecs/build/zephyr/cmake/flash && /usr/bin/cmake -E echo

-- west debug: using runner pyocd
-- runners.pyocd: pyOCD GDB server running on port 3333
GNU gdb (crosstool-NG 1.24.0.212_d7da3a9) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=arm-zephyr-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/noa/Documents/kecs/build/zephyr/zephyr.elf...
0001127:INFO:board:Target type is mimxrt1024
0001169:INFO:dap:DP IDR = 0x0bd11477 (v1 MINDP rev0)
0001188:INFO:ap:AHB-AP#0 IDR = 0x04770041 (AHB-AP var4 rev0)
0001246:INFO:rom_table:AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fd000 (designer=00e part=88c)
0001255:INFO:rom_table:[0]<e00fe000:ROM class=1 designer=43b part=4c8>
0001256:INFO:rom_table:  AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00fe000 (designer=43b part=4c8)
0001269:INFO:rom_table:  [0]<e00ff000:ROM class=1 designer=43b part=4c7>
0001269:INFO:rom_table:    AHB-AP#0 Class 0x1 ROM table #2 @ 0xe00ff000 (designer=43b part=4c7)
0001279:INFO:rom_table:    [0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0001329:INFO:rom_table:    [1]<e0001000:DWT class=14 designer=43b part=002>
0001337:INFO:rom_table:    [2]<e0002000:FPB class=14 designer=43b part=00e>
0001342:INFO:rom_table:    [3]<e0000000:ITM class=14 designer=43b part=001>
0001347:INFO:rom_table:  [1]<e0041000:ETM-M7 class=9 designer=43b part=975 devtype=13 archid=4a13 devid=0:0:0>
0001354:INFO:rom_table:  [2]<e0042000:CTI class=9 designer=43b part=906 devtype=14 archid=0000 devid=40800:0:0>
0001361:INFO:rom_table:[1]<e0040000:TPIU-M7 class=9 designer=43b part=9a9 devtype=11 archid=0000 devid=ca1:0:0>
0001367:INFO:rom_table:[2]<e0043000:TSGEN class=15 designer=43b part=101>
0001369:INFO:target_imxrt:IMXRT Boot Mode: Internal Boot
0001372:INFO:cortex_m:CPU core #0 is Cortex-M7 r1p1
0001436:INFO:cortex_m:FPU present: FPv5-D16-M
0001443:INFO:dwt:4 hardware watchpoints
0001445:INFO:fpb:8 hardware breakpoints, 1 literal comparators
0001616:INFO:server:Semihost server started on port 4444 (core 0)
0001710:INFO:gdbserver:GDB server started on port 3333 (core 0)
Remote debugging using :3333
0002462:INFO:gdbserver:Client connected to port 3333!
sys_dlist_init (list=0x6006aefc <mcux_lpuart_driver_api+56>) at /home/noa/zephyrproject/zephyr/include/sys/dlist.h:194
194		list->head = (sys_dnode_t *)list;
0002568:INFO:gdbserver:Attempting to load argon
0002568:INFO:gdbserver:Attempting to load freertos
0002568:INFO:gdbserver:Attempting to load rtx5
0002568:INFO:gdbserver:Attempting to load threadx
0002568:INFO:gdbserver:Attempting to load zephyr
Successfully halted device
Resetting target
Loading section rom_start, size 0x2278 lma 0x60000000
Loading section text, size 0x5bd84 lma 0x60002280
Loading section .ARM.exidx, size 0x8 lma 0x6005e004
Loading section initlevel, size 0x110 lma 0x6005e00c
Loading section sw_isr_table, size 0x470 lma 0x6005e11c
Loading section net_socket_register_area, size 0xc lma 0x6005e58c
Loading section log_const_sections, size 0xf0 lma 0x6005e598
Loading section log_backends_sections, size 0x10 lma 0x6005e688
Loading section shell_area, size 0x30 lma 0x6005e698
Loading section shell_root_cmds_sections, size 0x138 lma 0x6005e6c8
Loading section device_handles, size 0x9e lma 0x6005e800
Loading section rodata, size 0x20764 lma 0x6005e8a0
[... 18 lines cut ...]
0083067:INFO:target_imxrt:IMXRT Boot Mode: Internal Boot
0083069:INFO:target_imxrt:IMXRT Boot Device: 0
0083092:INFO:loader:Erased 532480 bytes (130 sectors), programmed 532480 bytes (2080 pages), skipped 8192 bytes (32 pages) at 6.57 kB/s
Start address 0x60015ee0, load size 536903
Transfer rate: 6 KB/sec, 1826 bytes/write.
(gdb) 
```
In another terminal I ran ```ps -eo pid,user,args``` and grepped the lines with pyocd in it and got this:
```
noa@noa-essiq ~/Documents/kecs (conf_str_read_write-786) $ ps -eo pid,user,args | grep pyocd
 459244 noa      /usr/bin/python3 /home/noa/.local/bin/pyocd gdbserver -p 3333 -T 4444 -t mimxrt1024
 459961 noa      grep --color=auto pyocd
```

I use ```west flash``` to program the board and I'm fairly sure that uses pyocd. This is the output I get from ```west flash```:
```
noa@noa-essiq ~/Documents/kecs (conf_str_read_write-786) $ west flash
-- west flash: rebuilding
[0/1] cd /home/noa/Documents/kecs/build/zephyr/cmake/flash && /usr/bin/cmake -E echo

-- west flash: using runner pyocd
-- runners.pyocd: Flashing file: /home/noa/Documents/kecs/build/zephyr/zephyr.hex
[====================] 100%
0032096:INFO:loader:Erased 8192 bytes (2 sectors), programmed 8192 bytes (32 pages), skipped 532480 bytes (2080 pages) at 17.63 kB/s
```

haneefdm, 2021-11-19:
Hmmm. They are doing pretty much what we are. When you tried to launch via VSCode we neget got to the point where gdb made a connection. We should have seen something like this (different ports of course)
```
Remote debugging using :3333
0002462:INFO:gdbserver:Client connected to port 3333!
```

When a gdb-server (pyocd) starts and ends there are messages logged into a file `/tmp/cortex-debug-server-exiting.log`. See if that file exists and see what is in it? The server is exiting as soon as it starts but leaving a zombie behind. But, we should have some indication of that.

haneefdm, 2021-11-19:
There are also files `/tmp/gdb-server-console-XXXXX`. Look at the latest. it should have what you see in the terminal and maybe a bit more

NoaEssiq, 2021-11-19:
I removed all ```gdb-server-console-XXXX```  files from ```/tmp/``` and attempted to debug via vs code. After getting the usual terminal and debug console messages I checked ```tmp``` and there were no ```gdb-server-console-XXXX```  files. I didn't have any ```/tmp/cortex-debug-server-exiting.log``` either. I could find some ```Cortex-Debug-XXXXX.json``` files though. About half of them said that they were accessed 2024-05-11 which is impossible.

After running ```west debug``` I also didn't have any  ``gdb-server-console-XXXX```  files in ```/tmp/```.

haneefdm, 2021-11-19:
The logfiles I mentioned are **only** generated by our extension when you are running it inside VSCode.

NoaEssiq, 2021-11-22:
I don't think it generates those files for me. At least I don't have any in ```/tmp/```. I ran: ```sudo find / -name *cortex-debug-server-exiting*``` and ```sudo find / -name *gdb-server-console-*``` to scan the whole computer for those files and it didn't find any.

haneefdm, 2021-11-22:
There is a new release out that maybe can help with the log files. Could you try the latest -- VSCode should have auto-updated to 0.4.8. See also changelog.

NoaEssiq, 2021-11-22:
When I updated I got an error message saying that ```gcc-arm-none-eabi``` couldn't be found. I solved this by adding ```"gdbPath": "/home/noa/gcc-arm-none-eabi-10.3-2021.10/",``` to my ```launch.json```. I also couldn't get the ```gdbServerLogfile``` to work. I figured out that ```/tmp/gdb-server-console-XXXXX``` were sometimes created when I restarted vs code. Here's the contents of one of those:
```
[2021-11-22T13:50:48.088Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
0000874:INFO:board:Target type is mimxrt1024
0000933:INFO:dap:DP IDR = 0x0bd11477 (v1 MINDP rev0)
0000951:INFO:ap:AHB-AP#0 IDR = 0x04770041 (AHB-AP var4 rev0)
0001004:INFO:rom_table:AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fd000 (designer=00e part=88c)
0001012:INFO:rom_table:[0]<e00fe000:ROM class=1 designer=43b part=4c8>
0001012:INFO:rom_table:  AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00fe000 (designer=43b part=4c8)
0001023:INFO:rom_table:  [0]<e00ff000:ROM class=1 designer=43b part=4c7>
0001023:INFO:rom_table:    AHB-AP#0 Class 0x1 ROM table #2 @ 0xe00ff000 (designer=43b part=4c7)
0001032:INFO:rom_table:    [0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0001039:INFO:rom_table:    [1]<e0001000:DWT class=14 designer=43b part=002>
0001044:INFO:rom_table:    [2]<e0002000:FPB class=14 designer=43b part=00e>
0001051:INFO:rom_table:    [3]<e0000000:ITM class=14 designer=43b part=001>
0001097:INFO:rom_table:  [1]<e0041000:ETM-M7 class=9 designer=43b part=975 devtype=13 archid=4a13 devid=0:0:0>
0001103:INFO:rom_table:  [2]<e0042000:CTI class=9 designer=43b part=906 devtype=14 archid=0000 devid=40800:0:0>
0001108:INFO:rom_table:[1]<e0040000:TPIU-M7 class=9 designer=43b part=9a9 devtype=11 archid=0000 devid=ca1:0:0>
0001116:INFO:rom_table:[2]<e0043000:TSGEN class=15 designer=43b part=101>
0001120:INFO:target_imxrt:IMXRT Boot Mode: Internal Boot
0001123:INFO:cortex_m:CPU core #0 is Cortex-M7 r1p1
0001135:INFO:cortex_m:FPU present: FPv5-D16-M
0001144:INFO:dwt:4 hardware watchpoints
0001150:INFO:fpb:8 hardware breakpoints, 1 literal comparators
0001222:INFO:server:Semihost server started on port 50001 (core 0)
0001495:INFO:gdbserver:GDB server started on port 50000 (core 0)
[2021-11-22T13:50:49.627Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
```
Seems to be the same as in the terminal. 

I couildn't get the ```/tmp/cortex-debug-server-exiting.log``` file to be created sadly.

haneefdm, 2021-11-22:
I don't know where else to go from here. I wish I could see what you are seeing. The `cortex-debug-server-exiting.log` HAS to be created unless we don't have write permission in /tmp or you have an environment variable like TMPDIR or TMP setting /tmp to someplace else. You can see the code here.

https://github.com/Marus/cortex-debug/blob/68b09c25913a9bde3a0ef60df56dadbfff1e78df/src/backend/server.ts#L10

It is `dbgServerLogfile` in your settings btw, not `gdbServerLogfile`

Only thing I can say is that you have to (painfully) run the *exact* same commands as what you see in the Debug Console in two separate terminals and issue the very same commands that Cortex-Debug is issuing.

NoaEssiq, 2021-11-23:
I fiddled around some more and managed to create the ```cortex-debug-server-exiting.log```. If I try to debug it's not created. If I don't unplug, restart anything or kill pyocd and try to debug again I get another error: ```Failed to launch PyOCD GDB Server: Timeout.```. This time ```cortex-debug-server-exiting.log``` is created and looks like this:
```
[2021-11-23T06:55:49.855Z] GDBServer: exited 1 null
[2021-11-23T06:55:58.811Z] Begin disconnectRequest
```

Even when fixing the typo, I feel dumb, I couldn't get ```gdbServerLogfile``` working. ```gdb-server-console-XXXXX``` are still created in ```/tmp```, and only sometimes when I restart vs code. 

My pyocds don't stop unless I use ```killall pyocd``` so maybe Cortex-Debug doesn't recognise that it has finished running and therefor doesn't create the files? I'm grasping at straws here.

How do I see the commands issued by Cortex-Debug? I tried doing what's in the debug console in two terminals.
Terminal 1(pyocd term):
```
pyocd gdbserver --port 50000 --telnet-port 50001
```

and in terminal 2(gdb term):
```
noa@noa-essiq ~/Documents/kecs (debug) $ gdb -q --interpreter mi2  build/zephyr/zephyr.elf
=thread-group-added,id="i1"
~"Reading symbols from build/zephyr/zephyr.elf...\n"
(gdb) 
set target-async on 
&"set target-async on\n"
=cmd-param-changed,param="mi-async",value="on"
^done
```
The formatting was weird for some reason in gdb term but it didn't give any error. In the pyocd term I seemingly got the same output as before.

I don't think the debug console told my to connect gdb but I tried that as well and now it crashed. After doing ```target remote :50000``` in gdb term I got this output:
```
target remote :50000
&"target remote :50000\n"
~"Remote debugging using :50000\n"
&"warning: Architecture rejected target-supplied description\n"
=thread-group-started,id="i1",pid="42000"
=thread-created,id="1",group-id="i1"
*** stack smashing detected ***: terminated
Aborted (core dumped)
```

and this in pyocd term:
```
noa@noa-essiq ~/Documents/kecs (debug) $ pyocd gdbserver --port 50000 --telnet-port 50001
0000930:INFO:board:Target type is mimxrt1024
0001007:INFO:dap:DP IDR = 0x0bd11477 (v1 MINDP rev0)
0001027:INFO:ap:AHB-AP#0 IDR = 0x04770041 (AHB-AP var4 rev0)
0001085:INFO:rom_table:AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00fd000 (designer=00e part=88c)
0001097:INFO:rom_table:[0]<e00fe000:ROM class=1 designer=43b part=4c8>
0001098:INFO:rom_table:  AHB-AP#0 Class 0x1 ROM table #1 @ 0xe00fe000 (designer=43b part=4c8)
0001108:INFO:rom_table:  [0]<e00ff000:ROM class=1 designer=43b part=4c7>
0001109:INFO:rom_table:    AHB-AP#0 Class 0x1 ROM table #2 @ 0xe00ff000 (designer=43b part=4c7)
0001161:INFO:rom_table:    [0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0001169:INFO:rom_table:    [1]<e0001000:DWT class=14 designer=43b part=002>
0001178:INFO:rom_table:    [2]<e0002000:FPB class=14 designer=43b part=00e>
0001186:INFO:rom_table:    [3]<e0000000:ITM class=14 designer=43b part=001>
0001195:INFO:rom_table:  [1]<e0041000:ETM-M7 class=9 designer=43b part=975 devtype=13 archid=4a13 devid=0:0:0>
0001257:INFO:rom_table:  [2]<e0042000:CTI class=9 designer=43b part=906 devtype=14 archid=0000 devid=40800:0:0>
0001266:INFO:rom_table:[1]<e0040000:TPIU-M7 class=9 designer=43b part=9a9 devtype=11 archid=0000 devid=ca1:0:0>
0001274:INFO:rom_table:[2]<e0043000:TSGEN class=15 designer=43b part=101>
0001277:INFO:target_imxrt:IMXRT Boot Mode: Internal Boot
0001281:INFO:cortex_m:CPU core #0 is Cortex-M7 r1p1
0001412:INFO:cortex_m:FPU present: FPv5-D16-M
0001452:INFO:dwt:4 hardware watchpoints
0001473:INFO:fpb:8 hardware breakpoints, 1 literal comparators
0001491:INFO:server:Semihost server started on port 50001 (core 0)
0001533:INFO:gdbserver:GDB server started on port 50000 (core 0)
0558514:INFO:gdbserver:Client connected to port 50000!
0559034:INFO:gdbserver:Client disconnected from port 50000!
```

NoaEssiq, 2021-11-24:
I can't believe it. I reinstalled the gcc armchain and now it works. Thank you for the help! This is truly going to change how I work and drastically improve my coding life.

closed 2021-11-24 by NoaEssiq