Marus/cortex-debug #536: Os specific setting bug?

darexa, 2021-11-22:
Going from 0.4.7 to 0.4.8 broke my debugging settings.
Specifically I used "cortex-debug.gdbPath" in my settings file and it now complains that the GDB file does not exist.
If I understand the release notes correctly it should be looking for .linux first(in my case) and then default to the regular one.
It seems neither .linux or the default path works correctly.

Looking at the code it seems it is looking for '-'linux rather than '.'linux and adding '-'linux to my settings file made it work again.

https://github.com/Marus/cortex-debug/blob/8fbeb1a85cb47d32e8008354ab02e61d03984ff7/src/frontend/configprovider.ts#L184

haneefdm, 2021-11-22:
Sorry, issue should be addressed. Thanks for reporting this. There was some confusion as to which suffix to use and code did not match the intent. Please try out the new release (should auto update shortly) and reopen if issues persist

closed 2021-11-22 by haneefdm

darexa, 2021-11-23:
Thanks, works fine now!