Marus/cortex-debug #538: SVD does not load since V0.4.8

ThomasPtO, 2021-11-23:
Cortex Peripherals box says: "Unable to load SVD file: <path-to-svd>"

Same launch.json works when switching back to version V0.4.7.

Problem occures also with V0.4.10.

Regards,
Thomas

jmbrederveld, 2021-12-02:
Same here, specifically for the plugin in combination with the STM32F0 platform. I seem to have no issues with the latest version with the STM32F4 platform.

haneefdm, 2021-12-02:
Could you please attach a compressed version of the SVD file? It could mean a parsing issue or even a crash. Were there any popups?

ThomasPtO, 2021-12-02:
No popups on my side. Just the the message in the box, where the registers normally show up.

[LPC55S06.zip](https://github.com/Marus/cortex-debug/files/7642367/LPC55S06.zip)

haneefdm, 2021-12-04:
We found and fixed the problem. Some code meant for debugging was left in the release which caused this problem. Will be fixed in the next release. Already pushed the changes to master

haneefdm, 2021-12-07:
@ThomasPtO @jmbrederveld I just published a pre-release if you want to try out the fix. See

https://github.com/Marus/cortex-debug/releases/tag/v0.4.11-pre1

There are other unrelated changes but they could affect you.

sjuut, 2021-12-07:
I can confirm this release solves the issue. Thank you very much!

haneefdm, 2021-12-10:
We are moving to a new versioning system that supports pre-releases. You have to update to the latest V1.63.0+ of VSCode. In the extensions area, you should now see a green dot
[image]
You will also see the following for install options.
[image]

closed 2021-12-10 by haneefdm