Marus/cortex-debug #539: Loading of symbol table is painfully slow

EvilEli, 2021-11-25:
I'm currently working on a rather big project where the resulting `.elf` file is around 100 KB.
When I load the `.elf` file containing the symbol table the launch process is taking pretty long: ~20s.
After that the actual download of the program to the device is started.
However when loading the `.hex` file, i.e. there are no symbols to load, the download of the program starts instantly.
This leads me to the assumption that the loading process of the symbol table is taking a lot of time.

On the other hand, when using Eclipse (which uses the same server and debugger interface) the loading time of the symbol table is not significant.
```json
"configurations": [
  {
      "type": "cortex-debug",
      "request": "launch",
      "name": "Cortex Debug",
      "executable": "<path to firmware>/firmware.elf/hex",
      "serverpath": "<path to jlink>/jlink-6.94d/JLinkGDBServerCL.exe",
      "servertype": "jlink",
      "device": "STM32MP15xx_A7_0_DualFlash",
      "interface": "jtag",
      "serverArgs": [
        "-speed",
        "15000",
        "-endian",
        "little",
        "-vd",
        "-ir",
        "-localhostonly",
        "1",
        "-strict",
        "-timeout",
        "0",
      ]
  }
]
```

Do there exist workarounds to speed up the loading times of the symbol table?
Or did anyone else experience this behavior?

haneefdm, 2021-11-25:
It depends on what is slow. Loading of the symbols is something we have to do or loading of te program into the device is something gdb does. When you use the hex file, gdb does not have extract sections and program sections at a time, it is done in bulk. Which method ismyour Eclipse using? You can do either in Eclipse?

You can enable `showDevDebugOutput` and  `showDevDebugTimestamps` to see if it is us. You can also ask gdb to load a hex file (instead of the executable) by using `overrideLaunchCommands`

Note that we cache the symbol results. 20 seconds is a long time. Can you provide some machine details?

haneefdm, 2021-11-25:
See for some doc. on launch.json options mentioned above.

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

EvilEli, 2021-11-25:
I enabled `showDevDebugOutput` and `showDevDebugTimestamps` already and from what I can see it is the cortex-debug extension which is taking long:
```
0000000007: Reading symbols from '<path to firmware>/firmware.elf'
0000052454: Finished reading symbols
0000052483: Please check TERMINAL tab (gdb-server) for output from <path to jlink>/jlink-6.94d/JLinkGDBServerCL.exe
```
On Eclipse the symbol loading is done as well and takes basically no time since the download of the program via GDB starts instantly and I am able to step through the code as expected.
GDB loading is started afterwards and the download of the program into the device is also taking a few seconds (on both Eclipse and cortex-debug) which I consider a normal amount of time.

How is it possible to cache the symbol results? Don't they change with every single compilation change?

Machine details:
Win10
11th Gen Intel Core i7-1185G7 @ 3.00GHz
32 GB RAM

haneefdm, 2021-11-25:
> How is it possible to cache the symbol results? Don't they change with every single compilation change?

Yes, we keep track of executable timestamps. Helpful when people go through many debug sessions before a re-compile. We can do the symbol loading asynchronously as well but a bit complicated, some things won't work (like showing Locals/Globals in Variable Window) immediately.

That is a pretty fast machine. Any chance, I can have the executable? Something is not right. 100KB is tiny -- I was thinking MBs. Maybe something wrong/changed with the GNU tools as well perhaps. Not the first time. Any gnu tools details?

Can you time the following command (in bash for example)
```bash
time gcc-arm-none-eabi-objdump -Wi -syms -C elffile > outfile.txt
```
This will tell us whether it is running the command or us parsing the output that is taking time
On Windows, a Virus Scanner can play havoc.

EvilEli, 2021-11-26:
I'm indeed working on a Windows machine.

I'm using gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major)
Here the timed output:
```
time arm-none-eabi-objdump.exe -Wi --syms -C firmware.elf > outfile.txt >   0.01s user 0.01s system 0% cpu 45.987 total
```

haneefdm, 2021-11-26:
Is that saying that it took ~46 seconds total to run that command even though the user/sys/cpu time was almost nothing? That is insane! Even for Windows with Defender on steroids. Hope that is ~46 milliseconds and even that is big.

Any chance I can have the elf file?

Does anyone else see this kind of (bad) performance? I am asking the community.

EvilEli, 2021-11-26:
It is indeed 46 seconds. At least it didn't feel like 46 milliseconds :)

I'm afraid that I cannot share the elf file, since I'm not the owner.

haneefdm, 2021-11-26:
Unless I have something to work with, I have done as much as I can. It appears to be a problem with your Machine. I don't know anyway of telling VSCode (actually NodeJS) to use another shell. I believe it already uses what is native to the machine (cmd.exe). It may not even use a shell at all and use Windows APIs directly. It may have something to do with fileio.

Reopen if there is something we can do.

EvilEli, 2021-11-29:
I'm not sure if it is enough to reopen, but also people on other machines reported the same symptoms.
I was also able to track the issue further:

- It's not the shell but the `-Wi` (which is supposed to display DWARF info) which makes the command take ~40 secs on either shell.

It most likely has something to do with our binary. Is there a way to replace the objdump step? Why is Eclipse that much faster? Do they use another approach?

haneefdm, 2021-11-29:
But you yourself told me that the ~40 secs depended on the shell. I still think that is us your system (file IO) related because there was no CPU time taken during those 40 secs.

How bug was the output file anyways? Gigabytes? If not 40 seconds is not explainable.

haneefdm, 2021-11-29:
And, yes, we have to use `-Wi` or Variables Window does not work properly for Statics and some other things don't work.

EvilEli, 2021-11-29:
> But you yourself told me that the ~40 secs depended on the shell. I still think that is us your system (file IO) related because there was no CPU time taken during those 40 secs.
> 
> How big was the output file anyways? Gigabytes? If not 40 seconds is not explainable.

Yes I accidentally forgot the `-Wi` flag when testing it with the other shell. That's why it was suddely fast.
The resulting output file has a size of 1.8 GB

haneefdm, 2021-11-29:
1.8 GB, I can't see how less than a megabyte binary gets results in a 1.8 GB of debug information.

We have looked at other ways to get the debug information but to this day, no luck. I will continue looking

EvilEli, 2021-11-29:
I have to correct myself (miscounted the zeros): The elffile is ~100 MB instead of KB.

haneefdm, 2021-11-29:
Can you time this for me. Sending the output to a null device. Wan't to know what fileio overhead is.

```
gcc-arm-none-eabi-objdump -Wi -syms -C elffile > /dev/null
```

haneefdm, 2021-11-29:
Also can you report the size of files `ls -lrt /tmp/Cortex-Debug-*.json`

EvilEli, 2021-11-30:
> gcc-arm-none-eabi-objdump -Wi -syms -C elffile > /dev/null

outfile.txt: arm-none-eabi-objdump   1.53s user 12.12s system 13% cpu 1:38.51 total
/dev/null:  arm-none-eabi-objdump   1.01s user   9.30s system 23% cpu 44.444 total

EvilEli, 2021-11-30:
> Also can you report the size of files `ls -lrt /tmp/Cortex-Debug-*.json`

Not sure where to find this file under Windows..

haneefdm, 2021-11-30:
On Windows, it would be in the dir specified by %TMP% or %TEMP% env. variables. It would be something like

C:\Users\<user-name>\AppData\Local\Temp

haneefdm, 2021-12-07:
@EvilEli I have modified the extension to get some debug info using GDB. It is also slow but you probably better than what we are getting with `-Wi` on Objdump. There is no way for me to tell.

Are you willing to try this. I can give you a custom build that you can install. For this to work, your GDB must be version 9 or higher.

EvilEli, 2021-12-07:
Wow that sounds great!
I´d be pleased to try it out!

haneefdm, 2021-12-07:
Just uploaded it. See

https://github.com/Marus/cortex-debug/releases/tag/v0.4.11-pre1

If you enable debug info, you will likely see a very large quantity of output in the Debug Console. at the start

EvilEli, 2021-12-08:
I just tested it and it is blazing fast!
The download of the binary to the device just starts instantly without any slight delay.
This modification is gold for large binaries!

haneefdm, 2021-12-08:
I am glad. The thing to look for is to make sure your Statics in the Variable Window are all accounted for. And make sure you are looking at the runtimes by removing Cortex-Debug*.json files so that the cache is not involved.

haneefdm, 2021-12-10:
We are moving to a new versioning system that supports pre-releases. You have to update to the latest V1.63.0+ of VSCode. In the extensions area, you should now see a green dot
[image]
You will also see the following for install options.
[image]

EvilEli, 2022-02-09:
Hello @haneefdm ,

I just updated my VS Code to a newer version (which in turns fetched the newest Cortex-Debug extension).
It seems like the issue fixed within this issue is present again.
Were those changes reverted due to #179 or is there another reason why this could be?
Is there a solution I could apply by a specific configuration in the launch.json?
Or simply installing version 0.4.11-pre manually again?

haneefdm, 2022-02-09:
Yes, some changes have been made but it should not result in that kind of slow down!!!. May I ask what version you are using now? The method (gdb) we switched to in 0.4.11-pre had many issues with gdb giving only partial information.

In fact, the old code that used the -Wi option has been ripped out. The latest you can try is v1.3.4 but it is a pre-release and you have to enable it in VSCode

haneefdm, 2022-02-09:
If you enable `showDevDebugOutput` in `launch.json`, you can see commands being run and you can run the same and time them in a shell. There are times printed in the debug output as well.

All this in the pre-release -- not the release version.

EvilEli, 2022-02-10:
I switched to the pre-release version (1.3.4).
The biggest step in the timestamps seems again to be caused by the reading of symbols which takes around 40 seconds:

```
[...]
0000004851: WARNING: Cortex-Debug will deprecate use of GDB version 8. Please upgrade to version 9+
0000004851: 3-interpreter-exec console "set print demangle on"
0000004851: 4-interpreter-exec console "set print asm-demangle on"
0000004866: GDB -> App: {"token":3,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0000004866: GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"cmd-param-changed","output":[["param","print asm-demangle"],["value","on"]]}]}
0000004866: GDB -> App: {"token":4,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
0000045890: Reading symbols from nm: Time: 45614 ms
[...]
```

haneefdm, 2022-02-10:
Could I ask you to run those two commands in a shell and report their run times?

haneefdm, 2022-02-11:
Could you try out a manual build I created for this? I made it so that we don't wait for `nm` to finish and we can let it catch up until **really** needed

https://github.com/Marus/cortex-debug/releases/tag/v1.3.5-pre1

I am still baffled by the size of the outputs. No one else has reported this. I wish I could have your executable -- maybe someway to obfuscate it. It is super hard to imagine stuff like this and provide solutions that I have no idea work.

I can sign an NDA if you wish.

EvilEli, 2022-02-14:
I ran the commands in a separate shell:
```
time arm-none-eabi-objdump --syms -C -h -w elffile  > /dev/null
arm-none-eabi-objdump 0.00s user 0.02s system 9% cpu 0.193 total
```
```
time arm-none-eabi-nm--defined-only -S -l -C -p elffile   > /dev/null
arm-none-eabi-nm-S -  0.03s user 0.09s system 0% cpu 52.517 total
```

I tried your manual build. It makes the whole thing prety fast again.
Personally I like the solution because it kind of parallelizes waiting time (we have to wait anyways for the firmware download to finish), but of course my use case may not be representative.

Regarding the NDA:
I'd need to figure this out first (because of the process compliance) but I fear that my hands are tied.
I feel really sorry since I really would like to support the development of this extension as much as possible.

closed 2022-02-19 by haneefdm