Marus/cortex-debug #540: since version 4.5 it can't start debug

ilia19911, 2021-11-29:
Hi, I am using ubuntu 20 and trying to start debugging my stm32f103 project using stlink 2.0. It used to work well, but since v 0.4.5 I get the error "Failed to start Openocd gdb server: Timeout". To fix this problem, I reboot version 0.4.4 and it works fine. My config

```json
           {
                "name": "stlink-v2",
                "request": "launch",
                "type": "cortex-debug",
                "cwd": "${workspaceRoot}",
                "servertype": "openocd",
                "executable": "${command:cmake.launchTargetPath}",
                "svdFile": "svd_file.svd",
                "configFiles": [
                    "/usr/share/openocd/scripts/interface/stlink-v2.cfg", 
                    "/usr/share/openocd/scripts/target/stm32f1x.cfg"
                ]
            },
```

It is bag or I a idiot?

haneefdm, 2021-11-29:
There have been issues but they were mostly because old openocd's never died,  causing new ones not to start. But could you try the latest release though? We are at V0.4.10.

haneefdm, 2021-11-29:
Also, do a `killall openocd` in a terminal shell to make sure there are no old copies of openocd running.

ilia19911, 2021-11-29:
Thanks for the answer, but it doesn't help. I see the stlink starts flashing but I still get the timeout error. 
Then I tried killall openocd but openocd was already killed

haneefdm, 2021-11-29:
And you are using 0.4.10? It would help if you post the contents of the 'Debug Console' and the gdb-server terminal windows. The error message says it could not start openocd and times out. But you also say it starts flashing the device. If it could not start then flashing cannot begin?!?! So, I am confused.

ilia19911, 2021-11-29:
I said wrong ( Stlink starts blinking, and not flashing. Yes, i use 0.4.10 as you said. Debug console contents
`
Please check TERMINAL tab (gdb-server) for output from /usr/bin/openocd
Launching server: "/usr/bin/openocd" "-c" "gdb_port 50000" "-c" "tcl_port 50001" "-c" "telnet_port 50002" "-s" "/home/ilia/projects/Copter_stand" "-f" "/usr/share/openocd/scripts/interface/stlink-v2.cfg" "-f" "/usr/share/openocd/scripts/target/stm32f1x.cfg"
`

and terminal contents 

Resuming connection to gdb server...
[2021-11-29T12:26:37.724Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v33 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.203125
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
[2021-11-29T12:26:47.727Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...

haneefdm, 2021-11-29:
Looks like the openocd either crashed or something else happened. If we cannot run OpenOCD, then we cannot debug. You can see the exact command-line used in the Debug Console. In a new Linux Terminal window, can you run the same command in the same directory?

```sh
"/usr/bin/openocd" "-c" "gdb_port 50000" "-c" "tcl_port 50001" "-c" "telnet_port 50002" "-s" "/home/ilia/projects/Copter_stand" "-f" "/usr/share/openocd/scripts/interface/stlink-v2.cfg" "-f" "/usr/share/openocd/scripts/target/stm32f1x.cfg"
```

We have to debug that first. This does not belong to us and has not changed due to us. We do rely on this. Perhaps the wrong config files for the board?

ilia19911, 2021-11-29:
I am trying to run openocd with this line and everything works fine. And openocd similarly runs in version 0.4.10, but I guess it can't connect to debug. All my configs are in my first post. Maybe there are other settings for debugging cortex with openocd?

haneefdm, 2021-11-29:
Does the output in the VSCode Terminal look identical to the one in the Terminal Window when you launched the OpenOCD manually? What I don't see is Openocd opening TCP ports to talk to. You should see something like

```
Listening on port 50000
```
If you don't see this coming from openocd, then gdb cannot talk to OpenOCD.

There is also a file `/tmp/cortex-debug-server-exiting.log` where you will see messages about a gdb-server quitting. Please note that openocd should start and not quit until we end the debug session. If it is quitting early for some reason, we have to find out why

ilia19911, 2021-11-29:
I tried start openocd manually, like this 
"/usr/bin/openocd" "-c" "gdb_port 50006" "-c" "tcl_port 50007" "-c" "telnet_port 50008" "-s" "/home/ilia/projects/Copter_stand" "-f" "/usr/share/openocd/scripts/interface/stlink-v2.cfg" "-f" "/usr/share/openocd/scripts/target/stm32f1x.cfg"

and then openocd threw this

Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v33 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.209902
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints

and then I started gdb-multiarch (not toolchain gdb, can it problem?)

gdb-multiarch
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb)

and then in gdb i insered target remote localhost:50009 (50009 because this socket was signed in openocd) and gdb threw
(gdb) target remote localhost:50009
Remote debugging using localhost:50009
warning: Architecture rejected target-supplied description
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
Truncated register 16 in remote 'g' packet
(gdb)

and in openocd adds this

Info : accepting 'gdb' connection on tcp/50009
Info : device id = 0x20036410
Info : flash size = 128kbytes
undefined debug reason 7 - target needs reset
Info : dropped 'gdb' connection

I can't find out  line like 
Listening on port 50009

but I saw it before.
and if I try start gdb with another socket, then a get error like in top of my issue

ilia19911, 2021-11-29:
can be proplem with wrong socket number in gdb command line ?

haneefdm, 2021-11-29:
OpenOCD is our first problem. Yes, we HAVE to see something like 'Listening on port...`. Are you sure you are using the right version of OpenoCD for STLink They ship a modified version with STMCube tools. That is the version I recommend you use.

If OpenOCD starts properly and shows you that message, only then you can start gdb (in a different terminal as the first one will be busy with OpenOCD) and it has to be the ARMs version of gdb and not the x86 version. Please make sure you are using the ARMs version.
https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

The socket numbers have to match between openocd and gdb. If you are using the non-default ones starting from 50000, then chances are you have other openocds running which will prevent any new openocds -- even with new socket numbers because the USB is busy

ilia19911, 2021-11-29:
i use 0.10.0 version
[image]
can you tell me which version i can use? 

on linux i can use gdb-multiarch instead of gdb. If it didn't work correctly, then I wouldn't be able to debug. And I normally debug in the old version
https://stackoverflow.com/questions/25143757/arm-linux-gnueabihf-gdb-versus-gdb-multiarch

Linux cannot stop the socket in time, so if I hard stop debugging, the socket stays for a few minutes and openocd opens the next socket. I cannot control this. and this does not mean that the debugger is busy with another process

haneefdm, 2021-11-29:
Unfortunately, 0.10.0 means nothing as there are a dozen of them distributed by various silicon vendors and they all have the same version number. If you install STMCube32 tools, you will find a copy of OpenOCD in there. This is what ST's Eclipse uses.

https://github.com/STMicroelectronics/OpenOCD

Building from sources is non-trivial so I would look for a precompiled version in the ST install directories.

Openocd does not open anything that we don't tell it. We are the ones that find which ports are open and tell openocd to use those ports.

bohdan-tymkiv, 2021-11-30:
OpenOCD v0.10.0 is almost five years old. The "Listening on port %hu for %s connections" messages were added on Jan 13, 2018, this happened AFTER v0.10.0 was released (Jan 22, 2017), see https://review.openocd.org/c/openocd/+/4316
@ilia19911, please update the OpenOCD. Ideally, you should build it from sources. Since you are using Ubuntu, the process is straightforward.

closed 2021-11-30 by haneefdm