Marus/cortex-debug #547: Documentation improvement planning

PhilippHaefele, 2021-12-21:
Hi together,

i open this issue to plan, discuss, collect things for a improvement of the documentation.

I‘m using a issue because it‘s better for discussion etc.

First of all i collect old issues regarding documentation (please comment if there‘re any missing that are currently not documented in the wiki):

- #515
- #323
- #283
- #277
- #200 ->  debuggerArgs example?
- #192 + #115 -> Check if behaviour is already described enough in wiki
- #139
- #111
- #104
- #76
- #59
- #55 => Something to mention as every new user is confused about that
- #36
- #1

I‘m also haveing thoughts about following things:
- „teach“ users to use intelliSense
- Use debug_attributes.md as much as possible so we don‘t need to document things twice.
- extend IntelliSense examples (maybe some with RTT/SWO output/graphing)
- Restructure wiki -> we have a lot of common settings and we most likely only need special hints on single parameters (sets) and maybe some special examples (others should be part of IntelliSense)
- Cleanup issues and mark open ones with labels, so users and contributers can more easily search trough them
- Move OpenOCD build manual to a seperate  wiki page
- Add a what‘s new page to better inform users about changes e.g. https://github.com/alefragnani/vscode-whats-new

That should be my first thoughts. Maybe not complete but i will update this post when working on the documentation.

Happy to any feedback or input

Best regards
Philipp

haneefdm, 2021-12-22:
@PhilippHaefele That is quite a list. Must have taken quite a bit of work to compile that. Thank you. A couple of comments

I would like to add a "theory of operations" sort of page, where we describe how the extension actually works -- the body parts (gdb, gdb-server, VSCode, etc.) and the process of how `launch.json` is used (including pre, default/override, post steps), etc. Some users think we interact directly with the HW (like IAR, Keil, etc.). Pictures are needed for this one.

haneefdm, 2021-12-22:
There is a problem with reliance on debug_attributes.md. package.json is not a good way of documenting anything in detail. So, a one-line has to do and it does help with IntelliSense. And, its duplicated (have to type it twice for most options).

Btw, I updated #55, #111, #183, #115, closed several

Btw, I don't get notifications on changes in the Wiki. So, when you are done, I will a look at everything.

Btw, if you see in the latest ChangeLog, we now support `runToEntryPoint` (sp?) for a Restart/Reset if setting a temporary breakpoint succeeds

hongshui3000, 2021-12-24:
> `runToEntryPoint`

When I reach the program runToEntryPoint position (my reset handler), I press the reset button again, and the program will run instead of still in the runToEntryPoint position

haneefdm, 2021-12-24:
@hongshui3000 Maybe I didn't understand your question.

After a reset the debugger may be showing a stale (old) stack-trace. This is what is reported by gdb. We report whatever location your gdb-server/gdb is showing. If you single-step however, things will look fine. We are trying to find a fix for this but it does not work across different types of gdb-servers and hardware.

This thread is not for asking questions like these though. I wish we had a Forum.

haneefdm, 2021-12-24:
@PhilippHaefele FYI: I started a new page as I am implementing the feature. So, things are in flux.

https://github.com/Marus/cortex-debug/wiki/Multi-core-debugging

haneefdm, 2022-01-05:
@PhilippHaefele Thanks again for all your help. I never imagined seeing such low numbers for open issues. There is now a new topic and I hope people will like it. I will push it out to a preview release soon.

https://github.com/Marus/cortex-debug/wiki/Disassembly-Debugging

Can't go in reverse -- from disassembly to source -- and you are supposed to be able to. Not sure why it is not working. People probably don't even know you can do that.

The UI is still pretty rough -- you will find out as you use it more and more. Sometimes, all of a sudden you get a second disassembly view. But moving through the stack frame and seeing the registers and the disassembly and source all getting updated was cool (for me at least).

haneefdm, 2022-01-05:
I am thinking that maybe the Wiki could do with some organization. Right now, it is flat with an alphabetical list of topics. I am thinking we can just number them and they will at least fall into place (but still flat)

sbixl, 2022-01-05:
Hi together, maybe it would be a good solution to carry over the wiki contents into ReStructuredText in order to use Sphinx as documentation generator. This has the advantage, that changes in the documentation can be reviewed in parallel to the changes in the code because the documentation ist stored in the same git repository. The generated documentation can then be uploaded at [1].  If required we can use github CI to build and upload the documentation automatically. I already use sphinx for documentation and I am very satisfied with the output. What do you think? 

 [1] https://readthedocs.org/

PhilippHaefele, 2022-01-05:
Hi,
didn't got that in mind for Cortex-Debug until now (as wiki was already there).
Had a short look and i really like the version feature of Read the Docs, so users are able to get older documentation versions when needed. Also markdown should work as input.

Downside of having the documentation in this (or separate) repo will be that someone needs to review changes made by others. Not that bad, but still something to keep in mind.

I like the idea in general, but i'm not 100% sure if we should do it like this.

@haneefdm @Marus What do you think about the suggestion from sbixl?

@sbixl Thanks for your input

haneefdm, 2022-01-05:
Thanks, I will look into `Sphinx `. @Marus will probably chime in. Where will the resulting docs hosted? Something like github.io?

Feels like stuff like `Sphinx` is more oriented towards API documentation rather than end-user users' manual kind of thing. Not saying you can't do both.

Caution: I don't own this repo. @Marus does.

sbixl, 2022-01-05:
The resulting docs can (if you want) hosted at [Read the Docs](https://readthedocs.org/) but you can als host it by using github.io. The benefit of using the first method is the possibility to access different versions of the documentation. If using github.io AFAIK, you only have access to the latest documentation.

Regarding the topic end-users users' manual while using Sphinx, I think it is good to show this with an example:

[1] Hosted documentation: https://bob-build-tool.readthedocs.io/en/latest/index.html
[2] Source Repository: https://github.com/BobBuildTool/bob

If you have a look in [2] you will find a "doc" folder where the complete documentation is included.

haneefdm, 2022-01-28:
Hi @PhilippHaefele, I added a chapter for an Overview of the debugging process and a pointer to it on the Home page. Let me know what you think. It was large enough that I downloaded I made a clone of the Wiki to edit.

https://github.com/Marus/cortex-debug/wiki/Overview

Let me know what you think, feel free to edit. I know have to cross-reference that page with other topics and there is some content missing.

haneefdm, 2022-01-28:
Actually, do not edit the Overiew page yet. SInce everything is long lines, merging can be difficult.

haneefdm, 2022-03-20:
About the following

>  Change Link for prebuild OpenOCD binaries to https://github.com/xpack-dev-tools/openocd-xpack/releases/ (at least for Linux and MacOS)

Initially, I thought that was great. But no. Already had to deal with a couple of users running into xPack version of OpenOCD which is not the same as what ST ships and switching to ST latest made problems go away. But we wasted a day or two trying to figure out why. Unlike any other gdb server, OpenOCD remains a problem. If a chip vendor ships it -- for sure use NOTHING but that.

haneefdm, 2022-04-09:
Hi @PhilippHaefele I started on a new page

https://github.com/Marus/cortex-debug/wiki/Toubleshooting

Frankly, I am inundated by a lot of issues related to OpenOCD/gdb installations and, I feel like there is not enough self help or people are not even trying to help themselves.

The page above is not done right, I know. I wanted to get some material in there first. Now I am thinking, I should have done it backwards.

pwnorbitals, 2022-08-26:
https://app.bountysource.com/issues/104525752-documentation-improvement-planning