Marus/cortex-debug #549: SWO ITM monitoring issue.

Ehsan2754, 2021-12-22:
Hi , I build my .elf target by following [Makefile](https://github.com/Ehsan2754/RTK-DW1000/blob/main/Makefile). I monitor the ITM0 on my VSCODE and I receive nothing. [ using STLINK and OPENOCD] please make a clear tutorial on how to config the extension for on STM32 / openocd to view the graph or have an SWO terminal . 
here is my Launch.json configuration: 
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "name":"Cortex Debug",
            "preLaunchTask": "Flash", 
            "preLaunchCommands": ["monitor arm semihosting enable"],
            "cwd": "${workspaceRoot}",
            "executable": "./build/RTK-DW1000.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "device": "STM32F103C8",
            "configFiles": [
                "interface/stlink.cfg",
                "target/stm32f1x.cfg"
            ],
            "svdFile": "./STM32F103.svd",
            "interface": "swd",
            "swoConfig":{
                "enabled":true,
                "source":"probe",
                "swoFrequency": 2000000,
                "cpuFrequency":72000000,
                "decoders": [
                    {
                        "port": 0,
                        "type": "console",
                        "label": "SWO output",
                        "encoding":"ascii"
                    }
                ]
            }
                
        }
    ]
}
```
Thanks

haneefdm, 2021-12-22:
Just wondering. Do you see a tab in the `TERMINAL` section and not find a window for 'SWO output'?

We try to enable SWO by setting the ITM (and other) registers but recently, vendors have been using non-default base addresses, so what we do might not work. You can also do this using the OpenOCD script/commands yourself.

You can see the gdb macros/functions used in the extension installation directory, but here is a copy online.

https://github.com/Marus/cortex-debug/blob/master/support/gdbsupport.init

You can see how those functions are used here...

https://github.com/Marus/cortex-debug/blob/88b69576624a8211f958947f39c7a75aaf852d44/src/openocd.ts#L128

Unlike RTT, SWO is super complicated. All the planets have to align. Even if most of the commands are right, the clock divider can be wrong. The built-in stuff may work for older devices but the newer devices are very different.

In the future, we may have to dump the existing auto-SWO setup and have users provide all critical info (the hard-coded stuff does not work). If you know how it works for your device, you can make your own gdbsupport.init using ours as a template.

haneefdm, 2021-12-22:
There is another failure mechanism, this following OpenOCD command can fail or not do the intended thing.

https://github.com/Marus/cortex-debug/blob/88b69576624a8211f958947f39c7a75aaf852d44/src/openocd.ts#L201

I wish I could fix this but it is a bit beyond my expertise and need help.

haneefdm, 2021-12-22:
STLink is a new addition and SWO support has not been added yet. I don't want to add that until we have a better solution for setup.

In the past, I just wrote the FW and added it to my project to enable SWO. You can find it here (look for SWO_Init(...)) It works provided you have proper CMSIS headers that define the base addresses correctly (I found issues here as well where the addresses were wrong and I had to fix them in my FW). This overrides whatever this extension does.

https://mcuoneclipse.com/2016/10/17/tutorial-using-single-wire-output-swo-with-arm-cortex-m-and-eclipse/

Ehsan2754, 2021-12-22:
Can I utilize my hardware such that I won't need to use SWO pin, but only classic SWD, by SWDIO and SICK pin for this purpose?

haneefdm, 2021-12-22:
That is what RTT does. But only OpenOCD and JLink support such tracing. It is much faster and easier to use.

https://github.com/Marus/cortex-debug/wiki/SEGGER-RTT-support

You cannot use SWD by yourself because OpenOCD/STLink would be using it. OpenOCD has an issue that there is a workaround for. I need to look it up if you want to try

Ehsan2754, 2021-12-22:
Well, although I have Jlink, for our laboratory, we made over 10pcs of 4pin STLinks with only 4pins of Gnd, clk, io and vcc. Thus I would be really greatful if you would help me to find out how to use printf by current setup of mine. Really appreciate your response.

haneefdm, 2021-12-22:
I am don't have a board with me, so I can give you the workaround in rough terms

* Download (install JLink SW if you have to and the RTT zip file contains the FW). It is free. Add the files to your project (not all of them, just the API ones and not the examples)
* Hopefully you can follow the RTT instructions in the link. In your main, include the RTT header file. RTT setup is similar to SWO setup in launch.json
* Place a print statement somewhere in main
* Start a debug session and go past the print statement (this initializes the RTT system or you can call the API to do it explicitly)
* In the Debug Console, enter `monitor rtt start`. Continue normally and you should see all your print statements in VSCode.

Ehsan2754, 2021-12-22:
But this method works for jlink only, could you possibly offer a method for stlink?

haneefdm, 2021-12-22:
No, it works for OpenOCD as well. Just -- that is how I did all the development.

There is an (semi) automated way to do the `monitor rtt start` but I need to be near board to try it out.

Once I am in a good place I will add these instructions to the Wiki. From their zip/tar file, add these to your FW project.

[image]

Then `#incllude "RTT/SEGGER_RTT.h"` in your main.c (of whichever .c file you want). Your FW should now be ready to use.

Note: The `RTT/SEGGER_RTT_ASM_ARMv7M.S` file is for GCC. There are others for other compilers

haneefdm, 2021-12-22:
RTT support was added to OpenOCD a few months ago.

Ehsan2754, 2021-12-22:
And after this I just can run my launch file and the printf would work accordingly.

haneefdm, 2021-12-22:
Sorry, Can't do it with STLink-GdbServer SW. But can use (OpenOCD SW + STLink-V2/V3 HW). Such a thing has to be supported by the SW that talks directly to the HW. We have no way to talk directly to the HW. We always go through GDB and a gdb-server

haneefdm, 2021-12-22:
Yes, your printf's will work just fine. You can even do getc() if you want.

Ehsan2754, 2021-12-22:
Wow that means scanf would also work?

haneefdm, 2021-12-22:
Yes sir! Our RTT terminal in VSCode is bidirectional. You can even output ANSI sequences for color coding. See code in the `Examples/Main_RTT_PrintfTest.c` in SEGGERs zip/tar file.

With input (getc()), there is a bit of a learning curve. Buffered vs. unbuffered, cooked vs. raw, etc.

Ehsan2754, 2021-12-22:
Wow. That's impressive. I gotta check this out tomorrow to close this issue. I would write a proper report which can be used as a tutorial for setting up this debug in wiki page of the project. Thanks for your response 🌹

Ehsan2754, 2021-12-23:
Hey Hannef. I'm afraid the RTT method doesn't work on STLINK. probably it's only meant to be used on jlink

haneefdm, 2021-12-23:
I tested it quite a bit with STLink, but there may be other issues. Here is my launch.json. Note that you need the proper version of OpenOCD. Stuff you download from internet is too old. I always rely on vendor provided OpenOCD if they provide one. I use ST's version for ST, Cypress' Version for Cypress, etc.

```JSON
            "cwd": "${workspaceRoot}",
            "executable": "./Debug/firststm.elf",
            "name": "Debug OpenOCD",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "device": "STM32F429ZI",
            // "showDevDebugOutput": "raw-only",
            "serverpath": "/Applications/STM32CubeIDE.app/Contents/Eclipse/plugins/com.st.stm32cube.ide.mcu.externaltools.openocd.macos64_2.0.0.202105311346/tools/bin/openocd",
            "searchDir": [
                "/Applications/STM32CubeIDE.app/Contents/Eclipse/plugins/com.st.stm32cube.ide.mcu.debug.openocd_2.0.0.202106290712/resources/openocd/st_scripts",
                "."
            ],
            "configFiles": [
                "firststm Debug OpenOCD.cfg"        // File created by STMCubeIDE, minor edits by me
            ],
            "rtos": "auto",
            "rttConfig": {
                "enabled": true,
                "address": "auto",
                "decoders": [
                    {
                        "label": "",
                        "port": 0,
                        "type": "console",
                        "iencoding": "ascii",
                    }
                ]
            },
            "swoConfig": {
                "enabled": false,
                "cpuFrequency": 168000000,
                "swoFrequency": 1500000,
                "source": "probe",
                "decoders": [
                    {
                        "type": "console",
                        "port": 0,
                        "label": "hdm",
                        "logfile": "/tmp/xyz.txt",
                        "showOnStartup": true,
                        "timestamp": false
                    }
                ],
            },
            "runToEntryPoint": "main"
```
Another thing I do differently is to use

haneefdm, 2021-12-23:
Since I am using the latest (pre-release) version of this extension, you should switch to that as well or I cannot debug/help you. You can always switch back. Go to the Extension section on the left side, find this extension, You will see a triangle with a green dot (meaning there is a pre-release available)
[image]

Select the extension and you will see a button to install the pre-release.
[image]

haneefdm, 2021-12-23:
You also have to tell me what you tried and what is not working. Any messages in the debug console, etc.

haneefdm, 2021-12-23:
There are a couple of other differences. Btw, I used the following in my FW. This ensures that the FW does compile and have RTT included
```C
  SEGGER_RTT_printf(0, "Hello from RTT\n");
```
What I noticed is that you are using a different way of configuring OpenOCD for STLink. That may work but I use a different method. I let STM32CubeIDE generate one for me (run the OpenOCD based debug at least once and this file will be generated. It works as is but I commented out a few lines for my testing purpose (lines begin with `#hdm`)
```TCL
# This is an NUCLEO-F429ZI board with a single STM32F429ZITx chip
#
# Generated by STM32CubeIDE
# Take care that such file, as generated, may be overridden without any early notice. Please have a look to debug launch configuration setup(s)

source [find interface/stlink-dap.cfg]

set WORKAREASIZE 0x8000

transport select "dapdirect_swd"

set CHIPNAME STM32F429ZITx
set BOARDNAME NUCLEO-F429ZI

# Enable debug when in low power modes
set ENABLE_LOW_POWER 1

# Stop Watchdog counters when halt
set STOP_WATCHDOG 1

# STlink Debug clock frequency
set CLOCK_FREQ 8000

# Reset configuration
# use hardware reset, connect under reset
# connect_assert_srst needed if low power mode application running (WFI...)
reset_config srst_only srst_nogate connect_assert_srst
set CONNECT_UNDER_RESET 1
set CORE_RESET 0

# ACCESS PORT NUMBER
set AP_NUM 0
# GDB PORT
#hdm set GDB_PORT 3333

# BCTM CPU variables

source [find target/stm32f4x.cfg]

# SWV trace
#hdm set USE_SWO 0
#hdm set swv_cmd "-protocol uart -output :61235 -traceclk 168000000"
#hdm source [find board/swv.tcl]
```

haneefdm, 2022-01-04:
You can try the new preview version of SWO support. It is available as a pre-release from the Market place.

You can customize how SWO works as well.

closed 2022-01-04 by haneefdm

olisnr, 2024-01-19:
if somebody is interested, ambiq's apollo2 works for me with folowing setings:

      {
        "cwd": "${workspaceFolder}",
        "executable": "boards/apollo2_fiber_probe/examples/adc_vbatt/gcc/bin/adc_vbatt.axf",
        "name": "Debug ADC with JLink",
        "request": "launch",
        "type": "cortex-debug",
        "device": "AMAPH1KK-KBR",
        "interface": "swd",
        "runToEntryPoint": "main",
        "showDevDebugOutput": "none",
        "servertype": "jlink",
        "svdFile": "pack/SVD/apollo2.svd",
        "swoConfig": {
            "enabled":true,
            "source":"probe",
            "swoFrequency": 1000000,
            "cpuFrequency":96000000,
            "decoders": [
                {
                    "port": 0,
                    "type": "console",
                    "label": "SWO output",
                    "encoding":"ascii"
                }
            ]
        }