Marus/cortex-debug #551: Is there any plan to support mem and register write window？

hongshui3000, 2021-12-24:
Is there any plan to support mem and register write window？

PhilippHaefele, 2021-12-24:
Yes this is at least on the TODO list (https://github.com/Marus/cortex-debug/blob/master/TODO.md).

But not sure when @haneefdm will work on that

haneefdm, 2021-12-24:
This may be something some people may not know. Using the Debug Console to interact directly with GDB. The bottom part is the input section and you can any gdb command so long as you don't drastically change the debugger state (like killing it). You can do the following
```
set $r0 = 0xdeadbeef
set mystruct.myfield = 23
print mystruct.myfield
call myFuncion(myArg)
```
My syntax may be wrong at places. Note: All register names a prefixed with a `$` and are pre-defined by gdb

[image]

This is what we did when we didn't have good Visual debuggers and still do since they don't support every possible thing I want to do.

That bottom input part of the window is called `REPL` (Read-Eval-Print Loop) 
https://code.visualstudio.com/docs/editor/debugging#_debug-console-repl

Some debuggers (Python, JavaScript, etc.)

closed 2021-12-24 by haneefdm