Marus/cortex-debug #555: How to insert hardware breakpoints？

hongshui3000, 2021-12-27:
I use the following configuration，But the gui can still only set software breakpoints。Unless I use the gdb command hbreak,
launch.json
```
            "hardwareBreakpoints": {
                "require": true,
                "limit": 2
             }
```

PhilippHaefele, 2021-12-28:
Hi,
can you write a little bit more about your setup please. Complete launch configuration would be nice.

As the consideration what kind of breakpoints is used when issuing a GDB `-break-insert` is upto the GDBServer (when not using the `-h` option), we really need that information to help you.

haneefdm, 2021-12-29:
Almost always what you get is a HW breakpoint when your program is in FLASH. gdb doesn't even know how else to set a breakpoint. GDB-servers also do not mess with FLASH on a byte basis. GDB knows with help with the gdb-server the memory types your devices have and does the right thing.

If your program is running from RAM, then that is different. So, what is the question here?

Also, the JSON snippet you posted in your `launch.json`? That is probably for the Microsoft debugger. Not ours. But you are welcome to use their debugger.

hongshui3000, 2021-12-29:
launch.json：
```
    {
            "name": "Openocd Attach",
            "type": "cortex-debug",
            "request": "attach",
            "servertype": "openocd",
            "cwd": "${workspaceRoot}",
            "executable": "./build/zephyr/zephyr.elf",
            "configFiles":[
                "board/xtensa-x-jlink.cfg",
                "board/xtensa-core-x.cfg"
            ],
            "debuggerArgs":[
              "--xtensa-core=hifi4vfp7",
              "-x", "${workspaceFolder}/ZCAN-sdk/scripts/gdb.init"
            ],
            "gdbPath":"C:\\usr\\xtensa\\XtDevTools\\install\\tools\\RI-2021.6-win32\\XtensaTools\\bin\\xt-gdb.exe",
            //"preLaunchTask": "build",
            "runToEntryPoint": "main",
            "hardwareBreakpoints": {
                "require": true,
                "limit": 2
             }
        }
```
Where should "hardwareBreakpoints" be placed? my program is in FLASH

haneefdm, 2021-12-29:
Oh, is this because you are using a non-ARM (Cortex) device?

It is unlikely we are going to add the -h breakpoints as it will interfere with which breakpoints go where and how they are allocated. If your program is in FLASH and the breakpoints are working, I bet your gdb and gdb-server is already doing the right thing. There is really no other (good) way for breakpoints to work when your code is in FLASH. How else would you think breakpoints would work? You can also check your debug HW debug registers. Here is what gdb reports for Cortex devices -- if we use the -h method, we will lose the `automatically using hardware breakpoints for read-only addresses` which would be super bad for 99% of our users and I don't want to duplicate the auto-behavior that gdb+gdb-server already do. You should address this with those teams.

[image]

I am very familiar with the MIEngine, btw, and have contributed there as well.

haneefdm, 2021-12-29:
Looking at your `launch.json` file, I can pretty much guarantee you are already getting HW breakpoints when/where needed. You are using OpenOCD AND JLink.

hongshui3000, 2021-12-29:
Okay, I understand what you said. I did use xtensa core instead of arm

haneefdm, 2021-12-29:
I have seen other debuggers for Xtensa and they use HW breakpoints when code is in FLASH automatically. No need for special handling.

I do have a question. Why do you think HW breakpoints are superior? They are expensive and not even any faster. If your code is in RAM (I frequently run out of RAM) then you are better off using SW breakpoints. And, gdb + openocd (both customized by ARM and your Silicon vendor) will do that for you.

hongshui3000, 2021-12-29:
I don't think hardware breakpoints are better, but my program is running in flash and I can only use hardware breakpoints. The gdb service  (xt-gdb) I am using is a commercial version provided by candence. I now suspect that this aspect has not been done very well.

haneefdm, 2021-12-29:
So, the breakpoints are not working for the code that is in FLASH?

hongshui3000, 2021-12-29:
yes

haneefdm, 2021-12-29:
Ohh... that is unfortunate. Let me think about this for a bit.

haneefdm, 2021-12-29:
What is the error you get (especially from gdb)? Is it a silent failure or do you actually see a failure reported by gdb

hongshui3000, 2021-12-29:
There is no failure, but the breakpoint does not stop

hongshui3000, 2021-12-29:
Another problem is that I set a breakpoint on the vscode gui, and the assembly code cannot set a breakpoint (there is no red dot). Is this related to vscode? C code is normal

haneefdm, 2021-12-29:
When you set a breakpoint (with `"showDevDebugOutput": true`) what do you see coming from gdb? Also, what does `info breakpoints` tell you after setting the breakpoint?. This will be very helpful for me.

VSCode does not know anything about assembly. We added some rudimentary support for it and a red dot should appear. We don't even know if we are looking at ARM code but we do care what the format for the disassembly (should like GNU/ARM assembly) so we can parse the addresses.

haneefdm, 2021-12-29:
> There is no failure, but the breakpoint does not stop

How do we know that we are executing this code? What if the PC never reached this place?

Also, did openocd report a problem? Since you are using JLink, have you tried using our extension with JLink? Maybe they have better support.

hongshui3000, 2021-12-29:
The log when I delete and insert a breakpoint
```
21-stack-list-frames --thread 1 0 2
GDB -> App: {"token":20,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["depth","3"]]}}
22-stack-list-frames --thread 1 0 2
GDB -> App: {"token":21,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["stack",[["frame",[["level","0"],["addr","0x0200192d"],["func","arch_cpu_idle"],["file","E:\\work\\MYMCU\\XMCU\\zephyr\\arch\\xtensa\\core\\cpu_idle.c"],["fullname","E:\\work\\MYMCU\\XMCU\\zephyr\\arch\\xtensa\\core\\cpu_idle.c"],["line","39"],["arch","xtensa"]]],["frame",[["level","1"],["addr","0x02003911"],["func","idle"],["file","E:\\work\\MYMCU\\XMCU\\zephyr\\include/kernel.h"],["fullname","E:\\work\\MYMCU\\XMCU\\zephyr\\include\\kernel.h"],["line","5503"],["arch","xtensa"]]],["frame",[["level","2"],["addr","0x02000a92"],["func","z_thread_entry"],["file","E:\\work\\MYMCU\\XMCU\\zephyr\\lib\\os\\thread_entry.c"],["fullname","E:\\work\\MYMCU\\XMCU\\zephyr\\lib\\os\\thread_entry.c"],["line","36"],["arch","xtensa"]]]]]]}}
GDB -> App: {"token":22,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["stack",[["frame",[["level","0"],["addr","0x0200192d"],["func","arch_cpu_idle"],["file","E:\\work\\MYMCU\\XMCU\\zephyr\\arch\\xtensa\\core\\cpu_idle.c"],["fullname","E:\\work\\MYMCU\\XMCU\\zephyr\\arch\\xtensa\\core\\cpu_idle.c"],["line","39"],["arch","xtensa"]]],["frame",[["level","1"],["addr","0x02003911"],["func","idle"],["file","E:\\work\\MYMCU\\XMCU\\zephyr\\include/kernel.h"],["fullname","E:\\work\\MYMCU\\XMCU\\zephyr\\include\\kernel.h"],["line","5503"],["arch","xtensa"]]],["frame",[["level","2"],["addr","0x02000a92"],["func","z_thread_entry"],["file","E:\\work\\MYMCU\\XMCU\\zephyr\\lib\\os\\thread_entry.c"],["fullname","E:\\work\\MYMCU\\XMCU\\zephyr\\lib\\os\\thread_entry.c"],["line","36"],["arch","xtensa"]]]]]]}}
23-stack-list-variables --thread 1 --frame 0 --simple-values
GDB -> App: {"token":23,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["variables",[]]]}}
24-break-insert "e:\\work\\MYMCU\\XMCU\\zephyr\\arch\\xtensa\\core\\cpu_idle.c:38"
GDB -> App: {"token":24,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["bkpt",[["number","1"],["type","breakpoint"],["disp","keep"],["enabled","y"],["addr","0x0200192a"],["func","arch_cpu_idle"],["file","E:\\work\\MYMCU\\XMCU\\zephyr\\arch\\xtensa\\core\\cpu_idle.c"],["fullname","E:\\work\\MYMCU\\XMCU\\zephyr\\arch\\xtensa\\core\\cpu_idle.c"],["line","38"],["thread-groups",["i1"]],["times","0"],["original-location","e:\\work\\MYMCU\\XMCU\\zephyr\\arch\\xtensa\\core\\cpu_idle.c:38"]]]]}}
25-break-delete 1
GDB -> App: {"token":25,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
26-break-insert "e:\\work\\MYMCU\\XMCU\\zephyr\\arch\\xtensa\\core\\cpu_idle.c:38"
GDB -> App: {"token":26,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["bkpt",[["number","2"],["type","breakpoint"],["disp","keep"],["enabled","y"],["addr","0x0200192a"],["func","arch_cpu_idle"],["file","E:\\work\\MYMCU\\XMCU\\zephyr\\arch\\xtensa\\core\\cpu_idle.c"],["fullname","E:\\work\\MYMCU\\XMCU\\zephyr\\arch\\xtensa\\core\\cpu_idle.c"],["line","38"],["thread-groups",["i1"]],["times","0"],["original-location","e:\\work\\MYMCU\\XMCU\\zephyr\\arch\\xtensa\\core\\cpu_idle.c:38"]]]]}}
27-data-evaluate-expression --thread 1 --frame 0 NOP128
GDB -> App: {"token":27,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","No symbol \"NOP128\" in current context."]]}}
hover: No symbol "NOP128" in current context. (from data-evaluate-expression --thread 1 --frame 0 NOP128)
```

hongshui3000, 2021-12-29:
assembly
[image]
c
[image]

haneefdm, 2021-12-29:
The assembly problem is because VSCode is not seeing that file as Code. I am sure there is an extension that makes it look like code. We don't do code editing.

We declare support with VSCode for languages [c, cpp, rust, asm, arm]. Note that these are not file extensions. Then you can configure certain file extensions to be for certain languages.

For instance, for ARM assembly, I use the following extension(s)

https://marketplace.visualstudio.com/items?itemName=dan-c-underwood.arm
And of course the MS cpptools extension.

The ARM extension may also work for you as what you have is GNU assembler format perhaps

hongshui3000, 2021-12-29:
> The assembly problem is because VSCode is not seeing that file as Code. I am sure there is an extension that makes it look like code. We don't do code editing.
>
> We declare support with VSCode for [...]

Thanks

haneefdm, 2022-01-04:
I don't want to entertain this. This should be taken care of between gdb and the gdb-server. I am closing this.

closed 2022-01-04 by haneefdm

mao-china, 2022-05-20:
@haneefdm 
when i add hardwareBreakpoints option in my launch.json, it shows : property hardwareBreakpoints is not allowed.
i don't know how to deal with it, would you give me some suggestions?
thanks!

vscode version:v1.67.2
cpptools version:v1.9.8
cortex-debug version:v1.4.4

my launch.json is:

haneefdm, 2022-05-20:
All breakpoints are HW breakpoints if in FLASH. GDB automatically figures it out depending on FLASH or RAM.

We don't have a property called `hardwareBreakpoints`. That is for some other debugger. Not Cortex-Debug. You may be using the wrong debugger, but you already know that :-)

https://github.com/microsoft/vscode-cpptools/issues/4745

mao-china, 2022-05-20:
Thanks for answer, but i think the option HW breakpoints is necessary, which is added in cpptools.
Although gdb can automatically figures it out, in some cases, it may works wrong.

Im my project, arm cortex-m7 core with cache, code running on extended sram, gdb uses sw breakpoints in default.
The program code at the breakpoint address is replaced with a bkpt instruction.
The debugger will send command to dap(arm coresight port), modify instruction at the breakpoint address in sram, but the cache don't know sram is modified. So the instruction cache is not update, m7 core can't fetch the bkpt instruction actually.  

I think force hw breakpoints is the best way to solve it. Otherwise cache must be disabled during debug.

haneefdm, 2022-05-20:
I see your point on cache coherency. But what cpptools did does not work. FYI, it is not cpptools, it is MIEngine that does that. Problem I have is that there is no guarantee which breakpoints get which type of breakpoint as there is no UI. And they change (randomly) through out your debug session. If you modify/add/delete one breakpoint, it can affect a whole  bunch of breakpoints in that file. Before anyone says first come first serve, there is no such thing. In VSCode these are random, and batched on a per file basis for line breakpoints and all function breakpoints are lumped, then you also have logpoints.

This is what I cannot deal with since there is no UI or known determinism or organization of breakpoints.

Have you considered adding your own breakpoints via gdb? Sure they are not managed but they should work. And if you add these as part of the startup script, they will always be the first ones to be created?

mao-china, 2022-05-20:
Hi,  haneef：

I understand your difficulties. From my perspective, I don't think you need to do so complicated management. Why do you do so complicated management, only to provide more breakpoints? In the case of problems (such as my cache consistency problem), being able to use is what I am concerned about.

We can make things simple and provide an option for all gdb breakpoints to become hardware breakpoints.
Others may share the same opinion.

1. We know that if the code is running on the flash, only hardware breakpoints can be set.
Then I put the code into the ram to run and force the hardware breakpoint, which is exactly the same as when the code is run on the flash. I do not think that will result in worse results.
2. For users, setting hardware breakpoints for some scenarios that may cause problems, such as my cache consistency problem, is a very effective way to solve the problem. In addition, I think this method may solve more problem scenarios.
3. On some chips, of course, there are few hardware breakpoints. But better than nothing.

In addition, adding breakpoints through gdb is a way to solve the problem, but I think it is not very convenient, even through scripts.

I use openocd as gdb server and find the gdb_breakpoint_override command of openocd. Openocd can convert all gdb breakpoints into hardware breakpoints. I add this command to the openocd.cfg, and my project runs very well even cache enabled. However, I am not sure whether jlink and pyocd can do the same. I think it is better to do it on gdb.

@haneefdm

haneefdm, 2022-05-20:
Thanks @mao-china

Important points, but I will play a bit of devil's advocate here,

Question: Do you seriously think that just copying what cpptools/MIEngine did makes sense? 

> I don't think you need to do so complicated management

Not sure you understand how many types of breakpoints exist and how they are managed by VSCode. Once I push something out in a release, there will be tons of objections on why things are not intuitive, simple, not the way I expected, etc. No one to help me, and I will be stuck with this.

> We can make things simple and provide an option for all gdb breakpoints to become hardware breakpoints.

I don't think it is that simple. It is an all-or-nothing hardcoded approach. I am not averse to it but I am not comfortable.

> In addition, adding breakpoints through gdb is a way to solve the problem, but I think it is not very convenient, even through scripts.

Why? It is a power user thing. Why would you need scripts? Just simple gdb commands

> I use openocd as gdb server and find the gdb_breakpoint_override command of openocd

So, why not ask other gdb-servers to provide the same options? If they already know it, it is trivial for them. Something like pyOCD, I/you can even submit a PR. It is eventually a gdb-server feature. I know one counterargument -- let gdb handle it because it has a HW option. Not good enough because we are in the middle of VSCode and gdb but most people think we own the whole chain including UX

If you are serious and detached from your immediate needs, let's discuss; and based on what I saw, I think you will.

mao-china, 2022-05-20:
Thanks for your reply. @haneefdm 

First of all, I want to apologize to you, I'm not good at developing vscode plugins, and I don't know how many types of breakpoints there are in vscode, you are an expert on this, I'm an embedded software engineer.

And my English is not very good, some words are not very accurate, they are translated by google, if I offend you, please forgive me!

> Why? It is a power user thing. Why would you need scripts? Just simple gdb commands

Yes, with gdb commands, everything can be done.But I have to repeatedly enter commands related to breakpoints, such as break mian.c:100/info breakpoints/delete 1/enable/disable/etcs, these things can obviously be done with the click of a mouse.
These can be done automatically by vscode and plugins.If you use the command line to interact, why use vscode debug, and there are obviously better solutions to this problem.

> I don't think it is that simple. It is an all-or-nothing hardcoded approach. I am not averse to it but I am not comfortable.

It might make you uncomfortable, if you have the same problem as me, if the forced hardware breakpoint is not applicable, do you have any good solution? The breakpoint command you said through the gdb console, but don't forget, it must also use hardware breakpoints, and it is manual.

> So, why not ask other gdb-servers to provide the same options? If they already know it, it is trivial for them. Something like pyOCD, I/you can even submit a PR. It is eventually a gdb-server [...]

I still think that forcing hardware breakpoints is the most efficient way to solve this kind of problem wherever it is implemented.
You're an expert on this, and I'm sure you're capable of judging where to implement forced hardware breakpoints, in cortex-debug or in gdb-servers.
Even with forced hardware breakpoints implemented in gdb-server, I think the plugin needs to have a way to tell gdb-server to turn it on, I'm not sure if all gdb-servers can use configuration files like openocd.
If cpptools is already implemented, why can't we refer to it?

good luck.

haneefdm, 2022-05-20:
>  if you have the same problem as me, if the forced hardware breakpoint is not applicable, do you have any good solution?

Yes I do. You can use launch.json to specify `preLaunchCommands` to specify any set of gdb commands (see `hbreak`). There are so many other ways to customize as well. See https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

https://sourceware.org/gdb/onlinedocs/gdb/Set-Breaks.html

I don't think I said enter breakpoints via the Debug Console.

Basically, I am not comfortable with what MS did. I have to think more about this and until then, you have your workarounds that are pretty good in my opinion. Randomly assigning the 6 HW bkpts to a set of VSCode breakpoints is not my cup of tea. That will be the first bug that will be submitted by users saying how stupid I am for doing it that way.

I do suggest getting familiar with gdb capabilities because all we are (cpptools too) is a frontend to gdb. But VSCode dicates user experience wherein and that is something we cannot change. So, I have to think in that direction as well to see if I can submit an idea (and write code) to VSCode for an enhancement.

If GDB cannot do something neither can we.

> Even with forced hardware breakpoints implemented in gdb-server, I think the plugin needs to have a way to tell gdb-server to turn it on, I'm not sure if all gdb-servers can use configuration files like openocd.

Not correct. All gdb-servers must implement the GDB remote interface spec. That includes the working of breakpoints and specifically `hbreak`. Even if we did implement something it will always be via gdb. Not via the gdb-server. And it will not be all or nothing which is what cpptools did. We never talk to the gdb-server after program starts.

mao-china, 2022-05-21:
Hi, haneef:

Thanks a lot for reply, i see what you mean.

> Yes I do. You can use launch.json to specify preLaunchCommands to specify any set of gdb commands (see hbreak). There are so many other ways to customize as well. See https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md
https://sourceware.org/gdb/onlinedocs/gdb/Set-Breaks.html
I don't think I said enter breakpoints via the Debug Console.

I know that some gdb commands can be customized through postLaunchCommands. My code is also loaded to ram through postLaunchCommand and jump to it. However, during debugging, I can only send commands via the Debug Console, for example: add new breakpoints, delete it, enable or disable it, etcs. If I do not do so, vscode will only set software breakpoints.

> Basically, I am not comfortable with what MS did. I have to think more about this and until then, you have your workarounds that are pretty good in my opinion. Randomly assigning the 6 HW bkpts to a [...]

Can we consider limiting the number of breakpoints after enabling forced hardware breakpoints? If a similar problem occurs, you need to enable the forced hardware breakpoint. It is meaningless to set the number of breakpoints that exceed the 6 hw breakpoints. Of course, this is only a suggestion. After all, I am not an expert in this aspect. I may not know enough about some places.

> Not correct. All gdb-servers must implement the GDB remote interface spec. That includes the working of breakpoints and specifically hbreak. Even if we did implement something it will always be via [...]

I also read the gdb manual, but did not find any command for setting forced hardware breakpoints . If using gdb, I understand that there are only two modes:
1. Use hbreak instead of break
2. Run the monitor command to send `gdb_breakpoint_override hard` to my gdb-server (`gdb_breakpoint_override` is a command of openocd. Different gdb-server may vary).
Am I right?