Marus/cortex-debug #556: SWO ITM no console, junk file created

g8ecj, 2021-12-28:
Using what (as far as I can see) a pretty standard setup in launch.json but I get no console, just a file called ':50003' that is full of binary data.
`         "interface": "swd",
         "swoConfig":{
             "enabled":true,
             "source":"probe",
             "swoFrequency":2000000,
             "cpuFrequency":72000000,
             "decoders": [
                 {
                     "port": 0,
                     "label": "SWO output",
                     "type": "console"
                 }
             ]
`
The command line generated to GDB seems to be as follows (I assume this is where the magic number for a filename comes from)

`Launching server: "/home/g8ecj/opt/xPacks/openocd/0.10.0-13/bin/openocd" "-c" "gdb_port 50000" "-c" "tcl_port 50001" "-c" "telnet_port 50002" "-s" "/home/g8ecj/work/src/WxSTM32Rx" "-f" "board/fake_cs32f1x.cfg" "-c" "tpiu config internal :50003 uart off 72000000 2000000"`

haneefdm, 2021-12-28:
Do you see any output from OpenOCD complaining about the `tpiu` command? See in the Terminal Tab (gdb-server). We are still using the old-style tpiu to enable SWO in OpenOCD. :50003 mean use TCP port 50003 -- that was the convention used for a very long time. What we were doing was still compatible until very recently. From openocd doc
```
- :port – configure TPIU/SWO and debug adapter to gather trace data, open a TCP server at port port and send the trace data to each connected client;
```

With the new syntax/commands OpenOCD want us to use, not only do we have compatibility issues, but we will need a lot more information fro the user to do the setup... to the point that you may be better off doing in the config file yourself.

We don't have any TPIU/SWO/DAP experts wrt. OpenOCD so, any help is appreciated. Iknow how it works in the device, just don't what OpenOCD configuration is needed (and one that does not conflict what the chip vendor provides).

haneefdm, 2021-12-28:
And, yes the output of OpenOCD for SWO is not human readable. It needs pretty complicated SW to figure out the frames and convert the info.

g8ecj, 2021-12-28:
Thanks for the quick response.
After more digging I found "ARM® v7-M ArchitectureReference Manual" which details the packet formats for ITM - still not worked out how TPIU wraps these up but at least I can read the data on a logic analyser now ;)
I guess this is one for the back-burner
Cheers

closed 2021-12-28 by g8ecj

haneefdm, 2021-12-29:
Btw, you may simply be running a very old version of OpenOCD. Not sure when TCP Port support was added but that should work for you. Many people (including me) are using it.

I would try building it from source. Looks like it was added just a year ago and a lot of Lunux/Mac packages are old.

g8ecj, 2022-01-03:
I've now updated openocd (0.11.0-3) and gdb (10.3.1.2-2.3) to the latest xpack versions [https://xpack.github.io/](url) and everything now works as expected. Again, many thanks (1 bug already found and fixed ;))

haneefdm, 2022-01-03:
Nice! Thanks for the update