Marus/cortex-debug #557: Debugging Renesas RA4M2

RolfNoot, 2021-12-29:
Hi!

I am currently working on Renesas RA4M2 and unfortunately only Windows/Eclipse is officially supported. I am trying to setup a VS Code environment (on MAC) to work with the EK-RA4M2 (development board with on-board JLink debugger).

I noticed that Renesas uses quite a quirky debugger setup (non default windows-only gdb server: e2-server-gdb and gdb 7.8.2 compiled for arm-none-eabi).

So I did the following:
- compiled gdb 7.8.2 for MAC
- run the e2-server-gdb on a windows machine with the board connected
- remote connect on the mac with gdb 7.8.2 to the e2-server-gdb on windows
- send init, load, post-load, reset, continue and interrupt commands successfully (after inspecting the eclipse commands).

So far, so good.

Now I tried to implement the gdb setup and communication using the Cortex-Debug extension but I cannot get it working. For some reason the gdb init differs from the default one and the debugger.

I cannot get to:
- send custom commands to gdb before the extension sends the 'remote-connect' command
- send custom commands to gdb directly after the extension sends the 'remote-connect' command

I am aiming to get the following commands to gdb:

```
// Init
"-environment-cd " + envLoc + "\r\n" +
"-gdb-set target-async on\r\n" +
"-gdb-set non-stop on\r\n" +
"symbol-file " + elfFile + "\r\n" +
"target extended-remote localhost:52222\r\n" +
"monitor set_target,R7FA4M2AD\r\n" +
"monitor is_target_connected"

// Load
"-thread-select 1\r\n" +
"show remote memory-map-packet\r\n" +
"monitor configuration_complete\r\n" +
"monitor prg_download_start_on_connect\r\n" +
"exec-file " + elfFile + "\r\n" +
 "load " + elfFile + " 0x0\r\n" +
"set remotetimeout 300\r\n" +
"monitor prg_download_end"

// Post Load
"set remotetimeout 10\r\n" +
"monitor enable_stopped_notify_on_connect\r\n" +
"monitor enable_execute_on_connect\r\n" +
"-thread-info 1\r\n" +
"set backtrace limit 11\r\n" +
"-break-insert -t -h -f main\r\n" +
"-gdb-set --thread-group i1 language c\r\n" +
"-interpreter-exec --thread-group i1 console \"p/x (char)-1\"\r\n" +
"-gdb-set --thread-group i1 language auto\r\n" +
"-interpreter-exec --thread-group i1 console \"show endian\"\r\n" +
"monitor ignore_continue_during_connect"

// Reset
"monitor reset"

// Continue
 "-exec-continue --thread 1"

// Interrupt
 "-exec-interrupt --thread 1"
```
The 'init', 'load', 'post load' and 'reset' commands have to be send in this order to connect well.

I tried to add commands to the preLaunchCommands and overrideLaunchCommands but no success so far. It might be that gdb is different to arm-none-eabi-gdb. I get the following output:

```
Reading symbols from '/Volumes/Data/VScode/Renesas/quickstart_ek_ra4m2_ep/build/quickstart_ek_ra4m2_ep.elf'
Finished reading symbols
Launching GDB: "/Volumes/Data/VScode/Renesas/quickstart_ek_ra4m2_ep/script/gdb" "-q" "--interpreter=mi2" "-data-directory=/Users/rolf/Downloads/gdb-7.8.2/gdb/data-directory"
 "/Volumes/Data/VScode/Renesas/quickstart_ek_ra4m2_ep/build/quickstart_ek_ra4m2_ep.elf"1-gdb-set target-async on
2-interpreter-exec console "source /Users/rolf/.vscode/extensions/marus25.cortex-debug-0.4.10/support/gdbsupport.init"
3-interpreter-exec console "set print demangle on"
4-interpreter-exec console "set print asm-demangle on"
5-target-select extended-remote 192.168.1.243:52222
6-interpreter-exec console "environment-cd /Volumes/Data/VScode/Renesas/quickstart_ek_ra4m2_ep"
7-gdb-set target-async on
8-gdb-set non-stop on
9-interpreter-exec console "symbol-file /Volumes/Data/VScode/Renesas/quickstart_ek_ra4m2_ep/build/quickstart_ek_ra4m2_ep.elf"
10-interpreter-exec console "monitor reset halt"
11-target-download
12-interpreter-exec console "monitor reset halt"
13-enable-pretty-printing
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-added","output":[["id","i1"]]}]}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"Reading symbols from /Volumes/Data/VScode/Renesas/quickstart_ek_ra4m2_ep/build/quickstart_ek_ra4m2_ep.elf..."}]}
Reading symbols from /Volumes/Data/VScode/Renesas/quickstart_ek_ra4m2_ep/build/quickstart_ek_ra4m2_ep.elf...
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"done.\n"}]}
done.
GDB -> App: {"token":1,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"token":2,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"token":3,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"cmd-param-changed","output":[["param","print asm-demangle"],["value","on"]]}]}
GDB -> App: {"token":4,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-group-started","output":[["id","i1"],["pid","1"]]}]}
GDB -> App: {"outOfBandRecord":[{"isStream":false,"type":"notify","asyncClass":"thread-created","output":[["id","1"],["group-id","i1"]]}]}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"thread.c:85: internal-error: struct thread_info *inferior_thread(void): Assertion `tp' failed.\nA problem internal to GDB has been detected,\nfurther debugging may prove unreliable.\nQuit this debugging session? "}]}
thread.c:85: internal-error: struct thread_info *inferior_thread(void): Assertion `tp' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? 
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"(y or n) [answered Y; input not from terminal]\n"}]}
(y or n) [answered Y; input not from terminal]
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"log","content":"\nThis is a bug, please report it."}]}

This is a bug, please report it.
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"log","content":"  For instructions, see:\n<http://www.gnu.org/software/gdb/bugs/>."}]}
  For instructions, see:
<http://www.gnu.org/software/gdb/bugs/>.
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"log","content":"\n\n"}]}

GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"thread.c:85: internal-error: struct thread_info *inferior_thread(void): Assertion `tp' failed.\nA problem internal to GDB has been detected,\nfurther debugging may prove unreliable.\nCreate a core file of GDB? "}]}
thread.c:85: internal-error: struct thread_info *inferior_thread(void): Assertion `tp' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Create a core file of GDB? 
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"(y or n) [answered Y; input not from terminal]\n"}]}
(y or n) [answered Y; input not from terminal]
14-break-delete
15-target-detach
```

This is my launch config:
```
"name": "J-Link RA4M2",
"cwd": "${workspaceFolder}",
"executable": "${command:cmake.launchTargetPath}",
"request": "launch",
"type": "cortex-debug",
"runToEntryPoint": "Reset_Handler",
"servertype": "external",
"gdbTarget": "192.168.1.243:52222",
"showDevDebugOutput": true,
"debuggerArgs": [
     "-data-directory=/Users/rolf/Downloads/gdb-7.8.2/gdb/data-directory"
],
"gdbPath": "${workspaceRoot}/script/gdb",
"preLaunchCommands": [
     "environment-cd ${workspaceRoot}",
     "-gdb-set target-async on",
     "-gdb-set non-stop on",
     "symbol-file ${workspaceRoot}/build/${workspaceFolderBasename}.elf",
],
"postDebugTask": "reset",
"device": "R7FA4M2AD"
```

Regards,
Rolf

PS: please see the GDB build for MAC attached.
[gdb.zip](https://github.com/Marus/cortex-debug/files/7789702/gdb.zip)

haneefdm, 2021-12-31:
Can you make sure you can debug on the command-line first? It appears that gdb is crashing
```
thread.c:85: internal-error: struct thread_info *inferior_thread(void): Assertion `tp' failed.
```
4th command returned a result but looks like it failed on the 5th command where it tries to connect to the specified gdb-server
```
5-target-select extended-remote 192.168.1.243:52222
```
It seems to have died trying to get program/thread information from the server. I don't think I can just look at the GDB you attached because it needs the whole setup to see what the problem is.

RolfNoot, 2021-12-31:
That's correct. It fails when the command '-gdb-set non-stop on' isn't sent before connecting. For some reason the init commands as shown in my previous post should be sent in that order.

I tried the same on the eclipse environment and there also gdb quits when the command isn't sent.

Renesas is really quirky...

RolfNoot, 2021-12-31:
With '-gdb-set non-stop on'
```
=thread-group-added,id="i1"
~"GNU gdb (GDB) 7.8.2-20210723-Build_1\n"
~"Copyright (C) 2014 Free Software Foundation, Inc.\n"
~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law.  Type \"show copying\"\nand \"show warranty\" for details.\n"
~"This GDB was configured as \"--host=i686-pc-mingw32 --target=arm-none-eabi\".\nType \"show configuration\" for configuration details."
~"\nFor bug reporting instructions, please see:\n"
~"<http://www.gnu.org/software/gdb/bugs/>.\n"
~"Find the GDB manual and other documentation resources online at:\n<http://www.gnu.org/software/gdb/documentation/>.\n"
~"For help, type \"help\".\n"
~"Type \"apropos word\" to search for commands related to \"word\".\n"
(gdb)
-environment-cd C:/Users/Rolf/Downloads/_quickstart2/quickstart_ek_ra4m2_ep/e2studio
^done
(gdb)
-gdb-set target-async on
^done
(gdb)
-gdb-set non-stop on
^done
(gdb)
symbol-file C:/Users/Rolf/Downloads/_quickstart2/quickstart_ek_ra4m2_ep/e2studio/Debug/quickstart_ek_ra4m2_ep.elf
&"symbol-file C:/Users/Rolf/Downloads/_quickstart2/quickstart_ek_ra4m2_ep/e2studio/Debug/quickstart_ek_ra4m2_ep.elf\n"
~"Reading symbols from C:/Users/Rolf/Downloads/_quickstart2/quickstart_ek_ra4m2_ep/e2studio/Debug/quickstart_ek_ra4m2_ep.elf..."
~"done.\n"
^done
(gdb)
target extended-remote localhost:52222
&"target extended-remote localhost:52222\n"
~"Remote debugging using localhost:52222\n"
=thread-group-started,id="i1",pid="1"
=thread-created,id="1",group-id="i1"
*running,thread-id="1"
^done
(gdb)
monitor set_target,R7FA4M2AD
&"monitor set_target,R7FA4M2AD\n"
^done
(gdb)
monitor is_target_connected
&"monitor is_target_connected\n"
@"Connection status=connected.\n"
^done
(gdb)

```

Without '-gdb-set non-stop on'  (gdb crashes indeed)

```
=thread-group-added,id="i1"
~"GNU gdb (GDB) 7.8.2-20210723-Build_1\n"
~"Copyright (C) 2014 Free Software Foundation, Inc.\n"
~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law.  Type \"show copying\"\nand \"show warranty\" for details.\n"
~"This GDB was configured as \"--host=i686-pc-mingw32 --target=arm-none-eabi\".\nType \"show configuration\" for configuration details."
~"\nFor bug reporting instructions, please see:\n"
~"<http://www.gnu.org/software/gdb/bugs/>.\n"
~"Find the GDB manual and other documentation resources online at:\n<http://www.gnu.org/software/gdb/documentation/>.\n"
~"For help, type \"help\".\n"
~"Type \"apropos word\" to search for commands related to \"word\".\n"
(gdb)
-environment-cd C:/Users/Rolf/Downloads/_quickstart2/quickstart_ek_ra4m2_ep/e2studio
^done
(gdb)
-gdb-set target-async on
^done
(gdb)
symbol-file C:/Users/Rolf/Downloads/_quickstart2/quickstart_ek_ra4m2_ep/e2studio/Debug/quickstart_ek_ra4m2_ep.elf
&"symbol-file C:/Users/Rolf/Downloads/_quickstart2/quickstart_ek_ra4m2_ep/e2studio/Debug/quickstart_ek_ra4m2_ep.elf\n"
~"Reading symbols from C:/Users/Rolf/Downloads/_quickstart2/quickstart_ek_ra4m2_ep/e2studio/Debug/quickstart_ek_ra4m2_ep.elf..."
~"done.\n"
^done
(gdb)
target extended-remote localhost:52222
&"target extended-remote localhost:52222\n"
~"Remote debugging using localhost:52222\n"
=thread-group-started,id="i1",pid="1"
=thread-created,id="1",group-id="i1"
&"\nThis is a bug, please report it."
&"  For instructions, see:\n<http://www.gnu.org/software/gdb/bugs/>."
&"\n\n"

```

RolfNoot, 2022-01-01:
I finally out the initial problem!

It seems the serverArgs didn't accept whitespaces well. I found a way around by separating the arguments as in the following example.

It now works with the internal GDB + J_Link server running on MAC. (J-Link GDB Server V7.60b, GDB GNU Arm Embedded Toolchain 10-2020-q4-major).

It would be nice if the serverArgs could be fixed.

Thanks!
Rolf

```
        {
            "name": "J-Link RA4M2",
            "cwd": "${workspaceFolder}",
            "executable": "${command:cmake.launchTargetPath}",
            "request": "launch",
            "type": "cortex-debug",
            "runToEntryPoint": "Reset_Handler",
            "servertype":"jlink",
            "serverArgs": [
                "-select","USB",
                "-endian","little",
                "-speed","500",
                "-noir",
                "-settingsfile","${workspaceRoot}/script/debug_Flat.jlink"
            ],
            "device": "R7FA4M2AD",
        }
```

haneefdm, 2022-01-01:
`serverArgs` does not handle or interpret/massage anything. The list goes verbatim to the server on its command-line. VSCode does do some substitutions but other than that, it doesn't mess with it either.

I thought you were running the server externally, so not sure how `serverArgs` entered the picture

closed 2022-01-01 by haneefdm

RolfNoot, 2022-01-01:
That's right, as the config wasn't working initially I tried the external server which worked. 

I don't know what's happening but if I change "-select","USB" into "-select USB" then it fails.

Output from terminal:
```
JLinkARM.dll V7.60b (DLL compiled Dec 22 2021 12:50:19)

WARNING: Unknown command line parameter -select USB found.
Command line: -singlerun -nogui -if swd -port 50007 -swoport 50008 -telnetport 50009 -device R7FA4M2AD -select USB -endian little -speed 500 -noir -settingsfile /Volumes/Data/VScode/Renesas/quickstart_ek_ra4m2_ep/script/debug_Flat.jlink
```

Output from Debug Console:
```
Launching server: "JLinkGDBServer" "-singlerun" "-nogui" "-if" "swd" "-port" "50007" "-swoport" "50008" "-telnetport" "50009" "-device" "R7FA4M2AD" "-select USB" "-endian" "little" "-speed" "500" "-noir" "-settingsfile" "/Volumes/Data/VScode/Renesas/quickstart_ek_ra4m2_ep/script/debug_Flat.jlink"
```

haneefdm, 2022-01-01:
The server (and the shell used to launch) sees everything in quotes as one argument. This is how things work for any command. If you specified as such then it went to the server as such. We do not split things up and if we did, then many other things will fail. 

In this case, the server even told you that there is something wrong. The server documentation probably also told you that '-server' and 'USB' are two separate arguments. Not one.

RolfNoot, 2022-01-01:
Not sure about the -server argument, but the "-endian little", "-speed 500" and "-settingsfile file.jlink" are (or should be treated as) single arguments. Please see https://wiki.segger.com/J-Link_GDB_Server

Anyway, as the arguments are passed verbatim to the server, the problem may be at the way jlinkgdbserver handles the arguments.

For now it all works with the workaround and I am fully satisfied with the extension.

Thanks.

haneefdm, 2022-01-02:
Not the way I read it. From what I see, from the example below it is very clear to me that '-speed' and '2000' have to be two separate arguments. Otherwise, they would have documented it differently. This is a very common way to document command-line arguments.

[image]