Marus/cortex-debug #558: Feature Request: Hex display toggle

pilcherd, 2021-12-31:
Hi

Would be great to have a button in the debugger area by the variables to toggle hex/dec display, sending GDB 
set output-radix 16
or
set output-radix 10

Thanks for a great extension!

Dave

haneefdm, 2021-12-31:
Thank you. We are working with VSCode real estate. The Variables window doesn't belong to us and can't add buttons. In general, VSCode frowns upon adding too many buttons anyways. Notice how few buttons/menus there are in all of VSCode.

We will see if a keyboard shortcut is possible rather than a button. Stay tuned.

Btw, if you hover on the variable name, you will get the other radix representations. You can also right-click on a variable name and set the radix to use.

RolfNoot, 2022-01-03:
+1

In the watch window there's no radix option when right clicking. Adding ',x' to the variable name will output it in hexadecimal but unfortunately this doesn't work for arrays and structures.

Or am I missing something?

pilcherd, 2022-01-03:
> Thank you. We are working with VSCode real estate. The Variables window doesn't belong to us and can't add buttons. In general, VSCode frowns upon adding too many buttons anyways. Notice how few [...]

That is good to know, I had not discovered that.  Thank you!

haneefdm, 2022-01-03:
In the next release, you will have a command to Toggle hex display for the variables window...here is a preview from the Command Palette (Command Palette is your friend btw.)

[image]

As you can see, the default keybinding is Ctrl-Shift-X but you can change that to anything you want. You also use keybindings on existing commands as well. It is just that this one has a default.

You can see it in action in the next preview releases today or tomorrow (if you are using preview releases). It will be a few weeks for the next full release.

I am closing this, the issue with the watch window is distinct and should not be mixed with this.

pilcherd, 2022-01-03:
Great. Thank you. That will be a very useful toggle!

haneefdm, 2022-01-03:
It was pushed as a preview releases at the marketplace. As usual preview releases have tons of other changes so beware.

closed 2022-01-03 by haneefdm