Marus/cortex-debug #560: Dump peripheral registers to file

bobatsar, 2022-01-04:
I would like to dump the current state of all peripherals to a file to diff it with another project.

Is there any way to dump the peripherals to a txt format?

I saw that for for the registers there is the `info registers` gdb command. Is there anything similar to dump the peripherals registers?

I found https://github.com/bnahill/PyCortexMDebug but I can't use it as I don't know how to start the GDB from the plugin in the right python virtual environment, modify gdbinit...

Is there anything like that already built in?

haneefdm, 2022-01-04:
I believe this requires a gdb compiled with python support. By default ARM does not include python with their version of GDB but you can build it yourself and use that. It is a build option for gdb.

Once you have that then you can use the scripts they mentioned. in PyCortexMDebug. You can run those scripts from the `Debug Console` of VSCode perhaps.

You may want to ask them what gdb version(s) they recommend.

haneefdm, 2022-01-04:
For stuff like tab-completion, you have to run gdb from the command line. Can't use VSCode.

bobatsar, 2022-01-12:
Ok I fixed my bug in another way but thought it could be handy to export the registers/peripherals.
I will close this for now. If anybody else thinks this is useful, reopen it.

closed 2022-01-12 by bobatsar