Marus/cortex-debug #561: Setting too many breakpoints and terminating debug session does not kill OpenOCD

bohdan-tymkiv, 2022-01-04:
Steps to reproduce: 
1. Set more breakpoints than HW supports, e.g. 7 for Cortex-M4
2. Launch the debug session

Debug session will be broken, this is kind of OKay, but VSCode and GDB get out-of-sync:
1. VSCode thinks the target is running while in fact, it is not
2. Clicking the "Pause" button has no effect
3. When debug session is terminated by clicking the "Stop" button, OpenOCD is left running in the background

Cortex-debug shows the following error in the Debug Console:

`Internal Error? Multiple results or no handler for query token '20''`

probably because GDB replies twice to the 'exec-continue' request:
```
20-exec-continue
GDB -> App: {"token":20,"outOfBandRecord":[],"resultRecords":{"resultClass":"running","results":[]}}
...
GDB -> App: {"token":20,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","Command aborted."]]}}
Internal Error? Multiple results or no handler for query token '20''
```

Is it possible to detect such situations and gracefully terminate the debug session?

Bohdan

haneefdm, 2022-01-04:
Hmmm. I was able to duplicate this. We were not expecting a `success` response and sometime later a `failure` output record for the same command. And other stuff happening in between. That is for the continue.

Even the HW breakpoints. The initial response was a success. But later it has failures on breakpoints.

To compound that, any future -exec-interrupt s fail (succeed as ^done but no actual *stopped message).

haneefdm, 2022-01-04:
Once those errors happen, I don't see a way to recover the gdb session. Everything is out of sync and it does respond properly (as expected) to many commands afterwords. Will try a few other things...

But, I did change the exiting procedure to recover from failed attempts and the server does get terminated before the debugger backend quits.

You will see the change in the preview release by tomorrow (if you are using previews) or a final release which is a couple of weeks away.
```
21-exec-continue
-> 21^running
-> *running,thread-id="all"
mi2.status = running
-> &"Warning:\n"
Warning:
-> &"Cannot insert hardware breakpoint 6.\n"
Cannot insert hardware breakpoint 6.
-> &"Cannot insert hardware breakpoint 7.\n"
Cannot insert hardware breakpoint 7.
-> &"Cannot insert hardware breakpoint 8.\n"
Cannot insert hardware breakpoint 8.
-> &"Could not insert hardware breakpoints:\n"
Could not insert hardware breakpoints:
-> &"You may have requested too many hardware breakpoints/watchpoints.\n"
You may have requested too many hardware breakpoints/watchpoints.
-> &"\n"

-> 21^error,msg="Command aborted."
Internal Error? Multiple results or no handler for query token '21''
Command aborted.
Run to 'main' timed out. Trying to pause program
22-exec-interrupt
-> 22^done
23-exec-interrupt
-> 23^done
GDB never responded to an interrupt request. Trying to end session anyways
24-break-delete
25-target-disconnect
-> 24^done
-> =thread-exited,id="1",group-id="i1"
-> =thread-group-exited,id="i1"
-> 25^done
-gdb-exit
-> ^exit
```

closed 2022-01-04 by haneefdm