Marus/cortex-debug #562: View Registers issue ：Variables Window Registers and deprecate Registers window are somewhat different

hongshui3000, 2022-01-07:
In the original window, when the mouse is swiped, the register of the changed value is always highlighted, but in the new window, once the mouse is swiped and then go back to view those changed registers, the register will not be highlighted. This is not a good change

haneefdm, 2022-01-07:
I suppose you are using the pre-release version then? That is good.

We have to deprecate. Currently, we are not showing you the correct register values -- with no hint that the info. is wrong. We have to make a choice. Accuracy, lack of features like Set Value, slower performance, etc. 

btw, the changed value feature was added by me just recently. It did not exist before that.

It is Accuracy that has bothered me for a long time. I noticed this when I first got introduced to Cortex-Debug two years ago. When working with multiple threads, processors, even multiple frames, not having accurate register values, is really bad. I wish we could have fixed this but VSCode does not provide any kind of API. I tried very to convince MS but did not fly.

haneefdm, 2022-01-07:
> This is not a good change

This is not a change. This is how VSCode works for all items in the Variables window. Also, this window is managed by Microsoft. We are doing this for accuracy.

hongshui3000, 2022-01-08:
Thanks for such a nice plugin, it's pretty close to what I want

closed 2022-01-08 by haneefdm