Marus/cortex-debug #563: Add "cortex-debug.objdumpPath" option to settings.json

nicreuss, 2022-01-08:
# Context
I am working on a project that uses the xPack package system to install the arm-gcc-none-eabi toolchain. 
xPack creates shims to the executable files (see https://xpack.github.io/).

# Current functionality
For vscode on Windows, I have to specify the extensions `.cmd` for any vscode extension to correctly use the arm binary tools 
(e.g. ms-vscode.makefile-tools and marus25.cortex-debug). 

In coretex-debug, the gdb path can be set using the option:
```json
"cortex-debug.gdbPath.windows": "${workspaceFolder}/xpacks/.bin/arm-none-eabi-gdb.cmd"
```
This option works as expected.

# Current problem
Without the ability to set the `arm-none-eabi-objdump` path, the coretex-debug extension is unable to populate the static and global variables in a debug session.

# Manual fix
Copying `arm-none-eabi-objdump.exe` into the path specified in `cortex-debug.armToolchainPath` resolved the issue. This is not ideal as the project installation will no longer be automated.

# Requested functionality
Would it be possible to add the option `cortex-debug.objdumpPath` to settings.json (and launch.json)? This would serve as a companion to the already existing `cortex-debug.gdbPath` option.

For my project, the option would take the value:
```json
"cortex-debug.objdumpPath.windows": "${workspaceFolder}/xpacks/.bin/arm-none-eabi-objdump.cmd",
```

haneefdm, 2022-01-08:
I knew this day would come. It is a bunch of busywork, but I will get it into the next release. 

My least favorite thing to do is edit large JSON files :-) Requires 6 new properties to mirror what was done for `gdbPath`

May I ask why your suffix has to be `.cmd`

nicreuss, 2022-01-09:
Thank you so much!

xPack allows one to install large binary tools. Since these tools can sometime be up to a few GB, it creates shims in the local project that point to the actual binary files (e.g. in Windows the binaries are stored in AppData). Since symlinks are unpredictable on windows, a cmd shim is used instead. As a result, the cmd extension has to be specified for windows to execute the binary.

PhilippHaefele, 2022-01-12:
According to latetst changelog (prerelease version V1.1.8) this is fixed
> Feature: you can now customize the objdumpPath like you can the gdbPath you your settings or in launch.json

haneefdm, 2022-01-14:
Just pushed out the latest pre-release 1.1.8? Let us know of any issues

closed 2022-01-14 by haneefdm

nicreuss, 2022-01-16:
Tested and its working! Thank you for this feature

haneefdm, 2022-01-16:
Welcome. Keep up the good work. I see a few people using xPack (when they submit issues with their launch.json).