Marus/cortex-debug #564: Can it be used with the debug virtual plugin?

hongshui3000, 2022-01-10:
vscode debug virtual：https://github.com/hediet/vscode-debug-visualizer

haneefdm, 2022-01-10:
That is nice. Yes, they can use cortex-debug as a supported debugger since we follow the same protocol specification all the other debuggers in VSCode have to. We don't have to do anything. It is their extension that will do it. When I see their code, they specialize a bit for python and JS but everything else should just work. Try it and let them (and us) know.

I see how they do it. They also have the same problem as my registers view but it should work (sometimes). Did you try their extension? It should just work. If not file a `feature request` with them and I can help if needed. They try to evaluate expressions using a frameId and that up and may or may not work.

See, this is the [problem they have](https://github.com/hediet/vscode-debug-visualizer/blob/e01c09eac58341a0660112216b4c4f0390189e92/extension/src/proxies/DebuggerViewProxy.ts#L8) and I even own the debugger and yet can't tell what the current program/thread/frame is.

hongshui3000, 2022-01-11:
I tried it and it doesn't seem to work   https://github.com/hediet/vscode-debug-visualizer/issues/131

hongshui3000, 2022-01-11:
Also I think that if you can configure the functions mentioned here to work together, it may be very good for debugging some code like state machines

haneefdm, 2022-01-11:
You tried it on an expression with a side effect? How about starting with baby steps. Like a normal looking variable pointing to a data structure, linked list or a tree? It appears that nothing was evaluated and nothing was found.

Also, you can see in our Debug Console, what exactly happened.

hongshui3000, 2022-01-11:
[image]
It seems that the return does not meet the requirements

haneefdm, 2022-01-11:
You got exactly what you asked for. Request `194` was `watch test` and that added a watch point for a variable called `test` and the response was `done` from gdb. Not  by us. Whatever you type there is going straight to gdb. Find out from them what you are supposed to put in that box. Mabe it is `print test` or just `test` or any C-expression and not a gdb command.

You can see it in the demo a little bit. Try to follow that.

hongshui3000, 2022-01-11:
[image]
I can only write gdb commands in the dialog

haneefdm, 2022-01-11:
You can do `p test` to get the value of test on the console.

But, other than JavaScript, it wont work. You have to create JSON strings in your program which is a waste of time and space. What the are expecting is that `test` evaluates to a JSON string that describes a data structure.

I will have to close this issue.

closed 2022-01-12 by hongshui3000