Marus/cortex-debug #565: Cortex peripherals not working - 1. PeripheralNode.collectRanges: address-range verification failed. Please report this problem to Cortex-Debug repo

cmorganBE, 2022-01-10:
Cortex peripherals window has: 'Unable to load SVD file: /home/cmorgan/.vscode/extensions/jeandudey.cortex-debug-dp-stm32h7-1.0.0/data/stm32h743.svd'

Verified this file exists.

Using cortex-debug with "Cortex-Debug: Device Support Pack - STM32H7" v1.0.0 by jeandudey

Seeing the error in the subject in the runtime status window for 'cortex-debug'.

launch.json of:

```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Cortex Debug",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/build/zephyr/zephyr.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "device": "STM32H743zi",
            "configFiles": [
                "${workspaceRoot}/zephyr/boards/arm/nucleo_h743zi/support/openocd.cfg"
            ],
            "gdbPath": "/home/cmorgan/zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gdb-no-py",
        },
    ]
}
```

cmorganBE, 2022-01-10:
I'm able to set breakpoints and debug etc so things are partially working, just not the svd loading.

haneefdm, 2022-01-10:
Probably means we encountered some errors in parsing the svd file. Can you compress and attach the svd file?

Also, did you try directly adding an `"svdFile": "xxx"` entry to your launch.json to see if the other extension is a problem? This way, we can rule out a few things. You should get the same error/behavior. Were there any popup boxes for errors/warnings?

I don't want to install that extension.

cmorganBE, 2022-01-10:
@haneefdm I can appreciate not wanting to install that extension. The svd file from the extension didn't load properly but I downloaded one from https://github.com/saschagrunert/stm32h7-rs/blob/master/utils/STM32H7x3.svd and that one works correctly.

I selected that extension because it looked like it was in the same vein as the other cortex extensions.

No popups or warnings either using "device": "stm32h7" or "svdFile": working file or non-working svd file.
[stm32h743.zip](https://github.com/Marus/cortex-debug/files/7841556/stm32h743.zip)

haneefdm, 2022-01-10:
@cmorganBE No, our extension is the entire debugger. Those other extensions are either clones (old) or just distribute svd files.

Yes, folks are creating other extensions or our extension API without our knowledge and we don't know what is in them. I am sure they are just fine, more power to them, but we can't control or monitor or debug them.

Wonder what the difference is between the one that did not work and the one that did. The one you attached though works just fine

cmorganBE, 2022-01-11:
I reached out to the author of the extension with the issue. It looks like it was created to match your extensions that act as device packs and we could really use stm32h7 support (the other file svdFile workaround works for now).

@haneefdm would you accept a docs PR showing our example vscode zephyr cortex-debug launch.json file? As far as we know there is no example of this use case on the web and it would make it easier to get things working.

haneefdm, 2022-01-11:
Can you attach the file that is not working. If you saw the pathname of the svd file and it is correct, then the extension is doing fine. Just attach the file that was not working with. I will do my best to fix the issue. Just need the actual file that is not working. Is it this *exact* file?

https://github.com/jeandudey/cortex-debug-dp-stm32h7/blob/main/data/stm32h743.svd

Our documentation is mostly on the WiKi. Anyone can edit. We can create a section for Zephyr.

haneefdm, 2022-01-11:
It appears it was a problem with that file some time ago and is fixed.

https://github.com/Marus/cortex-debug/blob/master/CHANGELOG.md#v110

Unfortunately, we don't have a new release since and there have been a TON of changes since. We are doing pre-release though and I will appreciate anyone trying them out. You can enable/subscribe pre-releases in VSCode yourself. You can switch back and forth or even switch to an older version.

[image]

cmorganBE, 2022-01-11:
That is exactly the file not working here. I pointed at it using its path using:

```
/home/cmorgan/.vscode/extensions/jeandudey.cortex-debug-dp-stm32h7-1.0.0/data/stm32h743.svd
```

as my 'svdFile' in launch.json.

This was after setting "device": "stm32h743" and this not working either, although it looks like that cortext-debug svd datapack uses "device" to find the proper svd file.

haneefdm, 2022-01-11:
Would you be willing to try and use the preview release and see if it is fixed. I cannot duplicate it in the newer version but it does fail in V 0.4.10. You can always revert back.

cmorganBE, 2022-01-11:
Sure, let me try the preview version here.

cmorganBE, 2022-01-11:
@haneefdm updated to preview, reloaded vscode after dropping the "svdFile" line from launch.json and I see cortex peripherals showing up correctly now. Looks like maybe the preview version fixed things.

haneefdm, 2022-01-11:
Unfortunately, the final release is still a couple of weeks out. If you plan to be using the STM32H7 dual core devices, this new release will be very helpful -- with better multi core/device/session support.

closed 2022-01-11 by haneefdm

cmorganBE, 2022-01-11:
@haneefdm thank you much for this plugin, no worries here about using a prerelease version either

SnazzyBlanket, 2022-07-06:
I just ran into the same Issue so can def confirm. Thanks for still allowing us to use the pre-release version as that took care of the problem and now I can finally see the peripherals !

haneefdm, 2022-07-06:
@SnazzyBlanket What version are you having an issue with? This particular issue was fixed and released long time ago. Please use the latest and file a separate issue if it is still there.

haneefdm, 2022-07-06:
@SnazzyBlanket Sorry, I misread your message. You are saying you are no longer seeing the issue.

SnazzyBlanket, 2022-07-07:
> Could we be specific with version numbers? The latest pre-release in the
> marketplace is 1.5.1. This is the version you are not seeing the issue with.
>
> The latest production release is 1.4.4. This is where you are seeing a
> problem

Sorry I am away from my computer as was just replying on my phone. Will add
the version numbers next time