Marus/cortex-debug #567: Can't restart a target if using stlink server type

mpekurin, 2022-01-12:
Cortex-Debug v0.4.10
ST-LINK_gdbserver v6.0.0

Raw output in the debug console:
```
interpreter-exec console "monitor reset halt"
@"Unknown reset option\n"
```

haneefdm, 2022-01-12:
Could you try the pre-release. I don't think the 'reset' is a problem although, you won't see that anymore. A restart was working fine for me even with that issue. And it also exists on a Launch.

You can get the pre-release from VSCode itself (marketplace).

[image]

If the problem still persists (or not), please report back.

mpekurin, 2022-01-13:
@haneefdm yep, it works fine in the pre-release.

closed 2022-01-13 by mpekurin