Marus/cortex-debug #568: Assembly does not show function names

hongshui3000, 2022-01-13:
I'm not sure if it's a vscode issue or a plugin issue
[image]

haneefdm, 2022-01-13:
Not a VSCode issue. GDB and our issue. We get just ONE line number per a set of instructions. That is all gdb gives us and that is what you see. Can we kinda figure out more lines, maybe, but we would be guessing and maybe getting it wrong in optimized code or printing too much code that is not code. Ton or work too. Will look at it after basic stuff is working.

In the example you show above, we may end up printing all/some of the comments above the function name as well. Note that the function name itself does not generate any instructions just like an assembly label. We definitely don't want to go all the way back to the previous instruction. Note that we don't parse C files or even open them and do not plan to.

Tomorrow, there will be another release. There are some bugs (when scrolling) fixed . Also, in the nest release, double-clicking on a disassembly line will sync with the source in the other window so it is easy to tell where you are. In fact perhaps better if the file was already open on the left side

hongshui3000, 2022-01-13:
Can insert blank lines between functions to separate them

haneefdm, 2022-01-13:
You mean users will insert blank lines? No, we can't ask for that.

We need to formulate a better strategy. Don't know what that is yet.

haneefdm, 2022-01-14:
Can you checkout the latest pre-release 1.1.8? Displaying up to three lines now. More than that, it looks very bad. Note that it will not show function names (neither does gdb). It will show up if it happens to be close enough to the first instructions. Reopen if needed

closed 2022-01-14 by haneefdm