Marus/cortex-debug #569: Disassembly view does not support arm(or other arch) assembly?

hongshui3000, 2022-01-13:
Unable to open disassembly view under assembly code

haneefdm, 2022-01-13:
Cannot be a VSCode issue. It has no clue what language is being used.

Yup, I can reproduce the problem. I am not even getting a request for disassembly. But it still has to be my problem because there may be missing info from GDB/us in the stack frame we supply to VSCode.

hongshui3000, 2022-01-13:
[image]
After executing this function call instruction, the display is abnormal、
[image]
After step into a few instructions, resume disassembly
[image]

haneefdm, 2022-01-13:
There are some bugs in the disassembly in 1.1.7, so not user you are seeing bad disassembly and sometimes failing/crashing which results in no disassembly.

There is nothing arch specific in the disassembly. Two things though

* We expect instructions to be on a 2-byte boundary (alignment). 
* We expect maximum instruction size to be 4 bytes.

Would these cause issues?

haneefdm, 2022-01-13:
Just FYI: you know that you can ask GDB to disassemble at a total garbage address and it will happily do that for you. The address off by two bytes can produce totally different disassembly.

haneefdm, 2022-01-13:
I found out what my problem was for assembly. Had nothing to do with assembly code regardless of the architecture. It had to do with memory regions. When VSCode makes a request, it will as for 200 instructions before the PC nad 200 after. Problem was the 200 instructions before fell into invalid memory region area. Now, I have am checking to see if the instructions are within known memory regions. I now clip the memory region to disassemble before asking gdb for instructions. You may see something like

[image]

and in the debug console, you may see something like...
[image]

You can use the `info mem` command to get your memory regions from GDB

The problem is not all targets give an accurate picture of various memory regions.

Can you go do run the `info mem` command in the Debug Console and let me know what gdb reports?

hongshui3000, 2022-01-14:
[disassembly NOTavailable.txt](https://github.com/Marus/cortex-debug/files/7867081/disassembly.NOTavailable.txt)
[disassemblyNOTavailable&&DisassemblyAvailable.txt](https://github.com/Marus/cortex-debug/files/7867082/disassemblyNOTavailable.DisassemblyAvailable.txt)
@haneefdm My problem is different from what you said. When the disassembly cannot be displayed, a huge log message is output. See the log file disassembly NOTavailable.txt

haneefdm, 2022-01-14:
Thank you, that could help. Lots of junk/empty lines in there but they should be simply be ignored already.

Now you know what we get from gdb and what we have to parse .... and fast!

But this is a bit confusing.
```
187-data-disassemble -s 0x000018ac -e 0x00001eec -- 5
188-data-disassemble -s 0x000018ac -e 0x00001eec -- 5
```
Two requests from VSCode for the same thing!?!?! And back to back!!! Even before we finished responding to the requet-187. What a waste of CPU time.

haneefdm, 2022-01-14:
Can you checkout the latest pre-release 1.1.8?

haneefdm, 2022-01-14:
[image]

hongshui3000, 2022-01-15:
I'm looking at 1.1.8 and there are still some issues
1. Function name issue
[image]
[image]
[image]

2.disassembly window  source code async 
Does it work only by double-clicking the C interleaved code in the disassembly? ?

3.The arm assembly code is disassembled, but the disassembly window cannot be opened directly from the assembly code. The disassembly code of the assembly can only be displayed when the C code is reset to the arm assembly code.

4.Temporary breakpoints were not inserted correctly
When I insert a breakpoint in the code, click the reset button, in most cases, no temporary breakpoint is inserted, and it runs directly to the code where I inserted the breakpoint.
[image]

haneefdm, 2022-01-15:
1. Function name is not an issue. You are seeing what gdb is telling us -- we cannot make stuff up. We have no intention of showing function names. Only disassembly with some source code around it. If it happens to be a function name, so be  it. You know that the compiler can totally rearrange your code and sometimes assembly can be hard to correlate back to source on an exact basis. If that is the case, your code would run 10 to 100 times slower. Please compare what you see with what you can find with gdb and if it is different, then we consider it a bug. Otherwise, no.
2. There is no auto sync between disassembly and source code and if you think about it you will know why. Besides, we don't control that.
3. For 3, I have no idea and we don't control it. But I can look into it. Very low priority for me. All disassembly happens around the stack trace anyways. You also cannot open a random file and ask for disassembly right now. VSCode does not support that and it is not in the spec. either.
4. That is a totally separate topic. It may be a real issue, but we should do that separately. I need to start with your launch.json and many other details which I have none.
5. For 5 I will look into that. It was working in the last build, I may have broken it because of other changes.

From that list, I will only look at last item. Can you provide details for 4th item in a separate issue?

hongshui3000, 2022-01-15:
> 

> 1. Function name is not an issue. You are seeing what gdb is telling us -- we cannot make stuff up. We have no intention of showing function names. Only disassembly with some source code around it. [...]
details for 4th item
https://github.com/Marus/cortex-debug/issues/570

haneefdm, 2022-01-16:
For # 3, sse https://code.visualstudio.com/docs/languages/overview#_adding-a-file-extension-to-a-language

What file extension are you using for assembly? We can add it to our debugger or you can change your file extension to asm.

hongshui3000, 2022-01-16:
my file ：reset.S

hongshui3000, 2022-01-17:
FOR #3, What I'm curious about is that there is no option to open disassebly view on .S files, but there is on .c files. I feel that the bottom layer of vscode only recognizes C/C++?
.S file
[image]

.C file
[image]

closed 2022-01-17 by haneefdm