Marus/cortex-debug #570: reset exception

hongshui3000, 2022-01-14:
```
"runToEntryPoint": "__start",
```
1.
[image]
log
```
Temporary breakpoint 1, z_arm_reset () at E:/work/TP/IOT/zephyr/arch/arm/core/aarch32/cortex_m\reset.S:104
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"104\t    movs.n r0, #_EXC_IRQ_DEFAULT_PRIO\n"}]}
104	    movs.n r0, #_EXC_IRQ_DEFAULT_PRIO
```
2.
[image]

3.

[image]

haneefdm, 2022-01-14:
What version of Cortex-Debug? It is helpful if you always let me know. Behavior changed in version 1.1.1

https://github.com/Marus/cortex-debug/blob/master/CHANGELOG.md#v111

haneefdm, 2022-01-14:
Btw, if the program did not stop because of a breakpoint, the gdb-server (and thus) reports it as an exception. Nothing wrong with that. Reset is one such event that causes an exception, there are others and it is normal (for the tools it is normal, but you should know why for your FW in case it is a bus error or something else like brownout, watchdog, etc.)

hongshui3000, 2022-01-15:
This issue seems to be caused by the debugger not inserting temporary breakpoints, in some cases
Configured "runToEntryPoint": "__start", then every reset should generate a temporary breakpoint?? But it doesn't seem to be done in every case

haneefdm, 2022-01-15:
Can you provide the details. launch.json is a must for all such bugs. Or else how would I know what you are doing?

hongshui3000, 2022-01-15:
This operation is not a normal operation. I found it by pressing the reset button twice in a row (note that I operated the reset continuously), and the temporary breakpoint was not inserted the second time. I read it from the log. I have tried different architectures of gdb and the phenomenon is the same

haneefdm, 2022-01-15:
I doubt this has anything to do with gdb. It has to do with our settings and what commands we send to gdb and the server via gdb. So, I do need the launch.json. For instance, certain things in that file will prevent us from setting the temp. breakpoint.

hongshui3000, 2022-01-15:
```
    {
            "name": "OCEAN JLink Launch",
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "jlink",
            "cwd": "${workspaceRoot}",
            "executable": "./build/zephyr/zephyr.elf",
            "device": "cortex-m3",
            "interface": "swd",
            "runToEntryPoint": "__start",
            "svdFile":"./MY-sdk/soc/arm/ocean/ocean.svd",
            "rtos": "Zephyr",
            "ipAddress": null,
            "serialNumber": "59522101",
            "runTOMain":true,
            "showDevDebugOutput": true,
            "swoConfig":{
                "enabled":true,
                "source":"probe",
                "swoFrequency": 2000000,
                "cpuFrequency":48000000,
                "decoders": [
                    {
                        "port": 0,
                        "type": "console",
                        "label": "SWO output",
                        "encoding":"ascii"
                    }
                ],
                "graphConfig": [
                    {
                        "label": "Graph 1",
                        "timespan": 5,
                        "type": "realtime",
                        "annotate": false,
                        "maximum": 256,
                        "minimum": 0,
                        "plots": [
                            {
                                "graphId": "1", 
                                "label": "data 1",
                                "color": "#53753c"
                            },
                            {
                                "graphId": "2", 
                                "label": "data 2",
                                "color": "#955f20"
                            }
                        ]
                    },
                ]
            }
       
        
        },
```

haneefdm, 2022-01-17:
I could not duplicate the problem and neither do I see one in the comments above. In the attached log, I do see a temp. bpt set, except it never got hit even after 3-5 seconds -- like the message said and your program was halted which of course is some random location.
```
43-interpreter-exec console "monitor halt"
GDB -> App: {"token":43,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
44-interpreter-exec console "monitor reset"
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"target","content":"Resetting target\r\n"}]}
Resetting target
GDB -> App: {"token":44,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
45-break-insert -t --function __start
GDB -> App: {"token":45,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["bkpt",[["number","2"],["type","breakpoint"],["disp","del"],["enabled","y"],["addr","0x00001b18"],["file","E:/work/TP/IOT/zephyr/arch/arm/core/aarch32/cortex_m\\reset.S"],["fullname","E:\\work\\TP\\IOT\\zephyr\\arch\\arm\\core\\aarch32\\cortex_m\\reset.S"],["line","104"],["thread-groups",["i1"]],["times","0"],["original-location","-function __start"]]]]}}
46-exec-continue
```

In the bug report above, you posted this...
```
Run to '__start' timed out. Trying to pause program
```

I tested it with my own program and JLink using ST and another vendor device and it all worked fine. Now, if there is a bug in your program, that is another item. Note that when the reset is done, it not like a power on reset. It is usually a SW reset which may not reset the whole device

haneefdm, 2022-01-17:
Btw, in your launch.json the`graphConfig` is inside `swoConfig`. It should be at the same level. Graphs can be produced both from SWO and SEGGER RTT.

I am closing this now, reopen if you see a real problem. A breakpoint not being hit is not a problem -- means the PC never reached there.

closed 2022-01-17 by haneefdm

hongshui3000, 2022-01-17:
Can you provide the gdb version you are using? I'd like to try that
Below is the toolchain version I am using
```
C:\gnu_arm_embedded\bin>arm-none-eabi-gdb.exe -v
C:\gnu_arm_embedded\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.10) 10.2.90.20210621-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

haneefdm, 2022-01-17:
I use three different versions depending on the chip. The one I use mostly is the same version as yours but built a quarter before.
```
hdm@HaneefMac-2021 cortex-debug-proxy % arm-none-eabi-gdb --version
GNU gdb (GNU Arm Embedded Toolchain 10.3-2021.07) 10.2.90.20210621-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

hongshui3000, 2022-01-17:
About this problem, I already know the problem. It seems that it is caused by not resetting the PC pointer when Resetting the target.
I don't know what command the debugger will issue to reset my target

hongshui3000, 2022-01-17:
Is it possible to automatically insert a gdb command before the reset command to clear some of my registers?

hongshui3000, 2022-01-17:
In addition, my device swd does not have a reset pin, this should probably be the biggest difference with your device, so maybe your device has no problem

hongshui3000, 2022-01-17:
When I add this command it works fine and everything is what I want
```
        "preRestartCommands" : [
              "set $pc = 0"
            ],
```

haneefdm, 2022-01-17:
That is exactly what your gdb server (JLink) is supposed to do.  Not gdb. Notice that 'reset' is a JLink command. Your device vendor (who is it?) simply did not do all the things you are supposed to do. They are supposed to also write some scripts to tell JLink what reset do and what kind.

For a cortex (and most other architectures) device, it is more than setting PC to zero. You also have to set the SP (stack) and LR (link) to the right value or your stack is corrupted and you will see weird results from gdb next time you hit a breakpoint. And there is much more to do on a reset -- I can't go into all that. I would not use reset/restart if I were you.

haneefdm, 2022-01-17:
For devices with no reset pin, there is a register you can set in most cortex devices to achieve the equivalent of a reset. And no, even for my device a reset does not toggle the reset pin -- so we can tell the difference between a cold boot and a warm boot.

What happens when you issue a reset command is up to whoever makes the chip. They have to teach JLink/OpenOCD (via scripts) what to do.