Marus/cortex-debug #571: Microcontroller independent SWO init

PhilippHaefele, 2022-01-14:
I recently found following documentation:
https://developer.arm.com/documentation/ddi0489/f/debug/about-debug?lang=en

And as i have seen that there seems to be a way to get a pointer to the ITM register set, i though that this may help with the problem that registers for enabling SWO are vendor specific.

Also not sure if all Cortex-M processors have that PPB ROM table!?

Maybe i'm wrong (haven't looked into that very deep), but may help to implement a nicer solution.

Best regards,
Philipp

haneefdm, 2022-01-14:
Yes, I am familiar with that document. The thing is there is no gdb interface and every gdb-server does it differently and there are no APIs for a query. In OpenOCD (since I can write TCL scripts), I can get some of that information but it is digging a bit deep and not all vendors do it the same way.

Yes, all Cortex-M have a ROM table(s). Some entries have been renamed, versions changed but they have been there from the very first Cortex-M (and Cortex-R/A). Part of my day job is knowing that :-) There are little details like in the most recent chips, you may not see an SWO even though there is an ITM

I know we have to do a debug access port discovery and follow the ROM tables (tree) until you find what you are looking for. In our case, we have to find ITM, DWT and TPIU at the minimum. Not sure you noticed but I factored out the SWO into a gdb script that anyone can copy and modify...

https://github.com/Marus/cortex-debug/blob/master/support/gdb-swo.init

There is a C version as well and I also created a tcl script to help me/user out. I think I put this info in the WiKi for SWO

https://github.com/Marus/cortex-debug/wiki/SWO-Output#swo-configuration

PhilippHaefele, 2022-03-28:
Just found out that #405 is very similar to this issue.

Not sure which to close or maybe both as there's now a 'manual' way of getting things working.