Marus/cortex-debug #572: Can the WATCH window add monitored memory blocks?

hongshui3000, 2022-01-15:
Can the WATCH window add monitored memory blocks? Arrays can be added, but I don't know how to add a memory block such as 0x80000000~0x80001000 memory addresses. Can only add one address and one address, can it be added like an array?

hongshui3000, 2022-01-15:
Has vscode already started doing this
https://github.com/microsoft/vscode/pull/133643

haneefdm, 2022-01-15:
We had memory window(s) for several years now.

https://github.com/Microsoft/vscode-cpptools/issues/1503#issuecomment-767481312

I know about the hex editor from Microsoft and also contribute to it. We wanted to switch but they changed the UI and it interferes with debugging. See my TODO list.

https://github.com/Marus/cortex-debug/blob/master/TODO.md

Microsoft/VSCode is considering making part of VSCode instead of a separate extension and when they do, so will we.

They are still working on their stuff, while we have ours.

closed 2022-01-15 by haneefdm

haneefdm, 2022-07-26:
This now works in 1.5.3 and beyond. This is provided by the VSCode core (not sure when it happened). Memory windows can be invoked both from the Variables and Watch windows (only for pointers).