Marus/cortex-debug #573: Are there old disassembly commands in v1.1.18?

hongshui3000, 2022-01-15:
[image]
Seems like the command doesn't work and also affects other functions

haneefdm, 2022-01-15:
Details? Yes that command from MS works. I seriously doubt it affects things unrelated to MS Disassembly window.

I don't know what the title of the issue means.

hongshui3000, 2022-01-16:
after run View Dissaembly cmd,gdb doesn't work anymore and commands don't complete
log
```
Debug-32: Enqueuing {"command":"disassemble","arguments":{"function":"main","file":null},"type":"request","seq":32}
Debug-32: Dequeuing...
64-interpreter-exec console "info mem"
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"Using memory regions provided by the target.\n"}]}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"There are no memory regions defined.\n"}]}
GDB -> App: {"token":64,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
```

haneefdm, 2022-01-16:
so, what do you get when you run `info mem`? Can I see the output? This is important. Are you saying after that, disassembly fails or the entire session?

I added `info mem` 3-4 days in v1.1.8 ago for the new disassembler. DIdn't touch the old one. Also, you are comparing it to v0.4.10 which never did full disassembly.

It should work even if `info mem` returns no memory regions but will fail near the boundaries of your memories -- depends on how your gdb was compiled.

hongshui3000, 2022-01-17:
gdb cmd info mem
```
56-thread-select 1
57-stack-select-frame 0
58-interpreter-exec console "info mem"
GDB -> App: {"token":56,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["new-thread-id","1"],["frame",[["level","0"],["addr","0x00001b18"],["func","z_arm_reset"],["args",[]],["file","E:/work/TP/IOT/zephyr/arch/arm/core/aarch32/cortex_m\\reset.S"],["fullname","E:\\work\\TP\\IOT\\zephyr\\arch\\arm\\core\\aarch32\\cortex_m\\reset.S"],["line","104"],["arch","armv7"]]]]}}
GDB -> App: {"token":57,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"Using memory regions provided by the target.\n"}]}
Using memory regions provided by the target.
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"There are no memory regions defined.\n"}]}
There are no memory regions defined.
GDB -> App: {"token":58,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
{"token":58,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
59-var-update --thread 1 --frame 0 --all-values watch_825651e0ccf40bc362de6213ee3093766e3496cd0e81d9b636f64dab229d8e04
GDB -> App: {"token":59,"outOfBandRecord":[],"resultRecords":{"resultClass":"error","results":[["msg","Variable object not found"]]}}
60-var-create watch_825651e0ccf40bc362de6213ee3093766e3496cd0e81d9b636f64dab229d8e04 @ "rfreg"
GDB -> App: {"token":60,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[["name","watch_825651e0ccf40bc362de6213ee3093766e3496cd0e81d9b636f64dab229d8e04"],["numchild","0"],["value","0x0"],["type","volatile uint32_t"],["has_more","0"]]}}
```

run View Disassembly，disassembly fails and the entire session doesn't work anymore. I can't use button functions like setp into or setp out, even gdb commands can't be sent

Only the following log appears 
```
Debug-28: Enqueuing {"command":"disassemble","arguments":{"function":"main","file":null},"type":"request","seq":28}
Debug-28: Dequeuing...
62-interpreter-exec console "info mem"
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"Using memory regions provided by the target.\n"}]}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"There are no memory regions defined.\n"}]}
GDB -> App: {"token":62,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
```

haneefdm, 2022-01-17:
May **I ask you to physically type `info mem`** in the Debug console and report what you see. And hopefully that command did not corrupt gdb somehow. The command seems to have succeeded. I have a feeling something unrelated occurred. Anyways, there is also a new update that prints valid memory regions that we were able to detect. It also allows '.S' as a valid assembly extension for setting breakpoints.

haneefdm, 2022-01-17:
Okay, I see the output of manual `info mem` command above. Nothing wrong with it. And you are able to continue debugging after manually running that command?

haneefdm, 2022-01-17:
In the next pre-release (MS is pushing it out right now, may take a few mins), you should see something like the following after the first time you invoke disassembly. The first half of the table is from gdb  (which you won't have) and the second half is from my executable file
```
Note: We detected the following memory regions as valid using gdb "info mem" and "objdump -h"
    This information is used to adjust bounds only when normal disassembly fails.
================================================================================
  Using following memory regions for disassembly
================================================================================
      Size   VMA Beg   VMA End   LMA Beg   LMA End  Attributes
================================================================================
  10000000  00000000  10000000  --------  --------  rw nocache 
  00100000  10000000  10100000  --------  --------  flash blocksize 0x200 nocache 
  03f00000  10100000  14000000  --------  --------  rw nocache 
  00008000  14000000  14008000  --------  --------  flash blocksize 0x200 nocache 
  01ff8000  14008000  16000000  --------  --------  rw nocache 
  00008000  16000000  16008000  --------  --------  flash blocksize 0x200 nocache 
  7a6f8000  16008000  90700000  --------  --------  rw nocache 
  00000400  90700000  90700400  --------  --------  flash blocksize 0x400 nocache 
  6f8ffc00  90700400  00000000  --------  --------  rw nocache 
  ------------------------------------------------------------------------------
  000094fc  10008000  100114fc  --------  --------  (.text) contents alloc load readonly code
  00000008  100114fc  10011504  --------  --------  (.ARM.exidx) contents alloc load readonly data
  00000018  10011504  1001151c  --------  --------  (.copy.table) contents alloc load data
  00000008  1001151c  10011524  --------  --------  (.zero.table) contents alloc load data
  0000068c  0800828c  08008918  10011524  10011bb0  (.data) contents alloc load readonly code
  0000000c  08008918  08008924  10011bb0  10011bbc  (.cy_sharedmem) contents alloc load data
  00000094  08008928  080089bc  10011bc0  10011c54  (.noinit) alloc
  0000071c  080089bc  080090d8  10011c54  10012370  (.bss) alloc
  0003d728  080090d8  08046800  10012370  1004fa98  (.heap) alloc readonly
================================================================================
```

hongshui3000, 2022-01-17:
> Okay, I see the output of manual `info mem` command above. Nothing wrong with it. And you are able to continue debugging after manually running that command?

Yes, it's OK

hongshui3000, 2022-01-17:
new log
```
Debug-27: Enqueuing {"command":"disassemble","arguments":{"function":"main","file":null},"type":"request","seq":27}
Debug-27: Dequeuing...
66-interpreter-exec console "info mem"
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"Using memory regions provided by the target.\n"}]}
GDB -> App: {"outOfBandRecord":[{"isStream":true,"type":"console","content":"There are no memory regions defined.\n"}]}
GDB -> App: {"token":66,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
Note: We detected the following memory regions as valid using gdb "info mem" and "objdump -h"
    This information is used to adjust bounds only when normal disassembly fails.
================================================================================
  Using following memory regions for disassembly
================================================================================
      Size   VMA Beg   VMA End   LMA Beg   LMA End  Attributes
================================================================================
  ------------------------------------------------------------------------------
  000000b4  00000000  000000b4  --------  --------  (rom_start) contents alloc load code
  00003544  000000b4  000035f8  --------  --------  (text) contents alloc load readonly code
  00000008  000035f8  00003600  --------  --------  (.ARM.exidx) contents alloc load readonly data
  00000020  00003600  00003620  --------  --------  (initlevel) contents alloc load readonly data
  00000030  00003620  00003650  --------  --------  (devices) contents alloc load readonly data
  000000e8  00003650  00003738  --------  --------  (sw_isr_table) contents alloc load data
  00000008  00003738  00003740  --------  --------  (log_const_sections) contents alloc load readonly data
  00000010  00003740  00003750  --------  --------  (device_handles) contents alloc load readonly data
  00000e34  00003750  00004584  --------  --------  (rodata) contents alloc load readonly data
  00000061  80000000  80000061  00004584  000045e5  (datas) contents alloc load data
  00000008  80000064  8000006c  000045e8  000045f0  (device_states) contents alloc load data
  00000227  80000070  80000297  --------  --------  (bss) alloc
  00001100  80000298  80001398  --------  --------  (noinit) alloc
================================================================================

```

hongshui3000, 2022-01-17:
BTW v1.1.19  
5.After setting a breakpoint in the disassembly, it cannot be deleted, it can only be deleted by using the gdb command
issue still exists

haneefdm, 2022-01-17:
We are off topic now. Is your issue of not being able to do anything after invoking disassembly fixed?

You should probably look into why your chip does not report valid memory regions like ST, Infineon do. JLink should have that information or else how does it program the chip?

I will look into the breakpoint issue. I forgot all about it.

haneefdm, 2022-01-17:
breakpoint issue should be fixed in the next release. I am going to close this now because, I don't see the original issue, or the second one reported.

closed 2022-01-17 by haneefdm