Marus/cortex-debug #574: Openocd + GDB launch pb on windows

kfstudio, 2022-01-15:
I found a problem launching gdb via openocd on windows. **On mac, it's working fine**.
I use a ST-Link v2.

I cannot launch GDB and unable to start debug session : 
[image]

But If I run them independently on a terminal with same commands, there is no problem, and I can run all gdb commands : 
[image]

So I don't thik that it's a bad path or something like that.

My setup for launch cortex debug :
        {
            "name": "Debug-imst",
            "type": "cppdbg",
            "servertype": "openocd",
            "request": "launch",
            "program": "${workspaceRoot}/build/imst",
            "args": [],
            "stopAtEntry": true,
            "cwd": "${workspaceRoot}",
            "environment": [],
            "externalConsole": false,
            "debugServerArgs": "-f interface/stlink.cfg -f ${workspaceRoot}/tools/openocd/stm32l0_192K.cfg -c init -c \"reset init\"",
            "serverLaunchTimeout": 20000,
            "filterStderr": true,
            "filterStdout": true,
            "serverStarted": "Info\\ :\\ [\\w\\d\\.]*:\\ hardware",
            "setupCommands": [
                { "text": "cd ${workspaceRoot}/build" },
                { "text": "file imst", "description": "load file", "ignoreFailures": false},
                { "text": "target extended-remote localhost:3333", "description": "connect to target", "ignoreFailures": false },
                { "text": "monitor reset halt", "description": "perform a reset and halt the target", "ignoreFailures": false },
                { "text": "load", "description": "flash target", "ignoreFailures": false },
                { "text": "set output-radix 16", "description": "set the default numeric base to 16", "ignoreFailures": false }
            ],
            "logging": {
                "moduleLoad": true,
                "trace": true,
                "engineLogging": true,
                "programOutput": true,
                "exceptions": true
            },
            "linux": {
                "postDebugTask": "openOcdKill",
            },
            "MIMode": "gdb",
            "MIDebuggerPath": "arm-none-eabi-gdb.exe",
            "debugServerPath": "C:/ST_tools/xpack-openocd-0.11.0-3/bin/openocd.exe",
            "showDevOutput": true
        }

Best regards,

Kevin

haneefdm, 2022-01-15:
Ummm, this is not our debugger. You are using Microsoft debugger. See below
```
"type": "cppdbg",
```
About 99% of your launch.json is not relevant to how we do it. We also do things quite differently. Our debugger is the following
```
"type": "cortex-debug",
```

haneefdm, 2022-01-15:
I meant, very much the same result but quite a bit is automated because we only cater to embedded users and support for various things are builtin. We diverged a long time ago. You may or may not like it. You will see that the properties we expect in launch.json are very different from what MS does 

https://github.com/Marus/cortex-debug/blob/master/debug_attributes.md

Check out our Wiki and the many examples on the web. Here is one.

https://sudonull.com/post/1526-Setting-up-VSCODE-for-ARM-development-using-the-example-of-the-stm32f429i-disco-debug-board

kfstudio, 2022-01-17:
Thank you very much, it work well now 😁

closed 2022-01-17 by kfstudio